VoltCyclone/PCILeechFWGenerator #1: I cant get it to Work.

salfirmanhir, 2025-06-03:
`╭──────────────────────────────────────────────────────────────────────────── Traceback (most recent call last) ─────────────────────────────────────────────────────────────────────────────╮
│ /home/a/Documents/PCILeechFWGenerator-0.1.0/venv/lib/python3.12/site-packages/textual/widgets/_select.py:617 in _on_mount                                                                  │
│                                                                                                                                                                                            │
│   614 │   def _on_mount(self, _event: events.Mount) -> None:                                   ╭───────────────── locals ─────────────────╮                                                │
│   615 │   │   """Set initial values."""                                                        │ _event = Mount()                         │                                                │
│   616 │   │   self._setup_options_renderables()                                                │   self = Select(id='device-type-select') │                                                │
│ ❱ 617 │   │   self._init_selected_option(self._value)                                          ╰──────────────────────────────────────────╯                                                │
│   618 │                                                                                                                                                                                    │
│   619 │   def _watch_expanded(self, expanded: bool) -> None:                                                                                                                               │
│   620 │   │   """Display or hide overlay."""                                                                                                                                               │
│                                                                                                                                                                                            │
│ /home/a/Documents/PCILeechFWGenerator-0.1.0/venv/lib/python3.12/site-packages/textual/widgets/_select.py:545 in _init_selected_option                                                      │
│                                                                                                                                                                                            │
│   542 │   │   """Initialises the selected option for the `Select`."""                          ╭──────────────── locals ────────────────╮                                                  │
│   543 │   │   if hint == self.BLANK and not self._allow_blank:                                 │ hint = 'generic'                       │                                                  │
│   544 │   │   │   hint = self._options[0][1]                                                   │ self = Select(id='device-type-select') │                                                  │
│ ❱ 545 │   │   self.value = hint                                                                ╰────────────────────────────────────────╯                                                  │
│   546 │                                                                                                                                                                                    │
│   547 │   def set_options(self, options: Iterable[tuple[RenderableType, SelectType]]) -> None:                                                                                             │
│   548 │   │   """Set the options for the Select.                                                                                                                                           │
│                                                                                                                                                                                            │
│ /home/a/Documents/PCILeechFWGenerator-0.1.0/venv/lib/python3.12/site-packages/textual/widgets/_select.py:582 in _validate_value                                                            │
│                                                                                                                                                                                            │
│   579 │   │   │   # so we provide a helpful message to catch this mistake in case people didn' ╭────────────────── locals ───────────────────╮                                             │
│   580 │   │   │   # realise we use a special value to flag "no selection".                     │ help_text = ''                              │                                             │
│   581 │   │   │   help_text = " Did you mean to use Select.clear()?" if value is None else ""  │      self = Select(id='device-type-select') │                                             │
│ ❱ 582 │   │   │   raise InvalidSelectValueError(                                               │     value = 'generic'                       │                                             │
│   583 │   │   │   │   f"Illegal select value {value!r}." + help_text                           ╰─────────────────────────────────────────────╯                                             │
│   584 │   │   │   )                                                                                                                                                                        │
│   585                                                                                                                                                                                      │
╰────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────╯
InvalidSelectValueError: Illegal select value 'generic'.
`

salfirmanhir, 2025-06-03:
is there anyway you can help me on my ubuntu desktop system??

salfirmanhir, 2025-06-03:
also just tried the generate.py but it ends up here every time. 

`[*] Launching build container...
✔ docker.io/library/dma-fw:latest
Trying to pull docker.io/library/dma-fw:latest...
Error: initializing source docker://dma-fw:latest: reading manifest latest in docker.io/library/dma-fw: requested access to the resource is denied
2025-06-04 01:15:29,093 - ERROR - Container build failed after 5.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/a/Downloads/PCILeechFWGenerator-0.1.0/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 125.
2025-06-04 01:15:29,093 - ERROR - Fatal error during firmware generation: Container build failed after 5.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/a/Downloads/PCILeechFWGenerator-0.1.0/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 125.

[✗] Fatal error: Container build failed after 5.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/a/Downloads/PCILeechFWGenerator-0.1.0/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 125.
2025-06-04 01:15:29,093 - INFO - Restoring original driver binding for 0000:29:00.0 (target driver: rtl8192ce)
`

ramseymcgrath, 2025-06-04:
You were missing the image locally. 
App should now automatically build the image from cli and tui. You can use https://github.com/ramseymcgrath/PCILeechFWGenerator/blob/main/podman_demo.md#2-building-the-container-image to run it manually as well

ramseymcgrath, 2025-06-04:
Closing this, open a new issue if you still see errors in v0.1.1

closed 2025-06-04 by ramseymcgrath