VoltCyclone/PCILeechFWGenerator #14: Fatal error: Failed to bind device to vfio-pci:

k1dua, 2025-06-05:
Current driver: r8169
2025-06-05 09:15:10,672 - INFO - Binding device 0000:0d:00.0 to vfio-pci driver (current driver: r8169)
[*] Binding device to vfio-pci driver...
/bin/sh: 1: echo: echo: I/O error
2025-06-05 09:15:10,673 - ERROR - Failed to bind device to vfio-pci: Command 'echo 10ec 8161 > /sys/bus/pci/drivers/vfio-pci/new_id' returned non-zero exit status 1.
2025-06-05 09:15:10,673 - ERROR - Fatal error during firmware generation: Failed to bind device to vfio-pci: Command 'echo 10ec 8161 > /sys/bus/pci/drivers/vfio-pci/new_id' returned non-zero exit status 1.

[✗] Fatal error: Failed to bind device to vfio-pci: Command 'echo 10ec 8161 > /sys/bus/pci/drivers/vfio-pci/new_id' returned non-zero exit status 1.
2025-06-05 09:15:10,673 - INFO - Restoring original driver binding for 0000:0d:00.0 (target driver: r8169)
[*] Restoring original driver binding..

k1dua, 2025-06-05:
Tried on another one I get

Enter number: 56
2025-06-05 12:42:16,566 - INFO - Selected device: 0000:16:00.6 (VID:1022 DID:15e3)

Selected device: 0000:16:00.6 (VID:1022 DID:15e3)
2025-06-05 12:42:16,566 - INFO - Device info - IOMMU group: 28, Current driver: snd_hda_intel
IOMMU group: 28
Current driver: snd_hda_intel
2025-06-05 12:42:16,566 - INFO - Binding device 0000:16:00.6 to vfio-pci driver (current driver: snd_hda_intel)
[*] Binding device to vfio-pci driver...
2025-06-05 12:42:16,783 - INFO - Successfully bound 0000:16:00.6 to vfio-pci
2025-06-05 12:42:16,783 - INFO - Starting container build for device 0000:16:00.6 on board 35t
[*] Launching build container...
/entrypoint.sh: line 12: syntax error near unexpected token `('
2025-06-05 12:42:17,113 - ERROR - Container build failed after 0.3 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/28           --device=/dev/vfio/vfio           -v /home/oj/Downloads/PCILeechFWGenerator-main/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:16:00.6 --board 35t' returned non-zero exit status 2.
2025-06-05 12:42:17,113 - ERROR - Fatal error during firmware generation: Container build failed after 0.3 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/28           --device=/dev/vfio/vfio           -v /home/oj/Downloads/PCILeechFWGenerator-main/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:16:00.6 --board 35t' returned non-zero exit status 2.

[✗] Fatal error: Container build failed after 0.3 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/28           --device=/dev/vfio/vfio           -v /home/oj/Downloads/PCILeechFWGenerator-main/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:16:00.6 --board 35t' returned non-zero exit status 2.
2025-06-05 12:42:17,113 - INFO - Restoring original driver binding for 0000:16:00.6 (target driver: snd_hda_intel)
[*] Restoring original driver binding...
2025-06-05 12:42:17,151 - INFO - Successfully restored 0000:16:00.6 to snd_hda_intel
2025-06-05 12:42:17,151 - INFO - Driver restoration completed successfully

ramseymcgrath, 2025-06-05:
What OS and device?

ramseymcgrath, 2025-06-05:
And build vers

salfirmanhir, 2025-06-05:
i got the same problem.

<pre>Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble
</pre>
Build Version:v0.1.3

ramseymcgrath, 2025-06-05:
should be sorted in v0.1.4. Open a new issue if you see another error

closed 2025-06-05 by ramseymcgrath