VoltCyclone/PCILeechFWGenerator #31: Vivado ERRO  " /bin/sh: 28: Syntax error: "}" unexpected (expecting "then") "

huidao1942, 2025-06-10:
```
_****** Vivado v2024.1 (64-bit)
  **** SW Build 5076996 on Wed May 22 18:36:09 MDT 2024
  **** IP Build 5075265 on Wed May 22 21:45:21 MDT 2024
  **** SharedData Build 5076995 on Wed May 22 18:29:18 MDT 2024
  **** Start of session at: Tue Jun 10 11:30:32 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

INFO: [Common 17-206] Exiting Vivado at Tue Jun 10 11:30:35 2025...
/bin/sh: 28: Syntax error: "}" unexpected (expecting "then")
[*] Cleaned up temporary file: /tmp/vivado_patch__cthpv3c.tcl
Traceback (most recent call last):
  File "/home/huidao1942/PCILeechFWGenerator/src/build.py", line 1969, in <module>
    main()
  File "/home/huidao1942/PCILeechFWGenerator/src/build.py", line 1931, in main
    vivado_run(board_root, gen_tcl, patch_tcl)
  File "/home/huidao1942/PCILeechFWGenerator/src/build.py", line 1505, in vivado_run
    run(f"{vivado_exe} -mode batch -source {gen_tcl_path} -notrace")
  File "/home/huidao1942/PCILeechFWGenerator/src/build.py", line 168, in run
    subprocess.run(cmd, shell=True, check=True, **kwargs)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '/tools/Xilinx/Vivado/2024.1/bin/vivado -mode batch -source_
```

OS:Ubuntu 24.04 LTS  Desktop
Kernel: Linux 6.11.0-26-generic

Is that my problem or will be fix soon.

ramseymcgrath, 2025-06-10:
Maybe a syntax error in the tcl output, can you send your config json

huidao1942, 2025-06-10:
> Maybe a syntax error in the tcl output, can you send your config json
well,i dont know where i can find "config json",i was use default setting.
`python3 src/build.py --bdf 0000:03:00.0 --board 75t --advanced-sv
`
ps:new 2 linux ,still learning.

closed 2025-06-10 by ramseymcgrath