VoltCyclone/PCILeechFWGenerator #37: TUI - All 5 compatible yet RED X + python3: can't open file '/app/build.py

imverynewhere, 2025-06-11:
Hello,

If I try using the TUI, all Icons show as compatible yet the RED X is present and i cant continue.

[image]

Yet in the /docs/, it says this "Example: ✅🔓🛡️🔒⚡ indicates a fully ready device."

If I try doing it via command line, I get the following;

ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ sudo python3 generate.py

========================================

2025-06-11 23:31:09,117 - INFO - Selected device: 0000:29:00.0 (VID:10ec DID:b822)

Selected device: 0000:29:00.0 (VID:10ec DID:b822)
2025-06-11 23:31:09,118 - INFO - Device info - IOMMU group: 15, Current driver: vfio-pci
IOMMU group: 15
Current driver: vfio-pci
2025-06-11 23:31:09,118 - INFO - Binding device 0000:29:00.0 to vfio-pci driver (current driver: vfio-pci)
[*] Device already bound to vfio-pci driver, skipping binding process...
2025-06-11 23:31:09,118 - INFO - Device 0000:29:00.0 already bound to vfio-pci, skipping binding process
2025-06-11 23:31:09,118 - INFO - Starting container build for device 0000:29:00.0 on board 35t
[*] Launching build container...
python3: can't open file '/app/build.py': [Errno 2] No such file or directory
2025-06-11 23:31:09,614 - ERROR - Container build failed after 0.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/ubuntu/Documents/PCILeechFWGenerator/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 2.
2025-06-11 23:31:09,614 - ERROR - Fatal error during firmware generation: Container build failed after 0.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/ubuntu/Documents/PCILeechFWGenerator/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 2.

[✗] Fatal error: Container build failed after 0.5 seconds: Command 'podman run --rm -it --privileged           --device=/dev/vfio/15           --device=/dev/vfio/vfio           -v /home/ubuntu/Documents/PCILeechFWGenerator/output:/app/output           dma-fw           sudo python3 /app/build.py --bdf 0000:29:00.0 --board 35t' returned non-zero exit status 2.
2025-06-11 23:31:09,614 - INFO - Restoring original driver binding for 0000:29:00.0 (target driver: vfio-pci)
[*] Restoring original driver binding...
2025-06-11 23:31:09,664 - INFO - Successfully restored 0000:29:00.0 to vfio-pci
2025-06-11 23:31:09,664 - INFO - Driver restoration completed successfully

ramseymcgrath, 2025-06-12:
Run locally and skip podman if it gives you issues

imverynewhere, 2025-06-12:
I'm new to linux and seem to have an installation issue I guess;

ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ sudo pcileech-build --bdf 0000:29:00.0 --board 75t --donor-info-file src/donor_dump/donor_info.json
sudo: pcileech-build: command not found
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ pcileech-build --bdf 0000:29:00.0 --board 75t --donor-info-file src/donor_dump/donor_info.json
[!] Modular build system not available
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$

imverynewhere, 2025-06-12:
I've spent many hours but thinking maybe i've installed it wrongly or its not installing properly.
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ pcileech-build --help
[!] Modular build system not available
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ pcileech-build-sudo --help
Running pcileech-build with sudo and preserved Python paths...
[!] Modular build system not available
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ 

and if I try to use the TUI, they show a X, even though all 5 icons are correct.

Everything I try either shows its looking for the /app/build.py 

or ubuntu@ubuntu:~/Documents/PCILeechFWGenerator/src$ ./build.py
[!] Modular build system not available
ubuntu@ubuntu:~/Documents/PCILeechFWGenerator/src$ ./build_cli.py
[!] Modular build system not available

k1dua, 2025-06-12:
> I've spent many hours but thinking maybe i've installed it wrongly or its not installing properly. ubuntu@ubuntu:~/Documents/PCILeechFWGenerator$ pcileech-build --help [!] Modular build system not [...]

I fixed that issue but adding build.py to PCILEEECHFWDIRECTORY/app/
Then I add to change some code because it doesnt want to got to /app/build.py. so changed it to app/build.py.
But after doing that It works until this shows up [!] Modular build system not available

closed 2025-06-12 by ramseymcgrath

k1dua, 2025-06-12:
> Fixed in v0.3

Im getting this error in 0.3

imverynewhere, 2025-06-12:
Yes, me too. 

I've even tried older builds too and still same errors 

Even when it detects Vivado, it still tries to use Podman then causes that other error. 

Then get the modular fault too.