VoltCyclone/PCILeechFWGenerator #57: sudo python3 tui_generate.py

saolong666, 2025-06-17:
Error starting TUI application: Node is missing data; Check you are calling super().__init__(...) in the PCILeechTUI() constructor, before setting reactives.

ramseymcgrath, 2025-06-17:
resolved by https://github.com/ramseymcgrath/PCILeechFWGenerator/pull/58

saolong666, 2025-06-17:
ka@ka:~/PCILeechFWGenerator-main$ python3 tui_generate.py 
Error starting TUI application: Node is missing data; Check you are calling super().__init__(...) in the PCILeechTUI() constructor, before setting reactives.
ka@ka:~/PCILeechFWGenerator-main$ sudo python3 tui_generate.py 
Error starting TUI application: No module named 'textual'
ka@ka:~/PCILeechFWGenerator-main$ pip show textual
Name: textual
Version: 3.3.0

saolong666, 2025-06-17:
I installed Textual, but the program can't be found.

closed 2025-06-18 by saolong666