VoltCyclone/PCILeechFWGenerator #70: PCILeechFWGenerator-0.5.3-beta.4 Errors

sthgi, 2025-06-19:
*Please no uuids in this*

## 📦 Environment

Host OS:
Ubuntu 24.04.2 LTS
Python Version:
Python 3.12.3
Vivado Version:
2025.1
Board Target:
pcileech_75t
Device type
USB3

Running as root or sudo I get same thing:

[BehaviorProfiler] Failed to setup debugfs: Failed to setup debugfs: Cannot determine user privileges: Debugfs setup requires root privileges. Current UID: 999. Please run with sudo or as root user.
[BehaviorProfiler] Monitoring started
[BehaviorProfiler] Failed to setup debugfs for register monitoring: Failed to setup debugfs: Cannot determine user privileges: Debugfs setup requires root privileges. Current UID: 999. Please run with sudo or as root user.
[BehaviorProfiler] Captured 16907 register accesses
[BehaviorProfiler] Monitoring stopped

Then I get this:

2025-06-19 21:15:39,361 [INFO] Processing MSI-X capabilities
2025-06-19 21:15:39,361 [WARNING] Configuration space is too small or invalid
2025-06-19 21:15:39,361 [ERROR] Fallback for msix not allowed: MSI-X validation failed: MSI-X table size is zero
2025-06-19 21:15:39,361 [ERROR] PCILeech firmware generation failed: MSI-X capability processing failed: MSI-X validation failed: MSI-X table size is zero
2025-06-19 21:15:39,361 [ERROR] Build failed: Firmware generation failed: MSI-X capability processing failed: MSI-X validation failed: MSI-X table size is zero
Traceback (most recent call last):
  File "/app/src/device_clone/pcileech_generator.py", line 416, in _process_msix_capabilities
    raise PCILeechGenerationError(error_msg)
src.device_clone.pcileech_generator.PCILeechGenerationError: MSI-X validation failed: MSI-X table size is zero

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/src/device_clone/pcileech_generator.py", line 197, in generate_pcileech_firmware
    msix_data = self._process_msix_capabilities(config_space_data)
  File "/app/src/device_clone/pcileech_generator.py", line 455, in _process_msix_capabilities
    raise PCILeechGenerationError(
src.device_clone.pcileech_generator.PCILeechGenerationError: MSI-X capability processing failed: MSI-X validation failed: MSI-X table size is zero

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/app/src/build.py", line 186, in main
    artifacts = builder.build(profile_secs=args.profile)
  File "/app/src/build.py", line 96, in build
    res = self.gen.generate_pcileech_firmware()
  File "/app/src/device_clone/pcileech_generator.py", line 242, in generate_pcileech_firmware
    raise PCILeechGenerationError(f"Firmware generation failed: {e}") from e
src.device_clone.pcileech_generator.PCILeechGenerationError: Firmware generation failed: MSI-X capability processing failed: MSI-X validation failed: MSI-X table size is zero
WARNING: VFIO binding failed - running diagnostics to identify issues

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Device : 0000:03:00.0
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline
✔ Kernel modules: All VFIO modules loaded
✔ vfio-pci driver: vfio-pci registered
✔ Device: 0000:03:00.0 (0x1912:0x0014) present
✔ IOMMU group: Group 33
✔ Driver: Already bound to vfio-pci
✔ /dev/vfio node: /dev/vfio/33

INFO: Restoring 0000:03:00.0 to xhci_hcd
ERROR: Failed to write file /sys/bus/pci/drivers/xhci_hcd/bind: [Errno 16] Device or resource busy
WARNING: Error during cleanup: Failed to write file /sys/bus/pci/drivers/xhci_hcd/bind: [Errno 16] Device or resource busy
Traceback (most recent call last):
  File "/root/PCILeechFWGenerator-0.5.3-beta.4/src/cli/container.py", line 167, in run_build
    subprocess.run(podman_cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'podman run --rm --privileged                   --device=/dev/vfio/33                   --entrypoint python3                   --device=/dev/vfio/vfio                   -v /root/PCILeechFWGenerator-0.5.3-beta.4/output:/app/output                   pcileech-fw-generator:latest                   /app/src/build.py --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/PCILeechFWGenerator-0.5.3-beta.4/generate.py", line 20, in <module>
    sys.exit(cli.main())
             ^^^^^^^^^^
  File "/root/PCILeechFWGenerator-0.5.3-beta.4/src/cli/cli.py", line 213, in main
    run_build(cfg)
  File "/root/PCILeechFWGenerator-0.5.3-beta.4/src/cli/container.py", line 169, in run_build
    raise ContainerError(f"Build failed (exit {e.returncode})") from e
cli.container.ContainerError: Build failed (exit 1)

imverynewhere, 2025-06-19:
Yep, I'm getting the same. 

Ramsey is working now to fix it. I'll update you here if we get it going

ramseymcgrath, 2025-06-19:
Trunk build should read more of the registers safely

sthgi, 2025-06-20:
STEP 15/17: HEALTHCHECK CMD python3 - <<'PY'\nimport psutil, pydantic, sys; sys.exit(0)\nPY
WARN[0007] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format
--> 769338969dc1
STEP 16/17: USER appuser
WARN[0007] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format
--> 55eeb3150ebf
STEP 17/17: ENTRYPOINT ["entrypoint"]
COMMIT pcileech-fw-generator:latest
WARN[0007] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format
--> 182cc490eb68
Successfully tagged localhost/pcileech-fw-generator:latest
182cc490eb68735e0987cfd3ce1ac516ea24022cd9e9dddaf9d61d2815e9a3ff
INFO: Binding 0000:03:00.0 (IOMMU grp 33) to vfio-pci
INFO: Launching build container – board=pcileech_75t484_x1, tag=latest
2025-06-20 01:55:38,882 [INFO] Production mode enabled: All fallbacks disabled
2025-06-20 01:55:38,882 [INFO] Initializing PCILeech generator for device 0000:03:00.0
2025-06-20 01:55:38,883 [INFO] AdvancedSVGenerator initialized with PCILeech primary: True
2025-06-20 01:55:38,883 [INFO] PCILeech generator components initialized successfully
2025-06-20 01:55:38,914 [INFO] ➤ Generating PCILeech firmware …
2025-06-20 01:55:38,914 [INFO] Starting PCILeech firmware generation for device 0000:03:00.0
2025-06-20 01:55:38,914 [INFO] Capturing device behavior profile for 30.0s
[BehaviorProfiler] Starting behavior capture for 30.0s
[BehaviorProfiler] Failed to setup debugfs: Failed to setup debugfs: Cannot determine user privileges: Debugfs setup requires root privileges. Current UID: 999. Please run with sudo or as root user.
[BehaviorProfiler] Monitoring started
[BehaviorProfiler] Failed to setup debugfs for register monitoring: Failed to setup debugfs: Cannot determine user privileges: Debugfs setup requires root privileges. Current UID: 999. Please run with sudo or as root user.
[BehaviorProfiler] Captured 18846 register accesses
[BehaviorProfiler] Monitoring stopped
2025-06-20 01:56:09,120 [INFO] Captured 18846 register accesses with 1 timing patterns
2025-06-20 01:56:09,120 [INFO] Analyzing configuration space for device 0000:03:00.0
2025-06-20 01:56:09,129 [INFO] Binding device 0000:03:00.0 to VFIO for configuration space access
2025-06-20 01:56:09,129 [INFO] Binding 0000:03:00.0 (IOMMU grp 33) to vfio-pci
2025-06-20 01:56:09,129 [INFO] 0000:03:00.0 already bound to vfio-pci
2025-06-20 01:56:09,129 [INFO] Reading configuration space via VFIO device /dev/vfio/33
2025-06-20 01:56:09,447 [INFO] Successfully read 64 bytes of configuration space via VFIO
2025-06-20 01:56:09,447 [INFO] Restoring 0000:03:00.0 to vfio-pci
2025-06-20 01:56:09,447 [INFO] Cleanup complete
2025-06-20 01:56:09,447 [INFO] VFIO node /dev/vfio/33 still exists – manual cleanup may be needed

ramseymcgrath, 2025-06-20:
Fixed by https://github.com/ramseymcgrath/PCILeechFWGenerator/releases/tag/v0.5.4-beta.1