VoltCyclone/PCILeechFWGenerator #77: how to solved Wrong driver devices: [('0000:04:00.0', 'pcieport')]

ssuawf, 2025-06-21:
*Please no uuids in this*

## 📦 Environment

Host OS:
e.g., Ubuntu 24
Python Version:
e.g., 3.12
Vivado Version:
e.g., 2024
Board Target:
e.g., pcileech_75t
Device type
Network 

## 🚨Problem or Feature Description

Wrong driver devices: [('0000:04:00.0', 'pcieport')]
All devices in an IOMMU group must be bound to vfio-pci for VFIO to work.
06:14:32 - WARNING - Failed to open VFIO device FD for region info: [Errno None] Failed to read IOMMU group for 0000:05:00.0: IOMMU group 15 has devices not bound to vfio-pci:
Wrong driver devices: [('0000:04:00.0', 'pcieport')]

muttti@muttti-System-Product-Name:~$ ls -la /sys/kernel/iommu_groups/15/devices/
total 0
drwxr-xr-x 2 root root 0 Jun 21 19:40 .
drwxr-xr-x 3 root root 0 Jun 21 19:40 ..
lrwxrwxrwx 1 root root 0 Jun 21 19:40 0000:04:00.0 -> ../../../../devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:00.0
lrwxrwxrwx 1 root root 0 Jun 21 19:40 0000:05:00.0 -> ../../../../devices/pci0000:00/0000:00:02.1/0000:03:00.0/0000:04:00.0/0000:05:00.0

## 🧠 Expected Behavior

i think error because 0000:04:00.0(amd pci bridge) cant vifo patch i cant solved it i tryed diffrent pcie slot donar card but  same

## 📎 Attachments (optional)

## ✅ Checklist

I’m using the latest main branch

closed 2025-06-21 by ssuawf

imverynewhere, 2025-06-21:
Did you manage to get it to successfully generate or just fixed that 1 issue?

ssuawf, 2025-06-21:
only fix Wrong driver devices: [('0000:04:00.0', 'pcieport')] issue ,device busy other bug same