VoltCyclone/PCILeechFWGenerator #79: Failed to open /dev/vfio/10: [Errno 16] Device or resource busy

liuck1108, 2025-06-23:
*Please no uuids in this*

## 📦 Environment

Host OS:
e.g., Ubuntu 24.04
Python Version:
e.g., 3.12.3
Vivado Version:
e.g., 2022.2
Board Target:
e.g., pcileech_75t
Device type
Network

## 🚨Problem or Feature Description

08:10:12 ERROR   08:10:12 │ ERROR  │ Failed to open /dev/vfio/20: [Errno 16] Device or resource busy: '/dev/vfio/20'
08:10:12 ERROR   08:10:12 │ ERROR  │ [VFIO] Group file busy - another process may be using this VFIO group
08:10:12 ERROR   08:10:12 │ ERROR  │ [BARA] Failed to open VFIO device FD for device 0000:07:00.0: [Errno 16] Device or resource busy: '/dev/vfio/20'

Did you forget to close container_fd
in src/cli/vfio_handler.py
 - VFIOBinderImpl

## ✅ Checklist

I’m using the latest main branch

ramseymcgrath, 2025-06-23:
Whats the vfio helper output say?

liuck1108, 2025-06-23:
13:02:11 INFO ➤ Generating PCILeech firmware …
13:02:11 INFO   13:02:11 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:07:00.0
13:02:11 INFO   13:02:11 │  INFO  │ Behavior profiling disabled, skipping device behavior capture
13:02:11 INFO   13:02:11 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:07:00.0
13:02:11 INFO   13:02:11 │  INFO  │ [BIND] Current driver for 0000:07:00.0: vfio-pci
13:02:11 INFO   13:02:11 │  INFO  │ [BIND] Device 0000:07:00.0 already bound to vfio-pci
13:02:11 INFO   13:02:11 │  INFO  │ [VFIO] VFIO binding established for device 0000:07:00.0 at /dev/vfio/20
13:02:11 INFO   13:02:11 │  INFO  │ [MSIX] Analyzing configuration space for device 0000:07:00.0 (VFIO already active)
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:07:00.0/config
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:07:00.0/config
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Reading up to 4096 bytes for extended config space
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Successfully read 4096 bytes from sysfs
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Read config space for device 10ec:8168
13:02:11 INFO   13:02:11 │  INFO  │ [CNFG] Starting BAR extraction from config space (4096 bytes)
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 0 is active (non-zero): 0x00003001
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 0 properties: type=io, prefetchable=False, is_64bit=False
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] Added BAR 0: BAR 0: io @ 0x0000000000003000 (32-bit, non-prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 2 is active (non-zero): 0x64c00004
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 2 properties: type=memory, prefetchable=False, is_64bit=True
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 2 full 64-bit address: 0x0000000064c00000
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] Added BAR 2: BAR 2: memory @ 0x0000000064c00000 (64-bit, non-prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 4 is active (non-zero): 0x6210000c
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 4 properties: type=memory, prefetchable=True, is_64bit=True
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] BAR 4 full 64-bit address: 0x0000000062100000
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] Added BAR 4: BAR 4: memory @ 0x0000000062100000 (64-bit, prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [BARX] Completed BAR extraction: found 3 active BARs
13:02:11 INFO   13:02:11 │  INFO  │ Successfully extracted device information:
13:02:11 INFO   13:02:11 │  INFO  │   Vendor ID: 0x10ec
13:02:11 INFO   13:02:11 │  INFO  │   Device ID: 0x8168
13:02:11 INFO   13:02:11 │  INFO  │   Class Code: 0x020000
13:02:11 INFO   13:02:11 │  INFO  │   Revision ID: 0x07
13:02:11 INFO   13:02:11 │  INFO  │   Command: 0x0003
13:02:11 INFO   13:02:11 │  INFO  │   Status: 0x0010
13:02:11 INFO   13:02:11 │  INFO  │   Header Type: 0x00
13:02:11 INFO   13:02:11 │  INFO  │   Subsystem Vendor: 0x10ec
13:02:11 INFO   13:02:11 │  INFO  │   Subsystem Device: 0x0123
13:02:11 INFO   13:02:11 │  INFO  │   Cache Line Size: 16
13:02:11 INFO   13:02:11 │  INFO  │   Latency Timer: 0
13:02:11 INFO   13:02:11 │  INFO  │   BIST: 0x00
13:02:11 INFO   13:02:11 │  INFO  │   Total BARs found: 3
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] Active BARs found: 3
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] BAR 0: io @ 0x0000000000003000 (32-bit, non-prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] BAR 2: memory @ 0x0000000064c00000 (64-bit, non-prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [BARS] BAR 4: memory @ 0x0000000062100000 (64-bit, prefetchable)
13:02:11 INFO   13:02:11 │  INFO  │ [DEVI] Extracted device info: vendor=10ec device=8168 class=020000
13:02:11 INFO   13:02:11 │  INFO  │ [MSIX] Configuration space analyzed: VID=4332, DID=33128, Class=131072
13:02:11 INFO   13:02:11 │  INFO  │ [MSIX] Processing MSI-X capabilities
13:02:11 INFO MSI-X capability found: 4 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.10/logging/__init__.py", line 1100, in emit
    msg = self.format(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 943, in format
    return fmt.format(record)
  File "/usr/lib/python3.10/logging/__init__.py", line 681, in format
    s = self.formatMessage(record)
  File "/usr/local/lib/python3.10/dist-packages/colorlog/formatter.py", line 131, in formatMessage
    escapes = self._escape_code_map(record.levelname)
  File "/usr/local/lib/python3.10/dist-packages/colorlog/formatter.py", line 144, in _escape_code_map
    codes.setdefault("log_color", self._get_escape_code(self.log_colors, item))
  File "/usr/local/lib/python3.10/dist-packages/colorlog/formatter.py", line 167, in _get_escape_code
    return colorlog.escape_codes.parse_colors(log_colors.get(item, ""))
  File "/usr/local/lib/python3.10/dist-packages/colorlog/escape_codes.py", line 105, in parse_colors
    return "".join(escape_codes[n] for n in string.split(",") if n)
  File "/usr/local/lib/python3.10/dist-packages/colorlog/escape_codes.py", line 105, in <genexpr>
    return "".join(escape_codes[n] for n in string.split(",") if n)
KeyError: 'orange'
Call stack:
  File "/app/src/build.py", line 282, in <module>
    sys.exit(main())
[... 9 lines cut ...]
    logger.warning(
[... cut ...]
13:02:11 ERROR   13:02:11 │ ERROR  │ ======================== Failed to open /dev/vfio/20: [Errno 16] Device or resource busy: '/dev/vfio/20' =================================
13:02:11 ERROR   13:02:11 │ ERROR  │ [VFIO] Group file busy - another process may be using this VFIO group
13:02:11 ERROR   13:02:11 │ ERROR  │ Failed to build PCILeech template context: [Errno 16] Device or resource busy: '/dev/vfio/20'
13:02:11 ERROR   13:02:11 │ ERROR  │ [PCIL] PCILeech firmware generation failed: [Errno 16] Device or resource busy: '/dev/vfio/20'
13:02:11 ERROR Build failed: [Errno 16] Device or resource busy: '/dev/vfio/20'

liuck1108, 2025-06-23:
BARA] Analyzing BARs for device 0000:07:00.0 with 3 BARs
09:58:48 - INFO -   09:58:48 │  INFO  │ Raw BAR Configuration:
09:58:48 - INFO -   09:58:48 │  INFO  │ ┌─────┬─────────┬────────────┬─────────┬──────────────┬─────────┐
09:58:48 - INFO -   09:58:48 │  INFO  │ │ BAR │ Type    │ Address    │ Size    │ Prefetchable │ 64-bit  │
09:58:48 - INFO -   09:58:48 │  INFO  │ ├─────┼─────────┼────────────┼─────────┼──────────────┼─────────┤
09:58:48 - INFO -   09:58:48 │  INFO  │ │ 0   │ unknown │ 0x00003000 │ unknown │ unknown      │ unknown │
09:58:48 - INFO -   09:58:48 │  INFO  │ │ 1   │ unknown │ 0x64C00000 │ unknown │ unknown      │ unknown │
09:58:48 - INFO -   09:58:48 │  INFO  │ │ 2   │ unknown │ 0x62100000 │ unknown │ unknown      │ unknown │
09:58:48 - INFO -   09:58:48 │  INFO  │ └─────┴─────────┴────────────┴─────────┴──────────────┴─────────┘
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Analyzing BAR 0 with data: BAR 0: io @ 0x0000000000003000 (32-bit, non-prefetchable)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Attempting to get VFIO region info for BAR 0 on device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Opening VFIO device FD for device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ Starting VFIO device fd acquisition for 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [VFIO] Device 0000:07:00.0 is in IOMMU group 20
09:58:48 - ERROR -   09:58:48 │ ERROR  │ Failed to open /dev/vfio/20: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [VFIO] Group file busy - another process may be using this VFIO group
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [BARA] Failed to open VFIO device FD for device 0000:07:00.0: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - WARNING -   09:58:48 │ WARNING│ [BARA] Failed to open VFIO device FD for region info: [Errno 16] [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] BAR 0: Skipped (invalid or not implemented)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Analyzing BAR 1 with data: BAR 2: memory @ 0x0000000064c00000 (64-bit, non-prefetchable)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Attempting to get VFIO region info for BAR 1 on device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Opening VFIO device FD for device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ Starting VFIO device fd acquisition for 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [VFIO] Device 0000:07:00.0 is in IOMMU group 20
09:58:48 - ERROR -   09:58:48 │ ERROR  │ Failed to open /dev/vfio/20: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [VFIO] Group file busy - another process may be using this VFIO group
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [BARA] Failed to open VFIO device FD for device 0000:07:00.0: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - WARNING -   09:58:48 │ WARNING│ [BARA] Failed to open VFIO device FD for region info: [Errno 16] [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] BAR 1: Skipped (invalid or not implemented)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Analyzing BAR 2 with data: BAR 4: memory @ 0x0000000062100000 (64-bit, prefetchable)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Attempting to get VFIO region info for BAR 2 on device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] Opening VFIO device FD for device 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ Starting VFIO device fd acquisition for 0000:07:00.0
09:58:48 - INFO -   09:58:48 │  INFO  │ [VFIO] Device 0000:07:00.0 is in IOMMU group 20
09:58:48 - ERROR -   09:58:48 │ ERROR  │ Failed to open /dev/vfio/20: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [VFIO] Group file busy - another process may be using this VFIO group
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [BARA] Failed to open VFIO device FD for device 0000:07:00.0: [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - WARNING -   09:58:48 │ WARNING│ [BARA] Failed to open VFIO device FD for region info: [Errno 16] [Errno 16] Device or resource busy: '/dev/vfio/20'
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] BAR 2: Skipped (invalid or not implemented)
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] === BAR Configuration Summary ===
09:58:48 - INFO -   09:58:48 │  INFO  │ [BARA] No valid MMIO BARs found - all BARs are either size 0 or I/O-port
09:58:48 - INFO -   09:58:48 │  INFO  │ ======================================================================
09:58:48 - ERROR -   09:58:48 │ ERROR  │ Failed to build PCILeech template context: No valid MMIO BARs found - all BARs are either size 0 or I/O-port
09:58:48 - ERROR -   09:58:48 │ ERROR  │ [PCIL] PCILeech firmware generation failed: No valid MMIO BARs found - all BARs are either size 0 or I/O-port
09:58:48 - ERROR - Build failed: No valid MMIO BARs found - all BARs are either size 0 or I/O-port

ramseymcgrath, 2025-06-25:
Fixed with your PR ✅

closed 2025-06-25 by ramseymcgrath