VoltCyclone/PCILeechFWGenerator #81: ERROR Build failed: 'dict' object has no attribute 'to_template_context'

liuck1108, 2025-06-25:
*Please no uuids in this*

## 📦 Environment

Host OS:
e.g., Ubuntu 24.04
Python Version:
e.g., 3.12.3
Vivado Version:
e.g., 2022.2
Board Target:
e.g., pcileech_75t
Device type
realtek r8168
## 🚨Problem or Feature Description

13:50:37 INFO 13:50:37 │ INFO │ Template context built successfully with 9 top-level keys
13:50:37 INFO 13:50:37 │ INFO │ [VFIO] VFIO binding cleanup completed for device 0000:07:00.0
13:50:37 INFO 13:50:37 │ INFO │ Generating SystemVerilog modules
13:50:37 INFO 13:50:37 │ INFO │ Generating PCILeech SystemVerilog modules
13:50:37 INFO 13:50:37 │ INFO │ Generated 3 PCILeech SystemVerilog modules
13:50:37 INFO 13:50:37 │ INFO │ Generated 3 SystemVerilog modules
13:50:37 INFO 13:50:37 │ INFO │ Generating additional firmware components
13:50:37 INFO 13:50:37 │ INFO │ [PCIL] PCILeech firmware generation completed successfully
13:50:37 INFO • Wrote 3 SystemVerilog modules
13:50:37 INFO ================== ctx ===================
13:50:37 ERROR Build failed: 'dict' object has no attribute 'to_template_context'

## 🧠 Expected Behavior

What did you expect to happen?

build_pcileech_project_script

## 📎 Attachments (optional)

in src/build.py

ctx = res["template_context"]
proj_tcl = self.out_dir / "vivado_project.tcl"
build_tcl = self.out_dir / "vivado_build.tcl"
proj_tcl.write_text(self.tcl.build_pcileech_project_script(ctx))
build_tcl.write_text(self.tcl.build_pcileech_build_script(ctx))

ctx is dict not BuildContext but self.tcl.build_pcileech_project_script need BuildContext

## ✅ Checklist

I’m using the latest main branch

Baskai959, 2025-06-25:
I have the same problem too! Has it been resolved?

JaydenDan, 2025-06-25:
> 中国人吗？能加邮箱联系吗？

可以交流吗？hardyyyy0629@gmail.com交换下联系方式。

ssuawf, 2025-06-26:
> sorry! I'm not sure how to instantiate BuildContext — it seems to require several parameters, but I can't find documentation or examples explaining them. Also, the Jinja template uses different parameter names, and I'm unclear about their meanings and how they map to the context.

can you try this build.py? now im outside so cant test it

[build.txt](https://github.com/user-attachments/files/20916093/build.txt)

liuck1108, 2025-06-26:
I’m unable to test this. You need to use create_build_context from src/templating/tcl_build.py because ctx["template_context"] doesn’t include fpga_part or other board config fields.

liuck1108, 2025-06-26:
你去看这个分支 [https://github.com/liuck1108/PCILeechFWGenerator/tree/fix-template-context](url)  board_name 匹配了src/device_clone/constants.BOARD_PARTS

Baskai959, 2025-06-27:
> 你去看这个分支 [https://github.com/liuck1108/PCILeechFWGenerator/tree/fix-template-context](url) board_name 匹配了src/device_clone/constants.BOARD_PARTS

to_template_context的问题解决了，但是在执行vivado_build.tcl 时还是有许多问题，请问你能够成功build吗？

ssuawf, 2025-06-27:
20:27:13 - INFO - Starting VFIO binding process for device 0000:01:00.0
20:27:13 - INFO - Current driver for 0000:01:00.0: vfio-pci
20:27:13 - INFO - Device 0000:01:00.0 already bound to vfio-pci
20:27:13 - INFO - Launching build container – board=pcileech_75t484_x1, tag=latest
11:27:13 - INFO - Production mode enabled: All fallbacks disabled
11:27:13 - INFO -   11:27:13 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:01:00.0
11:27:13 - INFO -   11:27:13 │  INFO  │ AdvancedSVGenerator initialized with PCILeech primary: True
11:27:13 - INFO -   11:27:13 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
11:27:13 - INFO - ➤ Generating PCILeech firmware …
11:27:13 - INFO -   11:27:13 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:01:00.0
11:27:13 - INFO -   11:27:13 │  INFO  │ Behavior profiling disabled, skipping device behavior capture
11:27:13 - INFO -   11:27:13 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:01:00.0
11:27:13 - INFO -   11:27:13 │  INFO  │ [BIND] Current driver for 0000:01:00.0: vfio-pci
11:27:13 - INFO -   11:27:13 │  INFO  │ [BIND] Device 0000:01:00.0 already bound to vfio-pci
11:27:13 - INFO -   11:27:13 │  INFO  │ [VFIO] VFIO binding established for device 0000:01:00.0 at /dev/vfio/12
11:27:13 - INFO -   11:27:13 │  INFO  │ [MSIX] Analyzing configuration space for device 0000:01:00.0 (VFIO already active)
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:01:00.0/config
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:01:00.0/config
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Reading up to 4096 bytes for extended config space
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Successfully read 4096 bytes from sysfs
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Read config space for device 10ec:8125
11:27:13 - INFO -   11:27:13 │  INFO  │ [CNFG] Starting BAR extraction from config space (4096 bytes)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 0 is active (non-zero): 0x0000f001
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 0 properties: type=io, prefetchable=False, is_64bit=False
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] Added BAR 0: BAR 0: io @ 0x000000000000f000 (32-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 2 is active (non-zero): 0xf6f00004
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 2 properties: type=memory, prefetchable=False, is_64bit=True
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 2 full 64-bit address: 0x00000000f6f00000
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] Added BAR 2: BAR 2: memory @ 0x00000000f6f00000 (64-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 4 is active (non-zero): 0xf6f10004
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 4 properties: type=memory, prefetchable=False, is_64bit=True
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] BAR 4 full 64-bit address: 0x00000000f6f10000
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] Added BAR 4: BAR 4: memory @ 0x00000000f6f10000 (64-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARX] Completed BAR extraction: found 3 active BARs
11:27:13 - INFO -   11:27:13 │  INFO  │ Successfully extracted device information:
11:27:13 - INFO -   11:27:13 │  INFO  │   Vendor ID: 0x10ec
11:27:13 - INFO -   11:27:13 │  INFO  │   Device ID: 0x8125
11:27:13 - INFO -   11:27:13 │  INFO  │   Class Code: 0x020000
11:27:13 - INFO -   11:27:13 │  INFO  │   Revision ID: 0x04
11:27:13 - INFO -   11:27:13 │  INFO  │   Command: 0x0003
11:27:13 - INFO -   11:27:13 │  INFO  │   Status: 0x0010
11:27:13 - INFO -   11:27:13 │  INFO  │   Header Type: 0x00
11:27:13 - INFO -   11:27:13 │  INFO  │   Subsystem Vendor: 0x10ec
11:27:13 - INFO -   11:27:13 │  INFO  │   Subsystem Device: 0x8125
11:27:13 - INFO -   11:27:13 │  INFO  │   Cache Line Size: 16
11:27:13 - INFO -   11:27:13 │  INFO  │   Latency Timer: 0
11:27:13 - INFO -   11:27:13 │  INFO  │   BIST: 0x00
11:27:13 - INFO -   11:27:13 │  INFO  │   Total BARs found: 3
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] Active BARs found: 3
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] BAR 0: io @ 0x000000000000f000 (32-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] BAR 2: memory @ 0x00000000f6f00000 (64-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [BARS] BAR 4: memory @ 0x00000000f6f10000 (64-bit, non-prefetchable)
11:27:13 - INFO -   11:27:13 │  INFO  │ [DEVI] Extracted device info: vendor=10ec device=8125 class=020000
11:27:13 - INFO -   11:27:13 │  INFO  │ [MSIX] Configuration space analyzed: VID=4332, DID=33061, Class=131072
11:27:13 - INFO -   11:27:13 │  INFO  │ [MSIX] Processing MSI-X capabilities
11:27:13 - INFO - MSI-X capability found: 32 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
11:27:13 - WARNING - MSI-X table offset 0x4 is not 8-byte aligned (actual offset: 0x4, aligned: 0x0)
11:27:13 - INFO -   11:27:13 │  INFO  │ MSI-X capabilities processed: 32 vectors, table BIR 4, offset 0x0
11:27:13 - INFO -   11:27:13 │  INFO  │ Building comprehensive template context
11:27:13 - INFO - Production mode enabled: All fallbacks disabled
11:27:13 - INFO -   11:27:13 │  INFO  │ Building PCILeech template context for device 0000:01:00.0 with msix interrupts (32 vectors)
[... cut ...]
11:27:14 - INFO -   • Wrote 3 SystemVerilog modules
11:27:14 - INFO -   • Emitted Vivado scripts → vivado_project.tcl, vivado_build.tcl
11:27:14 - INFO - Build finished in 1.3 s ✓

Generated artifacts (relative to output dir):
  – device_info.json
  – vivado_project.tcl
  – src
  – vivado_build.tcl
  – src/pcileech_tlps128_bar_controller.sv
  – src/pcileech_cfgspace_coe.sv
  – src/pcileech_fifo.sv
20:27:15 - INFO - Build completed in 1.8s
muttti@muttti-System-Product-Name:~/123$ 

im patched everything  but  not  working in vivado

Baskai959, 2025-06-27:
> ## --vivado 能生成 vivado project 但是 vivado_build.tcl 会报错.
> ERRORS:
> 1. ❌ ERROR: module 'msix_table' not found (in TLP (Transaction Layer Packet) processing module)
>    📍 Location: /home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:598
>    🏷️  Type: Syntax
>    💡 Suggestion: Check generated PCILeech SystemVerilog modules for syntax issues - may need to regenerate
>    📝 Raw: ERROR: [Synth 8-439] module 'msix_table' not found [/home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:598]
> 2. ❌ ERROR: failed synthesizing module 'pcileech_tlps128_bar_controller' (in TLP (Transaction Layer Packet) processing module)
>    📍 Location: /home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:23
>    🏷️  Type: Syntax
>    💡 Suggestion: Check generated PCILeech SystemVerilog modules for syntax issues - may need to regenerate
>    📝 Raw: ERROR: [Synth 8-6156] failed synthesizing module 'pcileech_tlps128_bar_controller' [/home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:23]
> 3. ❌ ERROR: module 'msix_table' not found (in TLP (Transaction Layer Packet) processing module)
>    📍 Location: /home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:598
>    🏷️  Type: Syntax
>    💡 Suggestion: Check generated PCILeech SystemVerilog modules for syntax issues - may need to regenerate
>    📝 Raw: ERROR: [Synth 8-439] module 'msix_table' not found [/home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:598]
> 4. ❌ ERROR: failed synthesizing module 'pcileech_tlps128_bar_controller' (in TLP (Transaction Layer Packet) processing module)
>    📍 Location: /home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:23
>    🏷️  Type: Syntax
>    💡 Suggestion: Check generated PCILeech SystemVerilog modules for syntax issues - may need to regenerate
>    📝 Raw: ERROR: [Synth 8-6156] failed synthesizing module 'pcileech_tlps128_bar_controller' [/home/long/workspace/PCILeechFWGenerator-main/output/src/pcileech_tlps128_bar_controller.sv:23]
>    可能是没有生成 msix_table.sv.

我也是这样，而且vivado project.tcl还有一些bug还要手动改一下

liuck1108, 2025-06-27:
i dont know. i cant build it. when i create vivado project, it has some file.

ssuawf, 2025-06-30:
> > 我不知道。我无法构建它。当我创建 vivado 项目时，它有一些文件。
> 
> Has it been solved?

see this project https://github.com/liuck1108/PCILeechFWGenerator/tree/fix-template-context

sthgi, 2025-07-01:
This project dead?

ramseymcgrath, 2025-07-04:
Fixed in 0.5.6