VoltCyclone/PCILeechFWGenerator #84: 0.55 err

i9meng, 2025-06-26:
22:51 - INFO - Production mode enabled: All fallbacks disabled
10:22:51 - INFO -   10:22:51 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ AdvancedSVGenerator initialized with PCILeech primary: True
10:22:51 - INFO -   10:22:51 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
10:22:51 - WARNING - Device config module unavailable: attempted relative import beyond top-level package
10:22:51 - WARNING - Using fallback constants: attempted relative import beyond top-level package
10:22:51 - INFO - ➤ Generating PCILeech firmware …
10:22:51 - INFO -   10:22:51 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ Behavior profiling disabled, skipping device behavior capture
10:22:51 - INFO -   10:22:51 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ [BIND] Current driver for 0000:01:00.0: vfio-pci
10:22:51 - INFO -   10:22:51 │  INFO  │ [BIND] Device 0000:01:00.0 already bound to vfio-pci
10:22:51 - INFO -   10:22:51 │  INFO  │ [VFIO] VFIO binding established for device 0000:01:00.0 at /dev/vfio/13
10:22:51 - INFO -   10:22:51 │  INFO  │ [MSIX] Analyzing configuration space for device 0000:01:00.0 (VFIO already active)
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:01:00.0/config
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:01:00.0/config
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Reading up to 4096 bytes for extended config space
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Successfully read 4096 bytes from sysfs
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Read config space for device 1106:3483
10:22:51 - INFO -   10:22:51 │  INFO  │ [CNFG] Starting BAR extraction from config space (4096 bytes)
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARX] BAR 0 is active (non-zero): 0xf6f00004
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARX] BAR 0 properties: type=memory, prefetchable=False, is_64bit=True
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARX] BAR 0 full 64-bit address: 0x00000000f6f00000
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARS] Added BAR 0: BAR 0: memory @ 0x00000000f6f00000 (64-bit, non-prefetchable)
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARX] Completed BAR extraction: found 1 active BARs
10:22:51 - INFO -   10:22:51 │  INFO  │ Successfully extracted device information:
10:22:51 - INFO -   10:22:51 │  INFO  │   Vendor ID: 0x1106
10:22:51 - INFO -   10:22:51 │  INFO  │   Device ID: 0x3483
10:22:51 - INFO -   10:22:51 │  INFO  │   Class Code: 0x0c0330
10:22:51 - INFO -   10:22:51 │  INFO  │   Revision ID: 0x01
10:22:51 - INFO -   10:22:51 │  INFO  │   Command: 0x0003
10:22:51 - INFO -   10:22:51 │  INFO  │   Status: 0x0010
10:22:51 - INFO -   10:22:51 │  INFO  │   Header Type: 0x00
10:22:51 - INFO -   10:22:51 │  INFO  │   Subsystem Vendor: 0x1106
10:22:51 - INFO -   10:22:51 │  INFO  │   Subsystem Device: 0x3483
10:22:51 - INFO -   10:22:51 │  INFO  │   Cache Line Size: 16
10:22:51 - INFO -   10:22:51 │  INFO  │   Latency Timer: 0
10:22:51 - INFO -   10:22:51 │  INFO  │   BIST: 0x00
10:22:51 - INFO -   10:22:51 │  INFO  │   Total BARs found: 1
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARS] Active BARs found: 1
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARS] BAR 0: memory @ 0x00000000f6f00000 (64-bit, non-prefetchable)
10:22:51 - INFO -   10:22:51 │  INFO  │ [DEVI] Extracted device info: vendor=1106 device=3483 class=0c0330
10:22:51 - INFO -   10:22:51 │  INFO  │ [MSIX] Configuration space analyzed: VID=4358, DID=13443, Class=787248
10:22:51 - INFO -   10:22:51 │  INFO  │ [MSIX] Processing MSI-X capabilities
10:22:51 - INFO - MSI-X capability not found
10:22:51 - INFO -   10:22:51 │  INFO  │ MSI-X capability not found or table_size is 0
10:22:51 - INFO -   10:22:51 │  INFO  │ [PCIL] MSI-X not available, checking for MSI capability
10:22:51 - INFO -   10:22:51 │  INFO  │ [PCIL] MSI capability found, using MSI with 1 vector
10:22:51 - INFO -   10:22:51 │  INFO  │ Building comprehensive template context
10:22:51 - INFO - Production mode enabled: All fallbacks disabled
10:22:51 - INFO -   10:22:51 │  INFO  │ Building PCILeech template context for device 0000:01:00.0 with msi interrupts (1 vectors)
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARA] Analyzing BARs for device 0000:01:00.0 with 1 BARs
10:22:51 - INFO -   10:22:51 │  INFO  │ Raw BAR Configuration:
10:22:51 - INFO -   10:22:51 │  INFO  │ ┌─────┬─────────┬────────────┬─────────┬──────────────┬─────────┐
10:22:51 - INFO -   10:22:51 │  INFO  │ │ BAR │ Type    │ Address    │ Size    │ Prefetchable │ 64-bit  │
10:22:51 - INFO -   10:22:51 │  INFO  │ ├─────┼─────────┼────────────┼─────────┼──────────────┼─────────┤
10:22:51 - INFO -   10:22:51 │  INFO  │ │ 0   │ unknown │ 0xF6F00000 │ unknown │ unknown      │ unknown │
10:22:51 - INFO -   10:22:51 │  INFO  │ └─────┴─────────┴────────────┴─────────┴──────────────┴─────────┘
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARA] Analyzing BAR 0 with data: BAR 0: memory @ 0x00000000f6f00000 (64-bit, non-prefetchable)
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARA] Attempting to get VFIO region info for BAR 0 on device 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ [BARA] Opening VFIO device FD for device 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ Starting VFIO device fd acquisition for 0000:01:00.0
10:22:51 - INFO -   10:22:51 │  INFO  │ [VFIO] Device 0000:01:00.0 is in IOMMU group 13
10:22:52 - INFO -   10:22:52 │  INFO  │ [VFIO] Successfully obtained device fd 6 for 0000:01:00.0
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] Successfully opened VFIO device FD 6 and container FD 5 for device 0000:01:00.0
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] Successfully opened VFIO device FD 6 and container FD 5 for region info query
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] Querying VFIO region info for index 0 with struct size 32
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] VFIO region info successful - index: 0, size: 4096, flags: 0x7
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] VFIO region 0 properties - readable: True, writable: True, mappable: True
10:22:52 - INFO -   10:22:52 │  INFO  │ [BARA] Closing VFIO device FD 6 and container FD 5
[... 35 lines cut ...]
10:22:52 - INFO -   10:22:52 │  INFO  │ Generating PCILeech SystemVerilog modules
10:22:52 - INFO -   10:22:52 │  INFO  │ Generated 3 PCILeech SystemVerilog modules
10:22:52 - INFO -   10:22:52 │  INFO  │ Generated 3 SystemVerilog modules
10:22:52 - INFO -   10:22:52 │  INFO  │ Generating additional firmware components
10:22:52 - INFO -   10:22:52 │  INFO  │ [PCIL] PCILeech firmware generation completed successfully
10:22:52 - INFO -   • Wrote 3 SystemVerilog modules
10:22:52 - ERROR - Build failed: 'dict' object has no attribute 'to_template_context'
Traceback (most recent call last):
  File "/home/ubuntu/下载/PCILeechFWGenerator-0.5.5/src/cli/container.py", line 192, in run_build
    subprocess.run(podman_cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'podman run --rm --privileged               --device=/dev/vfio/13               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/ubuntu/下载/PCILeechFWGenerator-0.5.5/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               src/build.py --bdf 0000:01:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/ubuntu/下载/PCILeechFWGenerator-0.5.5/generate.py", line 20, in <module>
    sys.exit(cli.main())
             ^^^^^^^^^^
  File "/home/ubuntu/下载/PCILeechFWGenerator-0.5.5/src/cli/cli.py", line 221, in main
    run_build(cfg)
  File "/home/ubuntu/下载/PCILeechFWGenerator-0.5.5/src/cli/container.py", line 194, in run_build
    raise ContainerError(f"Build failed (exit {e.returncode})") from e
cli.container.ContainerError: Build failed (exit 1)

i9meng, 2025-06-26:
Python 3.12.3

i9meng, 2025-06-26:
>  This question was mentioned in #81  and should not be asked again!

Has this problem been fixed?

Baskai959, 2025-06-26:
> > This question was mentioned in [#81](https://github.com/ramseymcgrath/PCILeechFWGenerator/issues/81)  and should not be asked again!
> 
> Has this problem been fixed?

no

babydh666, 2025-06-26:
怎么联系 做固件

ramseymcgrath, 2025-06-29:
Move your folder to a path without the Chinese characters. If not have everything needed for pcileech in the /output folder to finish in vivado

closed 2025-06-29 by ramseymcgrath

i9meng, 2025-06-29:
I have moved to a pure English environment folder, but there will be errors about python, the content is the same as the #81 error, can you tell me the details of your computer environment? Python version and Linux branch version, thank you.

----------

该邮件从移动设备发送

-----------------------------------