VoltCyclone/PCILeechFWGenerator #91: ERROR: [DRC MDRV-1] Multiple Driver Nets

liuck1108, 2025-07-04:
Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[0] has multiple drivers: error_status_reg_reg[0]/Q, and error_status_reg_reg[0]__0/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[1] has multiple drivers: error_status_reg_reg[1]/Q, and error_status_reg_reg[1]__0/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[2] has multiple drivers: error_status_reg_reg[2]/Q, and error_status_reg_reg[2]__0/Q.
INFO: [Project 1-461] DRC finished with 3 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

Time (s): cpu = 00:00:00.79 ; elapsed = 00:00:00.31 . Memory (MB): peak = 1736.918 ; gain = 178.445 ; free physical = 9014 ; free virtual = 27850
INFO: [Common 17-83] Releasing license: Implementation
13 Infos, 1 Warnings, 0 Critical Warnings and 4 Errors encountered.
opt_design failed
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.

INFO: [Common 17-206] Exiting Vivado at Fri Jul  4 09:52:42 2025...
[Fri Jul  4 09:52:53 2025] impl_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'impl_1'
wait_on_runs: Time (s): cpu = 00:00:00.62 ; elapsed = 00:00:16 . Memory (MB): peak = 2390.000 ; gain = 0.000 ; free physical = 9671 ; free virtual = 28507
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run impl_1"
    (file "vivado_build.tcl" line 103)
INFO: [Common 17-206] Exiting Vivado at Fri Jul  4 09:52:53 2025...

ramseymcgrath, 2025-07-04:
Wipe the output and cache folders and rerun. You have old generated files

Baskai959, 2025-07-06:
解决了吗？

ramseymcgrath, 2025-07-09:
v0.5.7 should run cleaner

closed 2025-07-14 by ramseymcgrath