VoltCyclone/PCILeechFWGenerator #106: 0.5.9 Error

sthgi, 2025-07-14:
2025-07-14 19:08:41,976 ERROR src.templating.template_renderer: Failed to render template 'python/pcileech_build_integration.py.j2': 'python/pcileech_build_integration.py.j2' not found in search path: '/app/src/templates'
2025-07-14 19:08:41,976 ERROR src.templating.template_renderer: Failed to render template 'python/build_integration.py.j2': 'python/build_integration.py.j2' not found in search path: '/app/src/templates'
2025-07-14 19:08:41,976 ERROR src.templating.systemverilog_generator:   19:08:41 │ ERROR  │ Failed to render build integration template: Failed to render template 'python/build_integration.py.j2': 'python/build_integration.py.j2' not found in search path: '/app/src/templates'
2025-07-14 19:08:41,976 ERROR fallback_manager: Fallback for build-integration not allowed: Failed to render template 'python/build_integration.py.j2': 'python/build_integration.py.j2' not found in search path: '/app/src/templates'
2025-07-14 19:08:41,976 ERROR src.device_clone.pcileech_generator:   19:08:41 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Build integration generation failed: Failed to render template 'python/build_integration.py.j2': 'python/build_integration.py.j2' not found in search path: '/app/src/templates'
2025-07-14 19:08:41,976 ERROR pcileech_builder: Build failed: name 'extract_root_cause' is not defined
[BehaviorProfiler] Starting behavior capture for 30.0s
[BehaviorProfiler] Found device: 03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
[BehaviorProfiler] Debugfs setup completed
[BehaviorProfiler] Ftrace monitoring enabled
[BehaviorProfiler] Monitoring started
[BehaviorProfiler] Debugfs setup completed for register monitoring
[BehaviorProfiler] Captured 38728 register accesses
[BehaviorProfiler] Monitoring stopped
Traceback (most recent call last):
  File "/root/PCILeechFWGenerator/src/cli/container.py", line 192, in run_build
    subprocess.run(podman_cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'podman run --rm --privileged               --device=/dev/vfio/18               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/PCILeechFWGenerator/generate.py", line 20, in <module>
    sys.exit(cli.main())
             ^^^^^^^^^^
  File "/root/PCILeechFWGenerator/src/cli/cli.py", line 215, in main
    run_build(cfg)
  File "/root/PCILeechFWGenerator/src/cli/container.py", line 194, in run_build
    raise ContainerError(f"Build failed (exit {e.returncode})") from e
cli.container.ContainerError: Build failed (exit 1)

sthgi, 2025-07-14:
cp -R ~/PCILeechFWGenerator/src/templating/templates/python /root/PCILeechFWGenerator/src/templates fixed this issue

sthgi, 2025-07-14:
Now I am getting this

****** Vivado v2024.2 (64-bit)
  **** SW Build 5239630 on Fri Nov 08 22:34:34 MST 2024
  **** IP Build 5239520 on Sun Nov 10 16:12:51 MST 2024
  **** SharedData Build 5239561 on Fri Nov 08 14:39:27 MST 2024
  **** Start of session at: Mon Jul 14 14:22:44 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

source pcileech_tlps128_bar_controller.tcl -notrace
Command: open_checkpoint /root/PCILeechFWGenerator/output/vivado_project/pcileech_firmware.runs/impl_1/pcileech_tlps128_bar_controller.dcp
INFO: [Device 21-403] Loading part xc7a75tfgg484-2
Netlist sorting complete. Time (s): cpu = 00:00:00.01 ; elapsed = 00:00:00.01 . Memory (MB): peak = 1701.625 ; gain = 0.000 ; free physical = 17827 ; free virtual = 28591
INFO: [Netlist 29-17] Analyzing 516 Unisim elements for replacement
INFO: [Netlist 29-28] Unisim Transformation completed in 0 CPU seconds
WARNING: [Netlist 29-101] Netlist 'pcileech_tlps128_bar_controller' is not ideal for floorplanning, since the cellview 'pcileech_tlps128_bar_controller' contains a large number of primitives.  Please consider enabling hierarchy in synthesis if you want to do floorplanning.
INFO: [Project 1-479] Netlist was created with Vivado 2024.2
INFO: [Project 1-570] Preparing netlist for logic optimization
Read ShapeDB Complete: Time (s): cpu = 00:00:00 ; elapsed = 00:00:00 . Memory (MB): peak = 1761.469 ; gain = 0.000 ; free physical = 17742 ; free virtual = 28506
Netlist sorting complete. Time (s): cpu = 00:00:00 ; elapsed = 00:00:00 . Memory (MB): peak = 1762.469 ; gain = 0.000 ; free physical = 17735 ; free virtual = 28499
INFO: [Project 1-111] Unisim Transformation Summary:
  A total of 64 instances were transformed.
  RAM128X1S => RAM128X1S (MUXF7, RAMS64E(x2)): 32 instances
  RAM64X1S => RAM64X1S (RAMS64E): 32 instances

INFO: [Project 1-604] Checkpoint was created with Vivado v2024.2 (64-bit) build 5239630
open_checkpoint: Time (s): cpu = 00:00:07 ; elapsed = 00:00:07 . Memory (MB): peak = 1770.469 ; gain = 304.754 ; free physical = 17735 ; free virtual = 28499
Command: opt_design -directive Explore
INFO: [Vivado_Tcl 4-136] Directive used for opt_design is: Explore
Attempting to get a license for feature 'Implementation' and/or device 'xc7a75t'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7a75t'
Running DRC as a precondition to command opt_design

Starting DRC Task
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[0] has multiple drivers: error_status_reg_reg[0]/Q, and error_status_reg_reg[0]__0/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[1] has multiple drivers: error_status_reg_reg[1]/Q, and error_status_reg_reg[1]__0/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net error_status[2] has multiple drivers: error_status_reg_reg[2]/Q, and error_status_reg_reg[2]__0/Q.
INFO: [Project 1-461] DRC finished with 3 Errors
INFO: [Project 1-462] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-78] Error(s) found during DRC. Opt_design not run.

Time (s): cpu = 00:00:00.98 ; elapsed = 00:00:00.41 . Memory (MB): peak = 1962.445 ; gain = 172.164 ; free physical = 17626 ; free virtual = 28390
INFO: [Common 17-83] Releasing license: Implementation
13 Infos, 1 Warnings, 0 Critical Warnings and 4 Errors encountered.
opt_design failed
ERROR: [Common 17-39] 'opt_design' failed due to earlier errors.

INFO: [Common 17-206] Exiting Vivado at Mon Jul 14 14:22:54 2025...
[Mon Jul 14 14:23:04 2025] impl_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'impl_1'
wait_on_runs: Time (s): cpu = 00:00:00.72 ; elapsed = 00:00:22 . Memory (MB): peak = 2618.348 ; gain = 0.000 ; free physical = 18558 ; free virtual = 29322
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run impl_1"
    (file "vivado_build.tcl" line 103)
INFO: [Common 17-206] Exiting Vivado at Mon Jul 14 14:23:04 2025...

ramseymcgrath, 2025-07-17:
v0.6.0-beta.1 should fix all this

closed 2025-07-17 by ramseymcgrath