VoltCyclone/PCILeechFWGenerator #108: Deeper Device Cloning Support (Extended Capabilities, BAR Fingerprinting, IRQ Topology)

ChaosFPGA, 2025-07-16:
Hi Ramsey,

Thanks again for your work on PCILeechFWGenerator — it’s been an essential tool for my low-level PCIe development projects.

I'm currently working on a stealth FPGA implementation that emulates the Samsung 980 Pro NVMe device. To ensure full compatibility with Windows drivers, BIOS whitelists, and advanced kernel-mode anti-tamper tools, I’m running into the need for deeper PCIe device cloning than what’s currently supported.

I'd like to request support for advanced emulation features that enable 1:1 spoofing of the Samsung 980 Pro (and similar devices) beyond standard config space and BAR exposure.

Specifically, it would be incredibly useful if the framework could support:

- **Extended PCIe Config Space Emulation**  
  Including full support for Power Management (PM), MSI/MSI-X, PCIe Capability Structure, Vendor Specific Extended Capabilities (VSEC), and Advanced Error Reporting (AER).  
  These are required to pass driver checks and avoid Code 10 errors during initialization.

- **Accurate BAR Sizing and Flag Emulation**  
  Correct prefetchable/non-prefetchable flags, alignment, and BAR sizes are often fingerprinted. Some tools check entropy in BAR layout and will reject slightly off configurations.

- **MSI/MSI-X IRQ Table Configuration**  
  Ability to define proper interrupt vector tables, PBA bits, and mask behavior is needed for accurate IRQ emulation. Especially important for NVMe devices with MSI-X and queue IRQs.

- **Link Behavior Control**  
  Tuning PCIe link parameters such as Max_Read_Request, ASPM state transitions (L0s, L1), and hot-plug capabilities helps defeat advanced fingerprinting from some EDR tools.

- **Optional VSEC Emulation**  
  For devices like the Samsung 980 Pro, there's vendor-specific logic embedded in VSEC. Even if the functionality is stubbed, the presence and offset are required.

- **Build-Time Entropy Injection**  
  This would allow embedding a unique signature or build hash (e.g., SHA-256) into unused config registers (like hidden VSEC space). Useful for stealth deployments across multiple systems.

If you’re open to this level of enhancement, I’d be happy to assist with sample PCI config dumps from the real 980 Pro, example capability layouts, and matching firmware expectations.

Let me know if this aligns with the project’s goals or if it would be better suited as an optional extension.

Thanks for your time and continued work.

ramseymcgrath, 2025-07-17:
will make feature tickets, here are a couple of those
https://github.com/ramseymcgrath/PCILeechFWGenerator/pull/111
(irq, sizing, some config space improvements)

ramseymcgrath, 2025-07-18:
Config space dynamic feature done, rng logic done

ramseymcgrath, 2025-08-01:
Done in v0.9.0

closed 2025-08-01 by ramseymcgrath