VoltCyclone/PCILeechFWGenerator #112: 0.6.0 error

Baskai959, 2025-07-17:
2025-07-17 08:34:51,231 INFO WritemaskGenerator: Generating writemask from /app/output/systemverilog/pcileech_cfgspace.coe
2025-07-17 08:34:51,231 ERROR WritemaskGenerator: Failed to read configuration space: [Errno 2] No such file or directory: '/app/output/systemverilog/pcileech_cfgspace.coe'
2025-07-17 08:34:51,231 WARNING src.device_clone.pcileech_generator:   08:34:51 │ WARNING│ [WRMASK] Failed to generate writemask COE: [Errno 2] No such file or directory: '/app/output/systemverilog/pcileech_cfgspace.coe'
2025-07-17 08:34:51,231 INFO src.device_clone.pcileech_generator:   08:34:51 │  INFO  │ [HEX] Generating configuration space hex file
2025-07-17 08:34:51,235 ERROR src.device_clone.pcileech_generator:   08:34:51 │ ERROR  │ [HEX] Configuration space hex generation failed: No configuration space data available in template context
2025-07-17 08:34:51,235 ERROR src.device_clone.pcileech_generator:   08:34:51 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Config space hex generation failed: No configuration space data available in template context
2025-07-17 08:34:51,235 ERROR pcileech_builder: Build failed: No configuration space data available in template context
2025-07-17 08:34:51,235 ERROR pcileech_builder: Unexpected error: No configuration space data available in template context
[BehaviorProfiler] Starting behavior capture for 30.0s
[BehaviorProfiler] Found device: 06:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
[BehaviorProfiler] Debugfs setup completed
[BehaviorProfiler] Ftrace monitoring enabled
[BehaviorProfiler] Monitoring started
[BehaviorProfiler] Debugfs setup completed for register monitoring
[BehaviorProfiler] Captured 25452 register accesses
[BehaviorProfiler] Monitoring stopped
Traceback (most recent call last):
  File "/home/zjy/PCILeechFWGenerator/src/cli/container.py", line 206, in run_build
    subprocess.run(podman_cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'podman run --rm --privileged               --device=/dev/vfio/19               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/zjy/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:06:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/zjy/PCILeechFWGenerator/generate.py", line 20, in <module>
    sys.exit(cli.main())
             ^^^^^^^^^^
  File "/home/zjy/PCILeechFWGenerator/src/cli/cli.py", line 252, in main
    run_build(cfg)
  File "/home/zjy/PCILeechFWGenerator/src/cli/container.py", line 208, in run_build
    raise ContainerError(f"Build failed (exit {e.returncode})") from e
cli.container.ContainerError: Build failed (exit 1)

Error message: Cannot find: /app/output/systemverilog/pcileech_cfgspace.coe

ramseymcgrath, 2025-07-17:
Can you give more of the log output? Need to see why the cfgspace wasnt made

ramseymcgrath, 2025-07-18:
Yup fixed

closed 2025-07-18 by ramseymcgrath