VoltCyclone/PCILeechFWGenerator #118: 0.6.0 beta 1 error

sthgi, 2025-07-17:
2025-07-17 14:27:17,874 INFO src.templating.systemverilog_generator:   14:27:17 │  INFO  │ Generating PCILeech SystemVerilog modules
2025-07-17 14:27:17,911 ERROR src.templating.template_renderer: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'hasattr' is undefined
2025-07-17 14:27:17,911 ERROR src.templating.systemverilog_generator:   14:27:17 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'hasattr' is undefined
2025-07-17 14:27:17,911 ERROR src.device_clone.pcileech_generator:   14:27:17 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'hasattr' is undefined
2025-07-17 14:27:17,912 ERROR pcileech_builder: Build failed: 'hasattr' is undefined
2025-07-17 14:27:17,912 ERROR pcileech_builder: Unexpected error: 'hasattr' is undefined
[BehaviorProfiler] Starting behavior capture for 30.0s
[BehaviorProfiler] Found device: 03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 04)
[BehaviorProfiler] Debugfs setup completed
[BehaviorProfiler] Ftrace monitoring enabled
[BehaviorProfiler] Monitoring started
[BehaviorProfiler] Debugfs setup completed for register monitoring
[BehaviorProfiler] Captured 37445 register accesses
[BehaviorProfiler] Monitoring stopped
Traceback (most recent call last):
  File "/root/PCILeechFWGenerator/src/cli/container.py", line 206, in run_build
    subprocess.run(podman_cmd, shell=True, check=True)
  File "/usr/lib/python3.12/subprocess.py", line 571, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'podman run --rm --privileged               --device=/dev/vfio/18               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/PCILeechFWGenerator/generate.py", line 20, in <module>
    sys.exit(cli.main())
             ^^^^^^^^^^
  File "/root/PCILeechFWGenerator/src/cli/cli.py", line 252, in main
    run_build(cfg)
  File "/root/PCILeechFWGenerator/src/cli/container.py", line 208, in run_build
    raise ContainerError(f"Build failed (exit {e.returncode})") from e
cli.container.ContainerError: Build failed (exit 1)

ssuawf, 2025-07-18:
maybe fixed 'hasattr' is undefined error

sthgi, 2025-07-18:
you fixed it @ssuawf

closed 2025-07-18 by sthgi