VoltCyclone/PCILeechFWGenerator #121: 0.6.0-beta.1 vivado errors

sthgi, 2025-07-18:
build runs clean

Generated artifacts in /app/output:

  SystemVerilog modules (5):
    - src/msix_capability_registers.sv
    - src/msix_implementation.sv
    - src/msix_table.sv
    - src/pcileech_fifo.sv
    - src/pcileech_tlps128_bar_controller.sv

  TCL scripts (2):
    - vivado_build.tcl
    - vivado_generate_project.tcl

  JSON files (2):
    - behavior_profile.json
    - device_info.json

  Other files (4):
    - src/msix_pba_init.hex
    - src/msix_table_init.hex
    - systemverilog/pcileech_cfgspace.coe
    - systemverilog/pcileech_cfgspace_writemask.coe

Total: 13 files
00:16:36 - INFO - Build completed in 71.7s

Getting this error on vivado generate:
ERROR: [Coretcl 2-101] Project '/root/PCILeechFWGenerator/output/vivado_project/pcileech_firmware.xpr' is already open.

here is the log:
vivado -mode batch -source vivado_generate_project.tcl

****** Vivado v2022.2 (64-bit)
  **** SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022
  **** IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source vivado_generate_project.tcl
# set_msg_config -id {Common 17-55} -suppress
# set_msg_config -id {Vivado 12-1387} -suppress
# set project_name "pcileech_firmware"
# set project_dir "./vivado_project"
# set fpga_part "xc7a75tfgg484-2"
# set board_name "pcileech_75t484_x1"
# set src_dir "src"
# set ip_dir "ip"
# file mkdir $project_dir
# create_project $project_name $project_dir -part $fpga_part -force
# set_property target_language Verilog [current_project]
# set_property simulator_language Mixed [current_project]
# set_property default_lib xil_defaultlib [current_project]
# file mkdir [file join $project_dir $src_dir]
# file mkdir [file join $project_dir $ip_dir]
# puts "Created PCILeech directory structure:"
Created PCILeech directory structure:
# puts "  Source directory: $src_dir"
  Source directory: src
# puts "  IP directory: $ip_dir"
  IP directory: ip
# if {[file exists $src_dir]} {
#     set sv_files [glob -nocomplain [file join $src_dir "*.sv"]]
#     set v_files [glob -nocomplain [file join $src_dir "*.v"]]
#
#     foreach sv_file $sv_files {
#         add_files -norecurse $sv_file
#         puts "  Added SystemVerilog: [file tail $sv_file]"
#     }
#
#     foreach v_file $v_files {
#         add_files -norecurse $v_file
#         puts "  Added Verilog: [file tail $v_file]"
#     }
# }
  Added SystemVerilog: pcileech_fifo.sv
  Added SystemVerilog: msix_capability_registers.sv
  Added SystemVerilog: msix_implementation.sv
  Added SystemVerilog: pcileech_tlps128_bar_controller.sv
  Added SystemVerilog: msix_table.sv
# puts "Configuring PCIe IP Core for pcie_7x..."
Configuring PCIe IP Core for pcie_7x...
# create_ip -name pcie_7x -vendor xilinx.com -library ip -module_name pcie_7x_bridge
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2022.2/data/ip'.
# set_property -dict [list \
#     CONFIG.Link_Speed {2.5_GT/s} \
#     CONFIG.Maximum_Link_Width X1 \
#     CONFIG.Device_ID 8125 \
#     CONFIG.Vendor_ID 10ec \
#     CONFIG.Revision_ID 04 \
#     CONFIG.Class_Code_Base {02} \
#     CONFIG.Class_Code_Interface {00} \
#     CONFIG.Class_Code_Sub {00} \
#     CONFIG.Subsystem_ID 7 \
#     CONFIG.Subsystem_Vendor_ID 4332 \
#     CONFIG.Bar0_Enabled {true} \
#     CONFIG.Bar0_Type {Memory} \
#     CONFIG.Bar0_64bit {true} \
#     CONFIG.Bar0_Size {1} \
#     CONFIG.MSI_Enabled {true} \
#     CONFIG.MSIx_Enabled {true} \
# ] [get_ips pcie_7x_bridge]
# puts "Adding constraint files..."
Adding constraint files...
# set xdc_files [glob -nocomplain "*.xdc"]
# foreach xdc_file $xdc_files {
#     add_files -fileset constrs_1 -norecurse $xdc_file
#     puts "  Added constraint: [file tail $xdc_file]"
# }
# set top_module ""
# set_property source_mgmt_mode None [current_project]
# if {$top_module != ""} {
#     set_property top $top_module [current_fileset]
#     puts "Set top module: $top_module"
# } else {
#     puts "Top module will be auto-detected by Vivado"
# }
Top module will be auto-detected by Vivado
# update_compile_order -fileset sources_1
# puts "Generating IP cores..."
Generating IP cores...
# generate_target all [get_files *.xci]
INFO: [IP_Flow 19-1686] Generating 'Instantiation Template' target for IP 'pcie_7x_bridge'...
INFO: [IP_Flow 19-1686] Generating 'Synthesis' target for IP 'pcie_7x_bridge'...
INFO: [IP_Flow 19-1686] Generating 'Simulation' target for IP 'pcie_7x_bridge'...
INFO: [IP_Flow 19-1686] Generating 'Change Log' target for IP 'pcie_7x_bridge'...
# catch {config_ip_cache -export [get_ips -all]}
INFO: [IP_Flow 19-6924] IPCACHE: Running cache check for IP inst: pcie_7x_bridge
# puts "Skipping design validation..."
Skipping design validation...
# puts "PCILeech project generation completed successfully!"
PCILeech project generation completed successfully!
# puts "Project: $project_name"
Project: pcileech_firmware
# puts "Location: $project_dir"
Location: ./vivado_project
# puts "FPGA Part: $fpga_part"
FPGA Part: xc7a75tfgg484-2
# puts "PCIe IP Type: pcie_7x"
PCIe IP Type: pcie_7x
# puts "Max Lanes: 8"
Max Lanes: 8
# puts "MSI Support: Enabled"
MSI Support: Enabled
# puts "MSI-X Support: Enabled"
MSI-X Support: Enabled
# save_project_as $project_name $project_dir -force
ERROR: [Coretcl 2-101] Project '/root/PCILeechFWGenerator/output/vivado_project/pcileech_firmware.xpr' is already open.
INFO: [Common 17-206] Exiting Vivado at Fri Jul 18 00:17:06 2025...

I get these errors on vivado build:

source pcileech_tlps128_bar_controller.tcl -notrace
Command: synth_design -top pcileech_tlps128_bar_controller -part xc7a75tfgg484-2 -no_iobuf
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a75t'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a75t'
INFO: [Device 21-403] Loading part xc7a75tfgg484-2
INFO: [Synth 8-7079] Multithreading enabled for synth_design using a maximum of 8 processes.
INFO: [Synth 8-7078] Launching helper process for spawning children vivado processes
INFO: [Synth 8-7075] Helper process launched with PID 43654
INFO: [Synth 8-11241] undeclared symbol 'REGCCE', assumed default net type 'wire' [/tools/Xilinx/Vivado/2022.2/data/verilog/src/unimacro/BRAM_SINGLE_MACRO.v:2170]
ERROR: [Synth 8-2716] syntax error near 'initial' [/root/PCILeechFWGenerator/output/src/msix_capability_registers.sv:18]
ERROR: [Synth 8-10307] SystemVerilog keyword 'initial' used in incorrect context [/root/PCILeechFWGenerator/output/src/msix_capability_registers.sv:18]
INFO: [Synth 8-9084] Verilog file '/root/PCILeechFWGenerator/output/src/msix_capability_registers.sv' ignored due to errors
Failed to read verilog '/root/PCILeechFWGenerator/output/src/msix_capability_registers.sv'
INFO: [Common 17-83] Releasing license: Synthesis
8 Infos, 0 Warnings, 0 Critical Warnings and 3 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Fri Jul 18 00:22:07 2025...
[Fri Jul 18 00:22:18 2025] synth_1 finished
WARNING: [Vivado 12-13638] Failed runs(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:00:46 ; elapsed = 00:01:04 . Memory (MB): peak = 2051.547 ; gain = 0.000 ; free physical = 19455 ; free virtual = 29896
# if {[get_property PROGRESS [get_runs synth_1]] != "100%"} {
#     puts "ERROR: Synthesis failed"
#     exit 1
# }
ERROR: Synthesis failed
INFO: [Common 17-206] Exiting Vivado at Fri Jul 18 00:22:18 2025...

ssuawf, 2025-07-18:
pcileech_firmware.xpr' is already open. error is fine vivado_generate_project need to fix

Baskai959, 2025-07-18:
> pcileech_firmware.xpr' is already open. error is fine vivado_generate_project need to fix

When I run vivado-build.tcl, the same error occurs.

ERROR: [Synth 8-2716] syntax error near 'initial' [/home/zjy/PCILeechFWGenerator/output/src/msix_capability_registers.sv:18]
ERROR: [Synth 8-10307] SystemVerilog keyword 'initial' used in incorrect context [/home/zjy/PCILeechFWGenerator/output/src/msix_capability_registers.sv:18]

[vivado.log](https://github.com/user-attachments/files/21310387/vivado.log)

closed 2025-07-18 by ramseymcgrath