VoltCyclone/PCILeechFWGenerator #130: Encounter import error

ApocKira, 2025-07-19:
when I run '''sudo ./dma_sw/bin/python3 ./pcileech.py build --bdf 0000:05:00.0 --board pcileech_75t484_x1''' on 0.64

Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 187, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.10/runpy.py", line 110, in _get_module_details
    __import__(pkg_name)
  File "/app/src/__init__.py", line 12, in <module>
    from .cli import BuildConfig, VFIOBinder, flash_firmware, run_build
  File "/app/src/cli/__init__.py", line 3, in <module>
    from .cli import get_parser, main
  File "/app/src/cli/cli.py", line 30, in <module>
    from .container import BuildConfig, run_build  # new unified runner
  File "/app/src/cli/container.py", line 17, in <module>
    from .vfio import (
  File "/app/src/cli/vfio.py", line 10, in <module>
    from .vfio_handler import (VFIOBinder, VFIOBindError, render_pretty,
  File "/app/src/cli/vfio_handler.py", line 39, in <module>
    from .vfio_constants import (
ImportError: cannot import name 'VFIO_GROUP_SET_CONTAINER' from 'src.cli.vfio_constants' (/app/src/cli/vfio_constants.py)
Build failed: Build failed (exit 1)

Never seen on 0.63, any fix for this?

Baskai959, 2025-07-19:
same error

ramseymcgrath, 2025-07-19:
will patch tomorrow, non-containerized builds should run as expected

ramseymcgrath, 2025-07-19:
resolved by https://github.com/ramseymcgrath/PCILeechFWGenerator/pull/134