VoltCyclone/PCILeechFWGenerator #137: [BUG] Build failed: name 'VFIO_DEVICE_NAME_MAX_LENGTH' is not defined

Bogzx, 2025-07-19:
(.venv) bogdan@bogdan-B550-GAMING-X-V2:~/PCILeechFWGenerator$ sudo python3 pcileech.py tui
TUI failed: name 'VFIO_DEVICE_NAME_MAX_LENGTH' is not defined
(.venv) bogdan@bogdan-B550-GAMING-X-V2:~/PCILeechFWGenerator$ sudo python3 pcileech.py build --bdf 0000:03:00.0 --board 75t
Build failed: name 'VFIO_DEVICE_NAME_MAX_LENGTH' is not defined
(.venv) bogdan@bogdan-B550-GAMING-X-V2:~/PCILeechFWGenerator$ ^C
(.venv) bogdan@bogdan-B550-GAMING-X-V2:~/PCILeechFWGenerator$

OmarTPX, 2025-07-19:
add VFIO_DEVICE_NAME_MAX_LENGTH = 64 in src/cli/vfio_constants.py file after imports

ramseymcgrath, 2025-07-19:
resolved by 
https://github.com/ramseymcgrath/PCILeechFWGenerator/pull/138

closed 2025-07-19 by ramseymcgrath