VoltCyclone/PCILeechFWGenerator #141: 执行sudo -E python3 pcileech.py tui的报错

Ckk-Dr, 2025-07-19:
作者你好，我在执行sudo -E python3 pcileech.py tui出现以下错误 我该如何解决

(.venv) lbwnb@lbwnb-H81:~/桌面/7.19/PCILeechFWGenerator$ sudo -E python3 pcileech.py tui
[sudo] lbwnb 的密码： 
Error: Textual framework not installed.
Please install with: pip install textual rich psutil

Ckk-Dr, 2025-07-19:
[image]

执行 pip install textual rich psutil 之后 再次 执行 sudo -E python3 pcileech.py依然会出现同样的问题

ramseymcgrath, 2025-07-19:
sudo $(which python3) pcileech.py tui

Ckk-Dr, 2025-07-19:
感谢作者回复

[image]

成功进入界面后该如何操作  设备status 显示错误该如何解决

ramseymcgrath, 2025-07-20:
你仍然可以尝试构建，即使它标记为红色，或者使用命令行获取更多详细信息。

closed 2025-07-20 by ramseymcgrath