VoltCyclone/PCILeechFWGenerator #160: /dev/vfio/vfio device not found

L1veScene, 2025-07-23:
Hey Guys.
I still have one problem that still occurs when trying to build.

I still receive the error -> /dev/vfio/vfio device not found:

[image]

After that it continues to proceed, and then before it ends there is a warning that it failed to restore to the original driver, and after that it doesn't finish and there are errors like -> VFIO group device /dev/vfio/1 does not exist and another VFIO error, as one can see in the next screenshot:

[image]
--
After trying to build a second time, the first error does not show, and it fails with another error:

[image]

Maybe someone can help me with that.

Thank you and best regards!

L1veScene, 2025-07-24:
> This error means that VFIO doesn't have the necessary permissions to access the IOMMU group that contains your PCIe device (0000:04:00.0).
> 
> Its a really easy fix just add VFIO group to that /dev/vfio/1

Thank you @k1dua very much for your help.
The problem is, that it should have the permissions and be added to the group.

It throws the error, that the VFIO group device does not exist...

[image]

----

...but when checking with:
**ls -l /dev/vfio**
it shows

[image]

Maybe I have something complety wrong.

Do you maybe have any other suggestions?

Best regards and again thank you for your help!!

k1dua, 2025-07-24:
> Thank you @k1dua very much for your help.
> The problem is, that it should have the permissions and be added to the group.
>
> It throws the error, that the VFIO group device does not exist...
>
> [image] [...]

Just do "sudo chmod 660 /dev/vfio/1"

And it will work

L1veScene, 2025-07-24:
Thank you again @k1dua.

For safety I did it before and it still doesnt work. It can be seen in the screenshot if im not wrong that is it set to his.

[image]

L1veScene, 2025-07-24:
@k1dua 
I am in CET too, I think I will be available at estimated 12:00/13:00 CET.
Just send me a message or friend request on Discord... you can see that on my profile on GitHub. I will write you when I'm available, if that is okay.

Thanks alot!!!

closed 2025-07-27 by ramseymcgrath