VoltCyclone/PCILeechFWGenerator #176: 0.7.8 error

sthgi, 2025-07-25:
vivado -mode batch -source vivado_generate_project.tcl

****** Vivado v2024.2 (64-bit)
  **** SW Build 5239630 on Fri Nov 08 22:34:34 MST 2024
  **** IP Build 5239520 on Sun Nov 10 16:12:51 MST 2024
  **** SharedData Build 5239561 on Fri Nov 08 14:39:27 MST 2024
  **** Start of session at: Fri Jul 25 10:46:29 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

source vivado_generate_project.tcl
# set_msg_config -id {Common 17-55} -suppress
# set_msg_config -id {Vivado 12-1387} -suppress
# set project_name "pcileech_firmware"
# set project_dir "./vivado_project"
# set fpga_part "xc7a75tfgg484-2"
# set board_name "pcileech_75t484_x1"
# set src_dir "src"
# set ip_dir "ip"
# file mkdir $project_dir
# create_project $project_name $project_dir -part $fpga_part -force
# set_property target_language SystemVerilog [current_project]
ERROR: [Common 17-162] Invalid option value specified for 'value'.
INFO: [Common 17-206] Exiting Vivado at Fri Jul 25 10:46:38 2025...

ramseymcgrath, 2025-07-26:
resolved in https://github.com/ramseymcgrath/PCILeechFWGenerator/releases/tag/v0.7.9

closed 2025-07-26 by ramseymcgrath

OmarTPX, 2025-07-26:
> resolved in https://github.com/ramseymcgrath/PCILeechFWGenerator/releases/tag/v0.7.9

i'm still getting the same error 
```
****** Vivado v2022.2 (64-bit)
  **** SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022
  **** IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

source vivado_generate_project.tcl
# set_msg_config -id {Common 17-55} -suppress
# set_msg_config -id {Vivado 12-1387} -suppress
# set project_name "pcileech_firmware"
# set project_dir "./vivado_project"
# set fpga_part "xc7a75tfgg484-2"
# set board_name "pcileech_75t484_x1"
# set src_dir "src"
# set ip_dir "ip"
# file mkdir $project_dir
# create_project $project_name $project_dir -part $fpga_part -force
# set_property target_language SystemVerilog [current_project]
ERROR: [Common 17-162] Invalid option value specified for 'value'.
INFO: [Common 17-206] Exiting Vivado at Sat Jul 26 10:33:55 2025...
```