VoltCyclone/PCILeechFWGenerator #188: Build failed: No valid MMIO BARs found - all BARs are either size 0 or I/O-port （或许没有修复）

zyycn, 2025-08-01:
我使用了一张PCIE捐赠卡，但是生成失败，我搜寻了issues，此问题以解决，但是我仍获得了相同错误，我不清楚为什么

```bash
(venv) zyy@Zyy-Linux:~/桌面/Code/PCILeechFWGenerator$ sudo $(which python3) pcileech.py build --bdf 0000:05:00.0 --board pcileech_75t484_x1
02:20:40 INFO   02:20:40 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
02:20:40 INFO   02:20:40 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
02:20:40 INFO   02:20:40 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
02:20:40 INFO   02:20:40 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
02:20:40 INFO   02:20:40 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
02:20:40 INFO   02:20:40 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
02:20:40 INFO   02:20:40 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
02:20:40 INFO   02:20:40 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
02:20:40 INFO   02:20:40 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
02:20:41 INFO   02:20:41 │  INFO  │ Discovered 8 boards
02:20:41 INFO   02:20:41 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
02:20:41 INFO   02:20:41 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
02:20:41 INFO   02:20:41 │  INFO  │ [BUILD] Starting build for device 0000:05:00.0 on board pcileech_75t484_x1
02:20:41 INFO   02:20:41 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 7d09646c0f2c6acc550fb61bd8408498f6df76b6b14167c1447418f80a3554b1
--> 7d09646c0f2c
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
--> Using cache 8c56fc38e8129e28aee79e8ce9290cea538b54a96c5a3e89f29b05bc1fbf92c8
--> 8c56fc38e812
[1/2] STEP 4/7: WORKDIR /src
--> Using cache 70f9b0c4580f6d9d465a43788af0aa7b7af8995e7b09a480badd4c337b276a09
--> 70f9b0c4580f
[1/2] STEP 5/7: COPY vfio_helper.c patch_vfio_constants.py build_vfio_constants.sh ./
--> Using cache 4ff02b5f8ad07dd3964072c9ea5dcb4b6e8e67322486bf9bcea10ab121849ee9
--> 4ff02b5f8ad0
[1/2] STEP 6/7: COPY src/cli/vfio_constants.py ./src/cli/
--> Using cache 7f3bd23367e4af9690563af0ade3195cfbc8a691f532a6ab716c12b17e3d53ce
--> 7f3bd23367e4
[1/2] STEP 7/7: RUN mkdir -p src/cli &&     chmod +x build_vfio_constants.sh &&     (./build_vfio_constants.sh && cp src/cli/vfio_constants.py vfio_constants_patched.py) ||     (echo "⚠ VFIO constants build failed, using original" && cp src/cli/vfio_constants.py vfio_constants_patched.py) &&     echo "Content of patched file:" && head -20 vfio_constants_patched.py | grep -A 10 "Ioctl numbers" || echo "No ioctl numbers section found"
--> Using cache 6d6d20a240dbe3e0e07c2f3fc8dda889cee096a83ab705bcc1f589a87a625f8a
--> 6d6d20a240db
[2/2] STEP 1/20: FROM ubuntu:22.04 AS runtime
[2/2] STEP 2/20: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PCILEECH_PRODUCTION_MODE=true     PCILEECH_ALLOW_MOCK_DATA=false
--> Using cache ef82187f9a0779c4367c6b78323d33f51e3365371fba325568c5fb85ae02687a
--> ef82187f9a07
[2/2] STEP 3/20: RUN apt-get update &&     apt-get install -y --no-install-recommends         python3 python3-pip pciutils bsdextrautils kmod ca-certificates git sudo     && apt-get clean && rm -rf /var/lib/apt/lists/*
--> Using cache a4d9cdfe920bb92f6f433832b4fe30ec35dae97077f1d16b9989c3dab35b7fba
--> a4d9cdfe920b
[2/2] STEP 4/20: RUN useradd -m -r appuser &&     echo "appuser ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers &&     echo "Defaults !requiretty" >> /etc/sudoers
--> Using cache c158d263c0c801a671fb4eecbec5ccee81a4bfef0305ad01cbbeb0f10dbcffc2
--> c158d263c0c8
[2/2] STEP 5/20: WORKDIR /app
--> Using cache c62dd2a313a01611c834c6a5371d3d7c1d80d268318d32279839104fd9ccaea7
--> c62dd2a313a0
[2/2] STEP 6/20: COPY requirements.txt requirements-tui.txt ./
--> Using cache 08b754b4d93178fd11863c756a59568360fc46bc29057c177859692712fd42ee
--> 08b754b4d931
[2/2] STEP 7/20: RUN pip3 install --no-cache-dir -r requirements.txt -r requirements-tui.txt
--> Using cache 4e783c43d27aac2fdc91e1a12d836f40872da2d4d55c53e65479cb995647e2cc
--> 4e783c43d27a
[2/2] STEP 8/20: COPY src ./src
--> 3052b37adcfc
[2/2] STEP 9/20: COPY boards ./boards
--> 066dba912c52
[2/2] STEP 10/20: COPY configs ./configs
--> 8f958b4b927e
[2/2] STEP 11/20: COPY pcileech.py .
--> 425c6c89e87c
[2/2] STEP 12/20: COPY --from=build /src/vfio_constants_patched.py ./src/cli/vfio_constants.py
--> 1d216486bb33
[2/2] STEP 13/20: RUN find ./src -type d -exec touch {}/__init__.py \; 2>/dev/null || true
--> 6b90d3361f8e
[... 218 lines cut ...]
git version 2.43.0
.git
02:27:35 INFO   02:27:35 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
02:27:35 INFO   02:27:35 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
02:27:35 INFO   02:27:35 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
02:27:35 INFO   02:27:35 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
02:27:35 INFO   02:27:35 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
02:27:35 INFO   02:27:35 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
02:27:35 INFO   02:27:35 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
02:27:35 INFO   02:27:35 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
02:27:35 INFO   02:27:35 │  INFO  │ Discovered 8 boards
02:27:35 INFO   02:27:35 │  INFO  │ [CHECK] Running VFIO diagnostics
02:27:35 INFO   02:27:35 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
02:27:35 INFO   02:27:35 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered
```

zyycn, 2025-08-02:
更新一下，我用AX201网卡成功的运行了脚本，或许是网卡不支持

ramseymcgrath, 2025-08-02:
v0.9.1 should now wake up devices properly

closed 2025-08-02 by ramseymcgrath