VoltCyclone/PCILeechFWGenerator #194: 0.9.3 Error

sthgi, 2025-08-02:
(.venv) root@beelink:~/PCILeechFWGenerator/output# vivado -mode batch -source vivado_build.tcl

****** Vivado v2024.2 (64-bit)
  **** SW Build 5239630 on Fri Nov 08 22:34:34 MST 2024
  **** IP Build 5239520 on Sun Nov 10 16:12:51 MST 2024
  **** SharedData Build 5239561 on Fri Nov 08 14:39:27 MST 2024
  **** Start of session at: Sat Aug  2 13:39:18 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

source vivado_build.tcl
# set_param general.maxThreads 4
# set_param synth.maxThreads 4
# set_param place.maxThreads 4
# set_param route.maxThreads 4
# set_msg_config -id {Common 17-55} -suppress
# set_msg_config -id {Vivado 12-1387} -suppress
# set_msg_config -id {Synth 8-3331} -suppress
# set_msg_config -id {Synth 8-3332} -suppress
# set project_name "pcileech_firmware"
# set project_dir "./vivado_project"
# set fpga_part "xc7a75tfgg484-2"
# set board_name "pcileech_75t484_x1"
# set synthesis_strategy "Vivado Synthesis Defaults"
# set implementation_strategy "Performance_Explore"
# puts "Opening PCILeech project: $project_name"
Opening PCILeech project: pcileech_firmware
# if {[file exists [file join $project_dir "${project_name}.xpr"]]} {
#     open_project [file join $project_dir "${project_name}.xpr"]
# } else {
#     puts "ERROR: Project file not found. Run project generation script first."
#     exit 1
# }
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2024.2/data/ip'.
# if {[get_property PART [current_project]] != $fpga_part} {
#     puts "ERROR: Project FPGA part mismatch. Expected: $fpga_part"
#     exit 1
# }
# puts "Project opened successfully"
Project opened successfully
# puts "FPGA Part: [get_property PART [current_project]]"
FPGA Part: xc7a75tfgg484-2
# puts "Target Language: [get_property TARGET_LANGUAGE [current_project]]"
Target Language: Verilog
# reset_run synth_1
# reset_run impl_1
# puts "Configuring synthesis strategy: $synthesis_strategy"
Configuring synthesis strategy: Vivado Synthesis Defaults
# set_property strategy $synthesis_strategy [get_runs synth_1]
# set_property -name {STEPS.SYNTH_DESIGN.ARGS.FLATTEN_HIERARCHY} -value {rebuilt} -objects [get_runs synth_1]
# set_property -name {STEPS.SYNTH_DESIGN.ARGS.GATED_CLOCK_CONVERSION} -value {off} -objects [get_runs synth_1]
# puts "Configuring implementation strategy: $implementation_strategy"
Configuring implementation strategy: Performance_Explore
# set_property strategy $implementation_strategy [get_runs impl_1]
# set_property -name {STEPS.PLACE_DESIGN.ARGS.MORE OPTIONS} -value {-timing_summary} -objects [get_runs impl_1]
# set_property -name {STEPS.ROUTE_DESIGN.ARGS.MORE OPTIONS} -value {-timing_summary} -objects [get_runs impl_1]
# set_property -name {STEPS.SYNTH_DESIGN.ARGS.MORE OPTIONS} -value {-no_iobuf} -objects [get_runs synth_1]
# puts "Starting synthesis..."
Starting synthesis...
# puts "Strategy: $synthesis_strategy"
Strategy: Vivado Synthesis Defaults
# puts "Jobs: 4"
Jobs: 4
# launch_runs synth_1 -jobs 4
INFO: [IP_Flow 19-6924] IPCACHE: Running cache check for IP inst: pcie_7x_bridge
WARNING: [Vivado 12-7122] Auto Incremental Compile:: No reference checkpoint was found in run synth_1. Auto-incremental flow will not be run, the standard flow will be run instead.
[Sat Aug  2 13:39:26 2025] Launched pcie_7x_bridge_synth_1...
Run output will be captured here: /root/PCILeechFWGenerator/output/vivado_project/pcileech_firmware.runs/pcie_7x_bridge_synth_1/runme.log
[Sat Aug  2 13:39:26 2025] Launched synth_1...
Run output will be captured here: /root/PCILeechFWGenerator/output/vivado_project/pcileech_firmware.runs/synth_1/runme.log
# wait_on_run synth_1
[Sat Aug  2 13:39:26 2025] Waiting for synth_1 to finish...

*** Running vivado
    with args -log pcileech_top.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source pcileech_top.tcl

****** Vivado v2024.2 (64-bit)
  **** SW Build 5239630 on Fri Nov 08 22:34:34 MST 2024
  **** IP Build 5239520 on Sun Nov 10 16:12:51 MST 2024
  **** SharedData Build 5239561 on Fri Nov 08 14:39:27 MST 2024
  **** Start of session at: Sat Aug  2 13:40:03 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

source pcileech_top.tcl -notrace
Command: synth_design -top pcileech_top -part xc7a75tfgg484-2 -no_iobuf
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xc7a75t'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xc7a75t'
INFO: [Device 21-403] Loading part xc7a75tfgg484-2
INFO: [Synth 8-7079] Multithreading enabled for synth_design using a maximum of 4 processes.
INFO: [Synth 8-7078] Launching helper process for spawning children vivado processes
INFO: [Synth 8-7075] Helper process launched with PID 46439
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:02 ; elapsed = 00:00:02 . Memory (MB): peak = 2120.930 ; gain = 420.797 ; free physical = 18821 ; free virtual = 29047
---------------------------------------------------------------------------------
INFO: [Synth 8-6157] synthesizing module 'pcileech_top' [/root/PCILeechFWGenerator/output/src/top_level_wrapper.sv:8]
ERROR: [Synth 8-439] module 'device_config' not found [/root/PCILeechFWGenerator/output/src/top_level_wrapper.sv:53]
ERROR: [Synth 8-6156] failed synthesizing module 'pcileech_top' [/root/PCILeechFWGenerator/output/src/top_level_wrapper.sv:8]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:03 ; elapsed = 00:00:03 . Memory (MB): peak = 2196.898 ; gain = 496.766 ; free physical = 18727 ; free virtual = 28953
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
7 Infos, 0 Warnings, 0 Critical Warnings and 3 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Sat Aug  2 13:40:15 2025...
[Sat Aug  2 13:40:25 2025] synth_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:00:43 ; elapsed = 00:00:59 . Memory (MB): peak = 1524.730 ; gain = 0.000 ; free physical = 20208 ; free virtual = 30435
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run synth_1"
    (file "vivado_build.tcl" line 76)
INFO: [Common 17-206] Exiting Vivado at Sat Aug  2 13:40:25 2025...

Dec-OwO, 2025-08-03:
having same issue

ramseymcgrath, 2025-08-03:
Link your build logs pls I need to see why youre missing a template

Dec-OwO, 2025-08-03:
> Link your build logs pls I need to see why youre missing a template

```
(.venv) xenn@xenn-Z390-DESIGNARE:~/Desktop/FWGEN/PCILeechFWGenerator$ sudo $(which python) pcileech.py build --bdf 0000:05:00.0 --board  pcileech_75t484_x1
18:50:02 INFO   18:50:02 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.48.1
.git
18:50:02 INFO   18:50:02 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
18:50:02 INFO   18:50:02 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
18:50:02 INFO   18:50:02 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
18:50:02 INFO   18:50:02 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
18:50:02 INFO   18:50:02 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
18:50:02 INFO   18:50:02 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
18:50:02 INFO   18:50:02 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
18:50:02 INFO   18:50:02 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
18:50:02 INFO   18:50:02 │  INFO  │ Discovered 8 boards
18:50:02 INFO   18:50:02 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
18:50:02 INFO   18:50:02 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
18:50:02 INFO   18:50:02 │  INFO  │ [BUILD] Starting build for device 0000:05:00.0 on board pcileech_75t484_x1
18:50:02 INFO   18:50:02 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 98a37294a70b47e368c2bf6b8177034e8bfa511dbbc8b7ca7b50950b40045b76
--> 98a37294a70b
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
--> Using cache 212a8496c48f42f52c46a7cba8fb343e9b07e87ff6ceb2330278985787c280ac
--> 212a8496c48f
[1/2] STEP 4/7: WORKDIR /src
--> Using cache a69bc095f7afa8d4db6ccd6ccef06d89a9bfcabffdf93ad9323b5547cbadbb95
--> a69bc095f7af
[1/2] STEP 5/7: COPY vfio_helper.c patch_vfio_constants.py build_vfio_constants.sh ./
--> Using cache aa142676d1fdfa312f870f5f711e0440d48ff8cd6776a0fb80d1332fd2a84c3a
--> aa142676d1fd
[1/2] STEP 6/7: COPY src/cli/vfio_constants.py ./src/cli/
--> Using cache 0d5759b1524aabe6b66ac5c1e0707dfa82a7200133882437e0f035677a23e559
--> 0d5759b1524a
[1/2] STEP 7/7: RUN mkdir -p src/cli &&     chmod +x build_vfio_constants.sh &&     (./build_vfio_constants.sh && cp src/cli/vfio_constants.py vfio_constants_patched.py) ||     (echo "⚠ VFIO constants build failed, using original" && cp src/cli/vfio_constants.py vfio_constants_patched.py) &&     echo "Content of patched file:" && head -20 vfio_constants_patched.py | grep -A 10 "Ioctl numbers" || echo "No ioctl numbers section found"
--> Using cache 3fccc9d41f57b50085313533529f8967ca1407297c4553639edf8b133c41070f
--> 3fccc9d41f57
[2/2] STEP 1/20: FROM ubuntu:22.04 AS runtime
[2/2] STEP 2/20: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PCILEECH_PRODUCTION_MODE=true     PCILEECH_ALLOW_MOCK_DATA=false
--> Using cache 4633e6a988444fcf045d864f52cbd94e182d9ed52b1157d41c3d7ef8c698d952
--> 4633e6a98844
[2/2] STEP 3/20: RUN apt-get update &&     apt-get install -y --no-install-recommends         python3 python3-pip pciutils bsdextrautils kmod ca-certificates git sudo     && apt-get clean && rm -rf /var/lib/apt/lists/*
--> Using cache 5350b3a48999d6801608fc8ad18f298e8a2756693c5664a5323a9b701e71163e
--> 5350b3a48999
[2/2] STEP 4/20: RUN useradd -m -r appuser &&     echo "appuser ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers &&     echo "Defaults !requiretty" >> /etc/sudoers
--> Using cache 1b8f149f41d7bdcdbee527c219176e7c98a769e2d6523153973f257248e1d503
--> 1b8f149f41d7
[2/2] STEP 5/20: WORKDIR /app
--> Using cache 28b8bba877fe0cc2e85ea5ee95c0e89cd24cd7fdc4f53f3677af55f5ab7a7b64
--> 28b8bba877fe
[2/2] STEP 6/20: COPY requirements.txt requirements-tui.txt ./
--> Using cache da176a7011ae13989061d273710b803586d863fcc794003103f84bfe0ba33a61
--> da176a7011ae
[2/2] STEP 7/20: RUN pip3 install --no-cache-dir -r requirements.txt -r requirements-tui.txt
--> Using cache 7cb871eeb40dab73c5db335a70aba8169d20c8d78576d9cdfd8750d0bbfcfc2d
--> 7cb871eeb40d
[2/2] STEP 8/20: COPY src ./src
--> 96daf40efc80
[2/2] STEP 9/20: COPY boards ./boards
--> 5c9cd3461322
[2/2] STEP 10/20: COPY configs ./configs
--> 1ea05de5ecc5
[2/2] STEP 11/20: COPY pcileech.py .
--> 37fbbfd5c539
[2/2] STEP 12/20: COPY --from=build /src/vfio_constants_patched.py ./src/cli/vfio_constants.py
--> 8fa93c6fd821
[2/2] STEP 13/20: RUN find ./src -type d -exec touch {}/__init__.py \; 2>/dev/null || true
--> 8e2ecfcb343b
[... 188 lines cut ...]
    - src/msix_capability_registers.sv
    - src/msix_implementation.sv
[... cut ...]
    - src/pcileech_fifo.sv
    - src/pcileech_tlps128_bar_controller.sv
    - src/top_level_wrapper.sv

  TCL scripts (2):
    - vivado_build.tcl
    - vivado_generate_project.tcl

  JSON files (2):
    - behavior_profile.json
    - device_info.json

  Other files (4):
    - src/msix_pba_init.hex
    - src/msix_table_init.hex
    - systemverilog/pcileech_cfgspace.coe
    - systemverilog/pcileech_cfgspace_writemask.coe

Total: 14 files
18:51:11 INFO   18:51:11 │  INFO  │ [CONT] Build completed in 65.1s

```