VoltCyclone/PCILeechFWGenerator #201: [BUG] ERROR simple build command "Operation not Permitted"

Danielix99, 2025-08-04:
user@fwgen:~/PCILeechFWGenerator-0.9.3$ sudo python3 pcileech.py build --bdf 0000:02:00.0 --board pcileech_75t484_x1
21:33:12 INFO   21:33:12 │  INFO  │ Discovering boards from pcileech-fpga repository...
21:33:12 INFO   21:33:12 │  INFO  │ Cloning https://github.com/ufrisk/pcileech-fpga.git -> /root/.cache/pcileech-fw-generator/repos/pcileech-fpga
21:33:12 INFO   21:33:12 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
21:33:12 INFO   21:33:12 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
21:33:12 INFO   21:33:12 │  INFO  │ [BUILD] Starting build for device 0000:02:00.0 on board pcileech_75t484_x1
21:33:12 INFO   21:33:12 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 99da50d171c83f98131218a3972343ec4e6c14feb58bbddd8a7f13b9afd3a7e0
--> 99da50d171c
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
--> Using cache 5d819067332f8788bb1f62f41a762aa3af4400cce0fda3c2a55d8d539a99470e
--> 5d819067332
[1/2] STEP 4/7: WORKDIR /src
--> Using cache acd515e7874120e9a5f5910e630dce17f7e081a96528f1fc0fce876fd305704b
--> acd515e7874
[1/2] STEP 5/7: COPY vfio_helper.c patch_vfio_constants.py build_vfio_constants.sh ./
--> Using cache abd8c8677ae67b3a451174a66dda5eab54bf020c9357cd33e28ff9f3199fd619
--> abd8c8677ae
[1/2] STEP 6/7: COPY src/cli/vfio_constants.py ./src/cli/
--> Using cache 45e82117b230b0e3691b4a647ee47bc173269b2bee5042f21a4a6facae1857ef
--> 45e82117b23
[1/2] STEP 7/7: RUN mkdir -p src/cli &&     chmod +x build_vfio_constants.sh &&     (./build_vfio_constants.sh && cp src/cli/vfio_constants.py vfio_constants_patched.py) ||     (echo "⚠ VFIO constants build failed, using original" && cp src/cli/vfio_constants.py vfio_constants_patched.py) &&     echo "Content of patched file:" && head -20 vfio_constants_patched.py | grep -A 10 "Ioctl numbers" || echo "No ioctl numbers section found"
--> Using cache efa5612df41fb14574f4fd0e1e9c314cadfe00a52bb97d073d73402f7a3500e3
--> efa5612df41
[2/2] STEP 1/20: FROM ubuntu:22.04 AS runtime
[2/2] STEP 2/20: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PCILEECH_PRODUCTION_MODE=true     PCILEECH_ALLOW_MOCK_DATA=false
--> Using cache 50ad95646812a11772271ca5663597a61e7cd64c62373ab3ae31953911c61c1c
--> 50ad9564681
[2/2] STEP 3/20: RUN apt-get update &&     apt-get install -y --no-install-recommends         python3 python3-pip pciutils bsdextrautils kmod ca-certificates git sudo     && apt-get clean && rm -rf /var/lib/apt/lists/*
--> Using cache 9ec4b55ba37f84ce69ac1944bcda5598d7d483a909faf1bcdfe1995e0960ed83
--> 9ec4b55ba37
[2/2] STEP 4/20: RUN useradd -m -r appuser &&     echo "appuser ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers &&     echo "Defaults !requiretty" >> /etc/sudoers
--> Using cache a0aecdf9268049958869e7a0204425ace0eabff40701bc215efef344adbdb092
--> a0aecdf9268
[2/2] STEP 5/20: WORKDIR /app
--> Using cache 4fe57ec5acee86236bb18bd80821ceb9f18e0d9c8ed801d9ca1f7cdb375da1ea
--> 4fe57ec5ace
[2/2] STEP 6/20: COPY requirements.txt requirements-tui.txt ./
--> Using cache 91a8b923124063bd6532e5fc9b1e201d76a97faa4cf6e74fa88a99db8f8a0089
--> 91a8b923124
[2/2] STEP 7/20: RUN pip3 install --no-cache-dir -r requirements.txt -r requirements-tui.txt
--> Using cache a61ffe74006055f756a017cd1238fd60af9e3ae6c3d222446e845d9aa07564ca
--> a61ffe74006
[2/2] STEP 8/20: COPY src ./src
--> 5c05a38fbed
[2/2] STEP 9/20: COPY boards ./boards
--> 0042152db04
[2/2] STEP 10/20: COPY configs ./configs
--> 976395b3389
[2/2] STEP 11/20: COPY pcileech.py .
--> 535a9eb0c42
[2/2] STEP 12/20: COPY --from=build /src/vfio_constants_patched.py ./src/cli/vfio_constants.py
--> 4fa6b4aec31
[2/2] STEP 13/20: RUN find ./src -type d -exec touch {}/__init__.py \; 2>/dev/null || true
--> b40d59b5808
[2/2] STEP 14/20: COPY entrypoint.sh /usr/local/bin/entrypoint
--> a2014607d7b
[2/2] STEP 15/20: RUN chmod 755 /usr/local/bin/entrypoint
--> 14e037dab29
[2/2] STEP 16/20: ENV PYTHONPATH=/app:/app/src
--> 3033f251026
[2/2] STEP 17/20: RUN mkdir -p /app/output && chown appuser /app/output
--> e905a732c80
[2/2] STEP 18/20: HEALTHCHECK --interval=30s --timeout=10s --start-period=5s --retries=3     CMD python3 -c "import psutil, pydantic, sys; sys.exit(0)" || exit 1
WARN[0003] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format 
--> 713846923d1
[2/2] STEP 19/20: USER appuser
WARN[0003] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format 
--> 4bf929fabc5
[2/2] STEP 20/20: ENTRYPOINT ["entrypoint"]
[2/2] COMMIT pcileech-fw-generator:latest
WARN[0004] HEALTHCHECK is not supported for OCI image format and will be ignored. Must use `docker` format 
--> ce856331628
Successfully tagged localhost/pcileech-fw-generator:latest
ce856331628d5509b517a9c1a0a691567cf15e8272103b850c9784207891e821
21:33:17 INFO   21:33:17 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:02:00.0
21:33:17 INFO   21:33:17 │  INFO  │ [BIND] Current driver for 0000:02:00.0: vfio-pci
21:33:17 INFO   21:33:17 │  INFO  │ [BIND] Device 0000:02:00.0 already bound to vfio-pci
21:33:17 INFO   21:33:17 │  INFO  │ [CONT] Launching build container - board=pcileech_75t484_x1, tag=latest
/usr/lib/python3.10/runpy.py:126: RuntimeWarning: 'src.build' found in sys.modules after import of package 'src', but prior to execution of 'src.build'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
19:33:17 - INFO - Production mode enabled: All fallbacks disabled
19:33:17 - INFO -   19:33:17 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:02:00.0
19:33:17 - INFO -   19:33:17 │  INFO  │ AdvancedSVGenerator initialized with PCILeech primary: True
19:33:17 - INFO -   19:33:17 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
19:33:17 - INFO - No device profile specified, using live device detection
19:33:17 - INFO - ➤ Preloading MSI-X data before VFIO binding
19:33:17 - INFO -   19:33:17 │  INFO  │ MSI-X capability found: 64 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x2000
19:33:17 - WARNING -   19:33:17 │ WARNING│ MSI-X table offset 0x4 is not 8-byte aligned (actual offset: 0x4, aligned: 0x0)
19:33:17 - INFO -   • Found MSI-X capability: 64 vectors
19:33:17 - INFO - ➤ Generating PCILeech firmware …
19:33:17 - INFO -   19:33:17 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:02:00.0
19:33:17 - INFO -   19:33:17 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
19:33:17 - INFO -   19:33:17 │  INFO  │ MSI-X capability found: 64 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x2000
19:33:17 - WARNING -   19:33:17 │ WARNING│ MSI-X table offset 0x4 is not 8-byte aligned (actual offset: 0x4, aligned: 0x0)
19:33:17 - INFO -   19:33:17 │  INFO  │ [MSIX] Preloaded MSI-X capability: 64 vectors, table BIR 4, offset 0x0
19:33:17 - INFO -   19:33:17 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
19:33:17 - INFO -   19:33:17 │  INFO  │ [PROFILER] Monitoring started
19:33:49 - INFO -   19:33:49 │  INFO  │ [MSIX] Captured 31072 register accesses with 2 timing patterns
19:33:49 - INFO -   19:33:49 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:02:00.0
19:33:49 - INFO -   19:33:49 │  INFO  │ [BIND] Current driver for 0000:02:00.0: vfio-pci
19:33:49 - INFO -   19:33:49 │  INFO  │ [BIND] Device 0000:02:00.0 already bound to vfio-pci
19:33:49 - ERROR -   19:33:49 │ ERROR  │ [VFIO] Failed to link group 1 to container: [Errno 1] Operation not permitted
19:33:49 - ERROR -   19:33:49 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Failed to open VFIO device FD for 0000:02:00.0: Failed to link group 1 to container: [Errno 1] Operation not permitted
19:33:49 - ERROR - Build failed: Failed to link group 1 to container: [Errno 1] Operation not permitted
19:33:49 - ERROR - Unexpected error: Failed to link group 1 to container: [Errno 1] Operation not permitted
21:33:50 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/1               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/user/Scrivania/PCILeechFWGenerator-0.9.3/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:02:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

Kojinko, 2025-08-04:
03:16:26 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/20               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/tanawat/PCILeechFWGenerator-0.9.4/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:05:00.2 --board pcileech_35t325_x1' returned non-zero exit status 1.

having the same one everything is greencheck and ready to go

Danielix99, 2025-08-05:
Idk why but I noticed that fw generation does not work if the device is not alone in iommu group, we can mark this as solved

closed 2025-08-05 by Danielix99