VoltCyclone/PCILeechFWGenerator #203: build were successfully generated but I have no idea what to do next.

Kojinko, 2025-08-05:
have 14 files complete build without any errors but lost on how to turn it into .bin 

already tried using vivado with some other pcileech open sources and also itself files I just got but always fail to compiled into .bin

Kojinko, 2025-08-05:
[image]

[image]

[image]

files are ready.

zyycn, 2025-08-05:
use vivado

```bash
// genrate project
vivado -mode batch -source vivado_generate_project.tcl

// build project to .bin
vivado -mode batch -source vivado_build.tcl
```

zyycn, 2025-08-05:
> use vivado
> 
> // genrate project
> vivado -mode batch -source vivado_generate_project.tcl
> 
> // build project to .bin
> vivado -mode batch -source vivado_build.tcl

but build stage maybe error  #199

Koxrusty, 2025-08-05:
source vivado_build.tcl
# set_param general.maxThreads 4
# set_param synth.maxThreads 4
# set_param place.maxThreads 4
# set_param route.maxThreads 4
# set_msg_config -id {Common 17-55} -suppress
# set_msg_config -id {Vivado 12-1387} -suppress
# set_msg_config -id {Synth 8-3331} -suppress
# set_msg_config -id {Synth 8-3332} -suppress
# set project_name "pcileech_firmware"
# set project_dir "./vivado_project"
# set fpga_part "xc7a35tcsg324-2"
# set board_name "pcileech_35t325_x1"
# set synthesis_strategy "Vivado Synthesis Defaults"
# set implementation_strategy "Performance_Explore"
# puts "Opening PCILeech project: $project_name"
Opening PCILeech project: pcileech_firmware
# if {[file exists [file join $project_dir "${project_name}.xpr"]]} {
#     open_project [file join $project_dir "${project_name}.xpr"]
# } else {
#     puts "ERROR: Project file not found. Run project generation script first."
#     exit 1
# }
INFO: [filemgmt 56-3] Default IP Output Path : Could not find the directory '/home/tanawat/Documents/PCILeechFWGenerator-0.9.5/output/vivado_project/pcileech_firmware.gen/sources_1'.
Scanning sources...
Finished scanning sources
WARNING: [Board 49-26] cannot add Board Part xilinx.com:emb-plus-vpr-4616:part0:1.0 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/emb-plus-vpr-4616/1.0/board.xml as part xcve2302-sfva784-2mp-e-s specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:emb-plus-vpr-4616:part0:1.1 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/emb-plus-vpr-4616/1.1/board.xml as part xcve2302-sfva784-2mp-e-s specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k24c:part0:1.0 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k24c/1.0/board.xml as part xck24-ubva530-2lv-c specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k24c:part0:1.1 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k24c/1.1/board.xml as part xck24-ubva530-2lv-c specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k24i:part0:1.0 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k24i/1.0/board.xml as part xck24-ubva530-2lvi-i specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26c:part0:1.2 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26c/1.2/board.xml as part xck26-sfvc784-2lv-c specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26c:part0:1.3 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26c/1.3/board.xml as part xck26-sfvc784-2lv-c specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26c:part0:1.4 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26c/1.4/board.xml as part xck26-sfvc784-2lv-c specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26i:part0:1.2 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26i/1.2/board.xml as part xck26-sfvc784-2lvi-i specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26i:part0:1.3 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26i/1.3/board.xml as part xck26-sfvc784-2lvi-i specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:k26i:part0:1.4 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/k26i/1.4/board.xml as part xck26-sfvc784-2lvi-i specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:kcu105:part0:1.6 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/kcu105/1.6/board.xml as part xcku040-ffva1156-2-e specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:kcu105:part0:1.7 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/kcu105/1.7/board.xml as part xcku040-ffva1156-2-e specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:kcu116:part0:1.4 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/kcu116/1.4/board.xml as part xcku5p-ffvb676-2-e specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:kcu116:part0:1.5 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/kcu116/1.5/board.xml as part xcku5p-ffvb676-2-e specified in board_part file is either invalid or not available
WARNING: [Board 49-26] cannot add Board Part xilinx.com:kcu1500:part0:1.2 available at /tools/Xilinx/2025.1/data/xhub/boards/XilinxBoardStore/boards/Xilinx/kcu1500/1.2/board.xml as part xcku115-flvb2104-2-e specified in board_part file is either invalid or not available
[... cut ...]
INFO: [Common 17-206] Exiting Vivado at Tue Aug  5 11:15:36 2025...
[Tue Aug  5 11:15:56 2025] synth_1 finished
ERROR: [Vivado 12-13638] Failed runs(s) : 'synth_1'
wait_on_runs: Time (s): cpu = 00:00:14 ; elapsed = 00:00:29 . Memory (MB): peak = 1487.930 ; gain = 0.000 ; free physical = 10920 ; free virtual = 23549
ERROR: [Common 17-39] 'wait_on_runs' failed due to earlier errors.

    while executing
"wait_on_run synth_1"
    (file "vivado_build.tcl" line 76)
INFO: [Common 17-206] Exiting Vivado at Tue Aug  5 11:15:56 2025...

t4bby, 2025-08-16:
To be honest, this project on its current state cannot not generate a buildable vivado project, with those outdated .sv files and missing files, and seems like the author is like vibe coding this project.

It will generate some useful tables that you can use but that's all. The rest is up to you implementing by implementing it to original pcileech from scratch. I have managed to build the project (bin file) with lots of adjustment but it doesnt not work since there is a lot of protocol missing.

I really suggest that you just start learning how to create firmware using guides available.