VoltCyclone/PCILeechFWGenerator #213: 0.9.8 Errors

sthgi, 2025-08-07:
=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Device : 0000:03:00.0
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=on
✔ ACS bypass: Device isolated in IOMMU group 32
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered
✔ Device: 0000:03:00.0 (0x10ec:0x8125) present
✔ IOMMU group: Group 32
✔ Driver: Already bound to vfio-pci
✔ /dev/vfio node: /dev/vfio/32

(.venv) root@beelink:~/PCILeechFWGenerator# sudo ~/.venv/bin/python3 ./pcileech.py build --bdf 0000:03:00.0 --board pcileech_75t484_x1 --vivado-path /tools/Xilinx/Vivado/2024.2 --vivado-jobs 8 --vivado-timeout 7200
22:55:28 INFO   22:55:28 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
22:55:28 INFO   22:55:28 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
22:55:28 INFO   22:55:28 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
22:55:28 INFO   22:55:28 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
22:55:28 INFO   22:55:28 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
22:55:28 INFO   22:55:28 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
22:55:28 INFO   22:55:28 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
22:55:28 INFO   22:55:28 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
22:55:28 INFO   22:55:28 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
22:55:28 INFO   22:55:28 │  INFO  │ Discovered 8 boards
22:55:28 INFO   22:55:28 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
22:55:28 WARNING   22:55:28 │ WARNING│ [VFIO] VFIO constants rebuild failed: Traceback (most recent call last):
  File "/root/PCILeechFWGenerator/patch_vfio_constants.py", line 29, in <module>
    from src.error_utils import log_error_with_root_cause
  File "/root/PCILeechFWGenerator/src/__init__.py", line 14, in <module>
    from .device_clone import (  # Board configuration; Device configuration; Config space management; Behavior profiling; PCILeech generator
  File "/root/PCILeechFWGenerator/src/device_clone/__init__.py", line 20, in <module>
    from .behavior_profiler import (BehaviorProfile, BehaviorProfiler,
  File "/root/PCILeechFWGenerator/src/device_clone/behavior_profiler.py", line 37, in <module>
    from .manufacturing_variance import DeviceClass, ManufacturingVarianceSimulator
  File "/root/PCILeechFWGenerator/src/device_clone/manufacturing_variance.py", line 22, in <module>
    from typing_extensions import TypedDict
ModuleNotFoundError: No module named 'typing_extensions'

22:55:28 WARNING   22:55:28 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
22:55:28 INFO   22:55:28 │  INFO  │ [BUILD] Starting build for device 0000:03:00.0 on board pcileech_75t484_x1
22:55:28 WARNING   22:55:28 │ WARNING│
============================================================
📦 A new version of PCILeech Firmware Generator is available!
   Current version: 0.5.8
   Latest version:  0.9.8

   Update with one of these commands:
   • pip install --upgrade pcileech-fw-generator
   • git pull (if installed from source)

   Release notes: https://github.com/ramseymcgrath/PCILeechFWGenerator/releases
============================================================

22:55:28 INFO   22:55:28 │  INFO  │ Building container image pcileech-fw-generator:latest

[INFO] Checking VFIO availability...
[ERROR] /dev/vfio/vfio device not found
[ERROR] Make sure VFIO is loaded and container has --device=/dev/vfio/vfio
[INFO] Building VFIO helper and patching constants...
[INFO] Running Python patcher...
Traceback (most recent call last):
  File "/src/patch_vfio_constants.py", line 29, in <module>
    from src.error_utils import log_error_with_root_cause
ModuleNotFoundError: No module named 'src.error_utils'
[ERROR] VFIO constants patching failed
[ERROR] Failed to build and patch VFIO constants
⚠ VFIO constants build failed, using original
Content of patched file:
No ioctl numbers section found

03:56:14 - INFO -   03:56:14 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
03:56:14 - ERROR -   03:56:14 │ ERROR  │ [VFIO] Failed to link group 32 to container: [Errno 25] Inappropriate ioctl for device
03:56:14 - ERROR -   03:56:14 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Failed to open VFIO device FD for 0000:03:00.0: Failed to link group 32 to container: [Errno 25] Inappropriate ioctl for device
03:56:14 - ERROR - Build failed: Failed to link group 32 to container: [Errno 25] Inappropriate ioctl for device
03:56:14 - ERROR - Unexpected error: Failed to link group 32 to container: [Errno 25] Inappropriate ioctl for device
22:56:15 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

ramseymcgrath, 2025-08-07:
try https://github.com/ramseymcgrath/PCILeechFWGenerator/releases/tag/v0.9.9

ramseymcgrath, 2025-08-08:
dupe of https://github.com/ramseymcgrath/PCILeechFWGenerator/issues/215

closed 2025-08-08 by ramseymcgrath