VoltCyclone/PCILeechFWGenerator #228: 0.9.13 ERROR PCILeech SystemVerilog generation failed: Failed to render template 'sv/pcileech_tlps128_bar_controller.sv.j2': Device signature is required

zyycn, 2025-08-10:
```bash
(venv) zyy@Zyy-Linux:~/桌面/Code/PCILeechFWGenerator$ sudo $(which python3) pcileech.py build --bdf 0000:03:00.0 --board pcileech_75t484_x1
17:55:17 INFO   17:55:17 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
17:55:17 INFO   17:55:17 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
17:55:17 INFO   17:55:17 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
17:55:17 INFO   17:55:17 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
17:55:17 INFO   17:55:17 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
17:55:17 INFO   17:55:17 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
17:55:17 INFO   17:55:17 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
17:55:17 INFO   17:55:17 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
17:55:17 INFO   17:55:17 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
17:55:17 INFO   17:55:17 │  INFO  │ Discovered 8 boards
17:55:17 INFO   17:55:17 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
17:55:18 INFO   17:55:18 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
17:55:18 INFO   17:55:18 │  INFO  │ [BUILD] Starting build for device 0000:03:00.0 on board pcileech_75t484_x1
17:55:18 INFO   17:55:18 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 846dd348827d2ea20a63e6a108981e9e4477d4693fc3972d37b616b4b189cf6c
--> 846dd348827d
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
--> Using cache 8ca1ade63a099edbc785bd5e16965b279130f00a2ff5957e93a8875074ffd29c
--> 8ca1ade63a09
[1/2] STEP 4/7: WORKDIR /src
--> Using cache d96e826d9a3c9a4559e2bfec73095dc625c1723938916f34251716137091a63c
--> d96e826d9a3c
[1/2] STEP 5/7: COPY vfio_helper.c patch_vfio_constants.py build_vfio_constants.sh ./
--> Using cache 9b334c634b071534841cfd8334b5bebd51a702f1e6ee4c5f71c5eecfcaaaea5f
--> 9b334c634b07
[1/2] STEP 6/7: COPY src/cli/vfio_constants.py ./src/cli/
--> Using cache 16a0887718270cedd4ca01da11e7812c035d5e498b82095541a2b8ad3fc50fca
--> 16a088771827
[1/2] STEP 7/7: RUN mkdir -p src/cli &&     chmod +x build_vfio_constants.sh &&     (./build_vfio_constants.sh && cp src/cli/vfio_constants.py vfio_constants_patched.py) ||     (echo "⚠ VFIO constants build failed, using original" && cp src/cli/vfio_constants.py vfio_constants_patched.py) &&     echo "Content of patched file:" && head -20 vfio_constants_patched.py | grep -A 10 "Ioctl numbers" || echo "No ioctl numbers section found"
--> Using cache fb1ba379503fcb661cfb85d0844b369e15dd5b3336f56dbf3003539c3c247f32
--> fb1ba379503f
[2/2] STEP 1/20: FROM ubuntu:22.04 AS runtime
[2/2] STEP 2/20: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PCILEECH_PRODUCTION_MODE=true     PCILEECH_ALLOW_MOCK_DATA=false
--> Using cache ef82187f9a0779c4367c6b78323d33f51e3365371fba325568c5fb85ae02687a
--> ef82187f9a07
[2/2] STEP 3/20: RUN apt-get update &&     apt-get install -y --no-install-recommends         python3 python3-pip pciutils bsdextrautils kmod ca-certificates git sudo     && apt-get clean && rm -rf /var/lib/apt/lists/*
--> Using cache a4d9cdfe920bb92f6f433832b4fe30ec35dae97077f1d16b9989c3dab35b7fba
--> a4d9cdfe920b
[2/2] STEP 4/20: RUN useradd -m -r appuser &&     echo "appuser ALL=(ALL) NOPASSWD: ALL" >> /etc/sudoers &&     echo "Defaults !requiretty" >> /etc/sudoers
--> Using cache c158d263c0c801a671fb4eecbec5ccee81a4bfef0305ad01cbbeb0f10dbcffc2
--> c158d263c0c8
[2/2] STEP 5/20: WORKDIR /app
--> Using cache c62dd2a313a01611c834c6a5371d3d7c1d80d268318d32279839104fd9ccaea7
--> c62dd2a313a0
[2/2] STEP 6/20: COPY requirements.txt requirements-tui.txt ./
--> Using cache 71731b04f3ac429684e51d7e82a32c6ec06d32450ed5b4ccedbe3d9127df9afe
--> 71731b04f3ac
[2/2] STEP 7/20: RUN pip3 install --no-cache-dir -r requirements.txt -r requirements-tui.txt
--> Using cache adf51a78aeab2a984a4e4e43ef413f3403d71353405ba6fb7cbcef99d7d86f58
--> adf51a78aeab
[2/2] STEP 8/20: COPY src ./src
--> 1c28c1e4378f
[2/2] STEP 9/20: COPY boards ./boards
--> 3f5686934132
[2/2] STEP 10/20: COPY configs ./configs
--> 9c7913377507
[2/2] STEP 11/20: COPY pcileech.py .
--> faffb9faf59d
[2/2] STEP 12/20: COPY --from=build /src/vfio_constants_patched.py ./src/cli/vfio_constants.py
--> 09d9b1adfe58
[2/2] STEP 13/20: RUN find ./src -type d -exec touch {}/__init__.py \; 2>/dev/null || true
--> 56116094a4e3
[2/2] STEP 14/20: COPY entrypoint.sh /usr/local/bin/entrypoint
--> 3faf491cea12
[... 125 lines cut ...]
09:55:53 - INFO -   09:55:53 │  INFO  │ [BARA] Primary BAR selected: index=0, size=16384 bytes (0.02 MB), type=memory
09:55:53 - INFO -   09:55:53 │  INFO  │ [BARA] Adjusting BAR configuration for device 0000:03:00.0 based on behavior profile
09:55:53 - INFO -   09:55:53 │  INFO  │ [BARA] Detected slow device 0000:03:00.0 with avg_interval: 8000.995613216444
09:55:53 - INFO -   09:55:53 │  INFO  │ [CONF] Building PCILeech configuration for device 0000:03:00.0
09:55:53 - INFO -   09:55:53 │  INFO  │ [CONF] Using base offset 0x0130 for device 0000:03:00.0
09:55:53 - INFO -   09:55:53 │  INFO  │ [CONF] Using max_payload_size=256 bytes, cfg_force_mps=1 for device 0000:03:00.0
09:55:53 - INFO -   09:55:53 │  INFO  │ [OVERLAY] Building overlay RAM configuration for device 0000:03:00.0
09:55:53 - WARNING -   09:55:53 │ WARNING│ [OVERLAY] No configuration space dword map available for overlay generation
09:55:53 - INFO -   09:55:53 │  INFO  │ [SIG] Generated device signature: ac6a9ea6
09:55:53 - INFO -   09:55:53 │  INFO  │ [ACTDEV] Building active device configuration for 0000:03:00.0 with msix interrupts
09:55:53 - INFO -   09:55:53 │  INFO  │ [ACTDEV] Active device config: enabled=False, mode=msix, timer=100000
09:55:53 - INFO -   09:55:53 │  INFO  │ [VLDN] Template context validation completed successfully
09:55:53 - INFO -   09:55:53 │  INFO  │ PCILeech template context built successfully with signature 32'hAC6A9EA6
09:55:53 - INFO -   09:55:53 │  INFO  │ Template context built successfully with 14 top-level keys
09:55:53 - INFO -   09:55:53 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:03:00.0
09:55:53 - INFO -   09:55:53 │  INFO  │ Generating SystemVerilog modules
09:55:53 - INFO -   09:55:53 │  INFO  │ Using PCILeech as primary SystemVerilog generation path
09:55:53 - INFO -   09:55:53 │  INFO  │ Generating PCILeech SystemVerilog modules
09:55:53 - ERROR -   09:55:53 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/pcileech_tlps128_bar_controller.sv.j2': Device signature is required - no fallback values allowed to prevent generic firmware
09:55:53 - ERROR -   09:55:53 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/pcileech_tlps128_bar_controller.sv.j2': Device signature is required - no fallback values allowed to prevent generic firmware
09:55:53 - ERROR -   09:55:53 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/pcileech_tlps128_bar_controller.sv.j2': Device signature is required - no fallback values allowed to prevent generic firmware
09:55:53 - ERROR - Build failed: Device signature is required - no fallback values allowed to prevent generic firmware
09:55:53 - ERROR - Unexpected error: Device signature is required - no fallback values allowed to prevent generic firmware
17:55:53 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/14               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/zyy/桌面/Code/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.
‵‵‵

zyycn, 2025-08-10:
```bash
(venv) zyy@Zyy-Linux:~/桌面/Code/PCILeechFWGenerator$ sudo lspci -vvv -s 0000:03:00.0
03:00.0 Network controller: Intel Corporation Wi-Fi 6E(802.11ax) AX210/AX1675* 2x2 [Typhoon Peak] (rev 1a)
	Subsystem: Intel Corporation Wi-Fi 6 AX210 160MHz
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 19
	IOMMU group: 14
	Region 0: Memory at 53200000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [c8] Power Management version 3
		Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [d0] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [40] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <512ns, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L1, Exit Latency L1 <8us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range B, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Via WAKE#, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 16ms to 55ms, TimeoutDis- LTR+ 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-5GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete- EqualizationPhase1-
			 EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
	Capabilities: [80] MSI-X: Enable- Count=16 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [14c v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [154 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=30us PortTPowerOnTime=18us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=59392ns
		L1SubCtl2: T_PwrOn=18us
	Kernel driver in use: vfio-pci
	Kernel modules: iwlwifi
```

Danielix99, 2025-08-10:
I tried hardcoding the device signature and got rid of the issue but a new issue come up

12:01:18 - INFO -   12:01:18 │  INFO  │ Template context built successfully with 14 top-level keys
12:01:18 - INFO -   12:01:18 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:06:00.0
12:01:18 - INFO -   12:01:18 │  INFO  │ Generating SystemVerilog modules
12:01:18 - INFO -   12:01:18 │  INFO  │ Using PCILeech as primary SystemVerilog generation path
12:01:18 - INFO -   12:01:18 │  INFO  │ Generating PCILeech SystemVerilog modules
12:01:18 - ERROR -   12:01:18 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'generation_metadata' is undefined
12:01:18 - ERROR -   12:01:18 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'generation_metadata' is undefined
12:01:18 - ERROR -   12:01:18 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'generation_metadata' is undefined
12:01:18 - ERROR - Build failed: 'generation_metadata' is undefined
12:01:18 - ERROR - Unexpected error: 'generation_metadata' is undefined