VoltCyclone/PCILeechFWGenerator #235: 0.10.3

sthgi, 2025-08-11:
08:08:29 INFO   08:08:29 │  INFO  │ [BUILD] Starting build for device 0000:03:00.0 on board pcileech_75t484_x1
08:08:29 WARNING   08:08:29 │ WARNING│
============================================================
📦 A new version of PCILeech Firmware Generator is available!
   Current version: 0.9.13
   Latest version:  0.10.1
   Build date:      2025-06-05T01:44:07.732192
   Commit hash:     39c13da

   Update with one of these commands:
   • pip install --upgrade pcileech-fw-generator
   • git pull (if installed from source)

   Release notes: https://github.com/ramseymcgrath/PCILeechFWGenerator/releases
============================================================

08:08:29 INFO   08:08:29 │  INFO  │ Building container image pcileech-fw-generator:latest

[INFO] Checking VFIO availability...
[ERROR] /dev/vfio/vfio device not found
[ERROR] Make sure VFIO is loaded and container has --device=/dev/vfio/vfio
[INFO] Building VFIO helper and patching constants...
[INFO] Running Python patcher...
Traceback (most recent call last):
  File "/src/patch_vfio_constants.py", line 29, in <module>
    from src.error_utils import log_error_with_root_cause
ModuleNotFoundError: No module named 'src.error_utils'
[ERROR] VFIO constants patching failed
[ERROR] Failed to build and patch VFIO constants
⚠ VFIO constants build failed, using original
Content of patched file:
No ioctl numbers section found

13:09:36 - INFO -   13:09:36 │  INFO  │ Generating PCILeech SystemVerilog modules
13:09:36 - ERROR -   13:09:36 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'pcileech_config' is undefined
13:09:36 - ERROR -   13:09:36 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'pcileech_config' is undefined
13:09:36 - ERROR -   13:09:36 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/pcileech_cfgspace.coe.j2': 'pcileech_config' is undefined
13:09:36 - ERROR - Build failed: 'pcileech_config' is undefined
13:09:36 - ERROR - Unexpected error: 'pcileech_config' is undefined
08:09:36 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

mantis-rng, 2025-08-11:
On your branch i still get : 

```
23:38:41 ERROR   23:38:41 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:38:41 ERROR   23:38:41 │ ERROR  │ Failed to render SystemVerilog template: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:38:41 ERROR   23:38:41 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:38:41 ERROR   23:38:41 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:38:41 ERROR Build failed: 'power_management' is undefined
23:38:41 ERROR Unexpected error: 'power_management' is undefined
23:38:41 ERROR   23:38:41 │ ERROR  │ [LOCAL] Local build failed after 32.2s: Local build failed with exit code 1
23:38:41 ERROR Build failed: Local build failed with exit code 1
```

Danielix99, 2025-08-11:
same

> On your branch i still get :
> 
> ```
> 23:38:41 ERROR   23:38:41 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
> 23:38:41 ERROR   23:38:41 │ ERROR  │ Failed to render SystemVerilog template: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
> 23:38:41 ERROR   23:38:41 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
> 23:38:41 ERROR   23:38:41 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
> 23:38:41 ERROR Build failed: 'power_management' is undefined
> 23:38:41 ERROR Unexpected error: 'power_management' is undefined
> 23:38:41 ERROR   23:38:41 │ ERROR  │ [LOCAL] Local build failed after 32.2s: Local build failed with exit code 1
> 23:38:41 ERROR Build failed: Local build failed with exit code 1
> ```
> 
> And
> 
> ```
> 23:43:16 ERROR   23:43:16 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/advanced_controller.sv.j2': 'error_handling' is undefined
> 23:43:16 ERROR   23:43:16 │ ERROR  │ Failed to render SystemVerilog template: Failed to render template 'sv/advanced_controller.sv.j2': 'error_handling' is undefined
> 23:43:16 ERROR   23:43:16 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'error_handling' is undefined
> 23:43:16 ERROR   23:43:16 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'error_handling' is undefined
> 23:43:16 ERROR Build failed: 'error_handling' is undefined
> 23:43:16 ERROR Unexpected error: 'error_handling' is undefined
> 23:43:16 ERROR   23:43:16 │ ERROR  │ [LOCAL] Local build failed after 31.1s: Local build failed with exit code 1
> 23:43:16 ERROR Build failed: Local build failed with exit code 1
> ```
> 
> Then on vivado build :
> 
> ```
> ---------------------------------------------------------------------------------
> Starting Synthesize : Time (s): cpu = 00:00:03 ; elapsed = 00:00:04 . Memory (MB): peak = 1945.684 ; gain = 452.730 ; free physical = 56606 ; free virtual = 67773
> ---------------------------------------------------------------------------------
> ERROR: [Synth 8-36] 'rst' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:333]
> ERROR: [Synth 8-36] 'bar_read_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:338]
> ERROR: [Synth 8-36] 'bar_read_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:338]
> ERROR: [Synth 8-36] 'bar_write_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:342]
> ERROR: [Synth 8-36] 'bar_write_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:342]
> ERROR: [Synth 8-36] 'rst' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:360]
> ERROR: [Synth 8-36] 'bar_read_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:367]
> ERROR: [Synth 8-36] 'bar_read_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:367]
> ERROR: [Synth 8-36] 'bar_write_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:370]
> ERROR: [Synth 8-36] 'bar_write_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:370]
> ERROR: [Synth 8-36] 'rst' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:397]
> ERROR: [Synth 8-36] 'bar_read_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:408]
> ERROR: [Synth 8-36] 'bar_write_valid' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:408]
> ERROR: [Synth 8-36] 'bar_read_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:414]
> ERROR: [Synth 8-36] 'bar_write_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:414]
> ERROR: [Synth 8-36] 'bar_read_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:415]
> ERROR: [Synth 8-36] 'bar_write_ready' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:422]
> ERROR: [Synth 8-36] 'rst' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:437]
> ERROR: [Synth 8-36] 'rst' is not declared [/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv:455]
> ERROR: [Synth 8-12188] Failed to read verilog '/root/PCILeechFWGenerator/output/src/pcileech_advanced_controller.sv'
> ```

closed 2025-08-12 by ramseymcgrath