VoltCyclone/PCILeechFWGenerator #237: Cant compile on post refactor fix

Danielix99, 2025-08-11:
Error is

23:03:28 - INFO -   23:03:28 │  INFO  │ Using actual MSI-X table data from hardware (32 entries)
23:03:28 - INFO -   23:03:28 │  INFO  │ Successfully extracted 2 registers from behavior profile
23:03:28 - WARNING -   23:03:28 │ WARNING│ [TEMPLATE] timing_config is missing, providing default values
23:03:28 - WARNING -   23:03:28 │ WARNING│ Template context validation warning: power_config.enable_power_management is missing
23:03:28 - INFO -   23:03:28 │  INFO  │ Template context validation passed (1 warnings)
23:03:28 - WARNING -   23:03:28 │ WARNING│ [TEMPLATE] timing_config is missing, providing default values
23:03:28 - ERROR -   23:03:28 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:03:28 - ERROR -   23:03:28 │ ERROR  │ Failed to render SystemVerilog template: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:03:28 - ERROR -   23:03:28 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:03:28 - ERROR -   23:03:28 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/advanced_controller.sv.j2': 'power_management' is undefined
23:03:28 - ERROR - Build failed: 'power_management' is undefined
23:03:28 - ERROR - Unexpected error: 'power_management' is undefined

ramseymcgrath, 2025-08-12:
Try 1 more time

ramseymcgrath, 2025-08-12:
merged

closed 2025-08-12 by ramseymcgrath