VoltCyclone/PCILeechFWGenerator #242: 0.10.6

sthgi, 2025-08-13:
Successfully tagged localhost/pcileech-fw-generator:latest
99d2871d5bccc07eb571095f1370c4b92148812c8913e45d80a69ba39ba13de8
20:49:02 INFO [BIND] Starting VFIO binding process for device 0000:03:00.0
20:49:02 INFO [BIND] Current driver for 0000:03:00.0: vfio-pci
20:49:02 INFO [BIND] Device 0000:03:00.0 already bound to vfio-pci
20:49:02 INFO [CONT] Launching build container - board=pcileech_75t484_x1, tag=latest
/usr/lib/python3.10/runpy.py:126: RuntimeWarning: 'src.build' found in sys.modules after import of package 'src', but prior to execution of 'src.build'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))

01:49:36 - INFO - Generating PCILeech SystemVerilog modules
01:49:36 - ERROR - [TEMPLATE] Unexpected error rendering template 'sv/pcileech_cfgspace.coe.j2': Template 'sv/pcileech_cfgspace.coe.j2' requires undefined vars: ['msix_pba_bir', 'msix_pba_offset', 'msix_table_bir', 'msix_table_offset']
01:49:36 - ERROR - PCILeech SystemVerilog generation failed: Unexpected error rendering template 'sv/pcileech_cfgspace.coe.j2': Template 'sv/pcileech_cfgspace.coe.j2' requires undefined vars: ['msix_pba_bir', 'msix_pba_offset', 'msix_table_bir', 'msix_table_offset']
01:49:36 - ERROR - [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Unexpected error rendering template 'sv/pcileech_cfgspace.coe.j2': Template 'sv/pcileech_cfgspace.coe.j2' requires undefined vars: ['msix_pba_bir', 'msix_pba_offset', 'msix_table_bir', 'msix_table_offset']
01:49:36 - ERROR - Build failed: Template 'sv/pcileech_cfgspace.coe.j2' requires undefined vars: ['msix_pba_bir', 'msix_pba_offset', 'msix_table_bir', 'msix_table_offset']
01:49:36 - ERROR - Unexpected error: Template 'sv/pcileech_cfgspace.coe.j2' requires undefined vars: ['msix_pba_bir', 'msix_pba_offset', 'msix_table_bir', 'msix_table_offset']
20:49:37 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/18               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

sthgi, 2025-08-13:
sudo ~/.venv/bin/python3 ./pcileech.py build --bdf 0000:03:00.0 --board pcileech_100t484_x1

mantis-rng, 2025-08-13:
Fixed by moving out the variable definition out the if msix_enabled scope

mantis-rng, 2025-08-13:
Then this one : 

```
14:52:15 ERROR [TEMPLATE] Unexpected error rendering template 'sv/advanced_controller.sv.j2': Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
14:52:15 ERROR ERROR TYPE: Template Error
CONTEXT: advanced SystemVerilog generation
DETAILS: Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
SUGGESTION: There's an issue with the template. Check the template syntax and ensure all required variables are provided.
14:52:15 ERROR PCILeech SystemVerilog generation failed: advanced SystemVerilog generation failed: Missing required template variables. Ensure all required context variables are provided. Details: Unexpected error rendering template 'sv/advanced_controller.sv.j2': Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
14:52:15 ERROR [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: advanced SystemVerilog generation failed: Missing required template variables. Ensure all required context variables are provided. Details: Unexpected error rendering template 'sv/advanced_controller.sv.j2': Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
14:52:15 ERROR Build failed: Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
14:52:15 ERROR Unexpected error: Template 'sv/advanced_controller.sv.j2' requires undefined vars: ['board_config']
14:52:15 ERROR [LOCAL] Local build failed after 30.7s: Local build failed with exit code 1
14:52:15 ERROR Build failed: Local build failed with exit code 1
```

ramseymcgrath, 2025-08-13:
fixed in overhaul