VoltCyclone/PCILeechFWGenerator #265: [BUG] Same error from 0.12.2 Inappropriate ioctl for device

StixuYT, 2025-08-15:
## 📋 Bug Summary

**Brief Description:**
Same error from 0.12.2 still happening.

**Severity Level:**

- [ ] **Critical** - System crashes, data loss, or complete failure
- [ ] **High** - Major functionality broken, significant impact
- [ ] **Medium** - Minor functionality issues, workarounds available
- [ ] **Low** - Cosmetic issues, minor inconveniences

## 📦 Environment Details

[systeminfo](https://imgur.com/3nI2EwG)

## 🚨 Problem Description

**Error Messages/Logs:**
```
(dma) daniel@daniel:~/compat2$ sudo /home/daniel/dma/bin/python3 pcileech.py check
12:05:20 INFO   12:05:20 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
12:05:21 INFO   12:05:21 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
12:05:21 INFO   12:05:21 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
12:05:21 INFO   12:05:21 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
12:05:22 INFO   12:05:22 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
12:05:22 INFO   12:05:22 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
12:05:22 INFO   12:05:22 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
12:05:23 INFO   12:05:23 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
12:05:23 INFO   12:05:23 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
12:05:24 INFO   12:05:24 │  INFO  │ Discovered 8 boards
12:05:24 INFO   12:05:24 │  INFO  │ [CHECK] Running VFIO diagnostics
12:05:24 INFO   12:05:24 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
12:05:24 INFO   12:05:24 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered

(dma) daniel@daniel:~/compat2$ sudo /home/daniel/dma/bin/python3 pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1
12:05:55 INFO   12:05:55 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
12:05:55 INFO   12:05:55 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
12:05:55 INFO   12:05:55 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
12:05:55 INFO   12:05:55 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
12:05:55 INFO   12:05:55 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
12:05:55 INFO   12:05:55 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
12:05:55 INFO   12:05:55 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
12:05:55 INFO   12:05:55 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
12:05:55 INFO   12:05:55 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
12:05:55 INFO   12:05:55 │  INFO  │ Discovered 8 boards
12:05:55 INFO   12:05:55 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
12:05:55 WARNING   12:05:55 │ WARNING│ [VFIO] VFIO constants rebuild failed: 
12:05:55 WARNING   12:05:55 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
12:05:55 INFO   12:05:55 │  INFO  │ [BUILD] Starting build for device 0000:08:00.0 on board pcileech_75t484_x1
12:05:56 WARNING   12:05:56 │ WARNING│ [BUILD] Podman not available or cannot connect

============================================================
⚠️  Podman is not available or cannot connect.
============================================================

[... 70 lines cut ...]

**What should happen instead?**
The build should finish without errors

## 🔬 Additional Context

**Frequency:**
- [ ] Always reproducible
- [ ] Intermittent (occurs sometimes)
- [ ] Rare (occurred once or twice)

**Workaround Available:**
- [ ] Yes (please describe below)
- [ ] No

## ✅ Pre-submission Checklist

**I have:**
- [ ] Checked that I'm using the latest version from the main branch
- [ ] Searched for existing similar issues
- [ ] Read the relevant documentation/wiki pages
- [ ] Included all required information above
- [ ] Removed any sensitive/personal information
- [ ] Tested with minimal reproduction case
- [ ] Verified this isn't a configuration issue

ramseymcgrath, 2025-08-15:
Your vfio is miisconfigured, please read the readme steps or use the vfio repair scripts. Most likely its your bios

closed 2025-08-15 by ramseymcgrath

StixuYT, 2025-08-15:
And how do I fix this problem? Because the command python3 pcileech.py check returns the following output:
```
23:58:50 INFO   23:58:50 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
23:58:52 INFO   23:58:52 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
23:58:52 INFO   23:58:52 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
23:58:52 INFO   23:58:52 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
23:58:52 INFO   23:58:52 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
23:58:53 INFO   23:58:53 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
23:58:53 INFO   23:58:53 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
23:58:53 INFO   23:58:53 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
23:58:54 INFO   23:58:54 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
23:58:54 INFO   23:58:54 │  INFO  │ Discovered 8 boards
23:58:54 INFO   23:58:54 │  INFO  │ [CHECK] Running VFIO diagnostics
23:58:54 INFO   23:58:54 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
23:58:54 INFO   23:58:54 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered
```
And this is my IOMMU table(after binding the vfio-pci using a binder)
```
daniel@daniel:~$ cd Downloads/
daniel@daniel:~/Downloads$ cd IOMMU-viewer-master/
daniel@daniel:~/Downloads/IOMMU-viewer-master$ sudo ./iommu_viewer.sh 
[sudo] password for daniel: 
Please be patient. This may take a couple seconds.
Group:  0   0000:00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Root Complex [1022:1480]
Group:  1   0000:00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  2   0000:00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]   Driver: pcieport
Group:  3   0000:00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]   Driver: pcieport
Group:  4   0000:00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  5   0000:00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  6   0000:00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse GPP Bridge [1022:1483]   Driver: pcieport
Group:  7   0000:00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  8   0000:00:05.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  9   0000:00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  10  0000:00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]   Driver: pcieport
Group:  11  0000:00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse PCIe Dummy Host Bridge [1022:1482]
Group:  12  0000:00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B] [1022:1484]   Driver: pcieport
Group:  13  0000:00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 61)
Group:  13  0000:00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
Group:  14  0000:00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 0 [1022:1440]
Group:  14  0000:00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 1 [1022:1441]
Group:  14  0000:00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 2 [1022:1442]
Group:  14  0000:00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 3 [1022:1443]   Driver: k10temp
Group:  14  0000:00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 4 [1022:1444]
Group:  14  0000:00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Matisse/Vermeer Data Fabric: Device 18h; Function 5 [1022:1445]
[... cut ...]
Group:  31  0000:0c:00.1 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse Cryptographic Coprocessor PSPCPP [1022:1486]   Driver: ccp
Group:  32  0000:0c:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Matisse USB 3.0 Host Controller [1022:149c]   Driver: xhci_hcd
Group:  33  0000:0c:00.4 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Starship/Matisse HD Audio Controller [1022:1487]   Driver: snd_hda_intel
daniel@daniel:~/Downloads/IOMMU-viewer-master$ 
```
The ethernet card is in group 25, with ID 0000:08:00.0 and as you can see it's using **Driver: vfio-pci**
What could be wrong?