VoltCyclone/PCILeechFWGenerator #272: 0.13.1 Error

sthgi, 2025-08-16:
04:53:35 - ERROR -   04:53:35 │ ERROR  │ [VALIDATE] Device identification validation failed for 0000:03:00.0: '<=' not supported between instances of 'int' and 'str'
04:53:35 - INFO -   04:53:35 │  INFO  │ Context built successfully: 14e4:1677_72f51f2d2cf3c27c
04:53:35 - ERROR - SECURITY VIOLATION: Template 'pcileech_firmware.j2' context validation failed:
- Required variable 'board_config' is missing or None

Explicit initialization of all template variables is required.
04:53:35 - ERROR -   04:53:35 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SECURITY VIOLATION: Template 'pcileech_firmware.j2' context validation failed:
- Required variable 'board_config' is missing or None

Explicit initialization of all template variables is required.
04:53:35 - ERROR - Build failed: SECURITY VIOLATION: Template 'pcileech_firmware.j2' context validation failed:
- Required variable 'board_config' is missing or None

Explicit initialization of all template variables is required.
04:53:35 - ERROR - Unexpected error: SECURITY VIOLATION: Template 'pcileech_firmware.j2' context validation failed:
- Required variable 'board_config' is missing or None

Explicit initialization of all template variables is required.
23:53:35 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/18               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

sthgi, 2025-08-16:
I am also getting this unless i create a boards directory in PCILeechFWGenerator

[2/2] STEP 9/20: COPY boards ./boards
Error: building at STEP "COPY boards ./boards": checking on sources under "/root/PCILeechFWGenerator": copier: stat: "/boards": no such file or directory
00:09:27 ERROR Build failed: Command '['podman', 'build', '-t', 'pcileech-fw-generator:latest', '-f', 'Containerfile', '.']' returned non-zero exit status 125.

ramseymcgrath, 2025-08-16:
loop should be fixed

ramseymcgrath, 2025-08-16:
relabel as bug if it errors again

closed 2025-08-16 by ramseymcgrath