VoltCyclone/PCILeechFWGenerator #274: [BUG] I really have no idea what I am doing wrong

StixuYT, 2025-08-16:
## 📋 Bug Summary

**Brief Description:**
Inappropriate ioctl for device

**Severity Level:**

- [ ] **Critical** - System crashes, data loss, or complete failure
- [ ] **High** - Major functionality broken, significant impact
- [ ] **Medium** - Minor functionality issues, workarounds available
- [ ] **Low** - Cosmetic issues, minor inconveniences

## 📦 Environment Details

[systeminfo](https://imgur.com/3nI2EwG)

## 🚨 Problem Description

**Error Messages/Logs:**
```
(dma) daniel@daniel:~/0.13.2$ sudo /home/daniel/dma/bin/python3 pcileech.py check
09:30:49 INFO   09:30:49 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
09:30:49 INFO   09:30:49 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
09:30:49 INFO   09:30:49 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
09:30:49 INFO   09:30:49 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
09:30:49 INFO   09:30:49 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
09:30:49 INFO   09:30:49 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
09:30:49 INFO   09:30:49 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
09:30:49 INFO   09:30:49 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
09:30:49 INFO   09:30:49 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
09:30:49 INFO   09:30:49 │  INFO  │ Discovered 8 boards
09:30:49 INFO   09:30:49 │  INFO  │ [CHECK] Running VFIO diagnostics
09:30:49 INFO   09:30:49 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
09:30:49 INFO   09:30:49 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered

(dma) daniel@daniel:~/0.13.2$ sudo /home/daniel/dma/bin/python3 pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1
09:31:38 INFO   09:31:38 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
09:31:38 INFO   09:31:38 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
09:31:38 INFO   09:31:38 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
09:31:38 INFO   09:31:38 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
09:31:38 INFO   09:31:38 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
09:31:38 INFO   09:31:38 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
09:31:38 INFO   09:31:38 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
09:31:38 INFO   09:31:38 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
09:31:38 INFO   09:31:38 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
09:31:38 INFO   09:31:38 │  INFO  │ Discovered 8 boards
09:31:38 INFO   09:31:38 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
09:31:38 WARNING   09:31:38 │ WARNING│ [VFIO] VFIO constants rebuild failed: 
09:31:38 WARNING   09:31:38 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
09:31:38 INFO   09:31:38 │  INFO  │ [BUILD] Starting build for device 0000:08:00.0 on board pcileech_75t484_x1
09:31:38 WARNING   09:31:38 │ WARNING│ [BUILD] Podman not available or cannot connect

============================================================
⚠️  Podman is not available or cannot connect.
============================================================

[... 127 lines cut ...]
- [ ] Always reproducible
- [ ] Intermittent (occurs sometimes)
- [ ] Rare (occurred once or twice)

**Impact Assessment:**
- [ ] Blocks development/testing
- [ ] Affects firmware functionality
- [ ] Synthesis/build failures
- [ ] Runtime errors
- [ ] Performance issues

**Workaround Available:**
- [ ] Yes (please describe below)
- [ ] No

## ✅ Pre-submission Checklist

**I have:**
- [ ] Checked that I'm using the latest version from the main branch
- [ ] Searched for existing similar issues
- [ ] Read the relevant documentation/wiki pages
- [ ] Included all required information above
- [ ] Removed any sensitive/personal information
- [ ] Tested with minimal reproduction case
- [ ] Verified this isn't a configuration issue

ramseymcgrath, 2025-08-16:
what kernel are you on

StixuYT, 2025-08-16:
I always run the check first, you can see it in the issue itself, i posted it above, I run 6.15.9 xanmod so I can break IOMMU groups.

StixuYT, 2025-08-16:
I have tried changing the kernel to liquorix 6.15.10(6.15.10-1-liquorix-amd64 to be exact), and still same issues, tried installing podman(because I was building locally, but another error occured)

Here's the build with podman:
```
(dma) daniel@daniel:~/0.13.2$ sudo /home/daniel/dma/bin/python3 pcileech.py check
15:43:47 INFO   15:43:47 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
15:43:47 INFO   15:43:47 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
15:43:47 INFO   15:43:47 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
15:43:47 INFO   15:43:47 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
15:43:47 INFO   15:43:47 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
15:43:47 INFO   15:43:47 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
15:43:47 INFO   15:43:47 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
15:43:47 INFO   15:43:47 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
15:43:47 INFO   15:43:47 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
15:43:47 INFO   15:43:47 │  INFO  │ Discovered 8 boards
15:43:47 INFO   15:43:47 │  INFO  │ [CHECK] Running VFIO diagnostics
15:43:47 INFO   15:43:47 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
15:43:47 INFO   15:43:47 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered

(dma) daniel@daniel:~/0.13.2$ sudo /home/daniel/dma/bin/python3 pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1
15:44:12 INFO   15:44:12 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
15:44:12 INFO   15:44:12 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
15:44:12 INFO   15:44:12 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
15:44:12 INFO   15:44:12 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
15:44:12 INFO   15:44:12 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
15:44:12 INFO   15:44:12 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
15:44:12 INFO   15:44:12 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
15:44:12 INFO   15:44:12 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
15:44:12 INFO   15:44:12 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
15:44:12 INFO   15:44:12 │  INFO  │ Discovered 8 boards
15:44:12 INFO   15:44:12 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
15:44:12 WARNING   15:44:12 │ WARNING│ [VFIO] VFIO constants rebuild failed: 
15:44:12 WARNING   15:44:12 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
15:44:12 INFO   15:44:12 │  INFO  │ [BUILD] Starting build for device 0000:08:00.0 on board pcileech_75t484_x1
15:44:12 INFO   15:44:12 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> 0413ac3acf74
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
Get:1 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]
Get:2 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:3 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [1271 kB]
Get:4 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [3209 kB]
Get:5 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [48.5 kB]
Get:6 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [5111 kB]
Get:7 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
Get:8 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [127 kB]
[... cut ...]
15:47:47 ERROR   15:47:47 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Failed to open VFIO device FD for 0000:08:00.0: Failed to link group 25 to container: [Errno 25] Inappropriate ioctl for device
15:47:47 ERROR Build failed: Failed to link group 25 to container: [Errno 25] Inappropriate ioctl for device
15:47:47 ERROR Unexpected error: Failed to link group 25 to container: [Errno 25] Inappropriate ioctl for device
15:47:47 ERROR   15:47:47 │ ERROR  │ [LOCAL] Local build failed after 31.3s: Local build failed with exit code 1
15:47:47 ERROR Build failed: Local build failed with exit code 1
(dma) daniel@daniel:~/0.13.2$ 
```

StixuYT, 2025-08-16:
But that's what I am doing? I have created a virtual environment using python3 -m dma(thats what I called it) and before starting any of the checks and trying to build i start it using source dma/bin/activate. What do you mean exactly?

sthgi, 2025-08-16:
Ah sorry missed that.

try deleting your existing folders for PCILeechFWGenerator then do a fresh
git clone https://github.com/ramseymcgrath/PCILeechFWGenerator.git
cd PCILeechFWGenerator
sudo modprobe vfio vfio_pci vfio_iommu_type1
sudo /home/daniel/dma/bin/python3 ./patch_vfio_constants.py

sudo /home/daniel/dma/bin/python3 ./pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1

sthgi, 2025-08-16:
booyaa_