VoltCyclone/PCILeechFWGenerator #277: 0.13.3

AMRICHASFUCK, 2025-08-16:
07:57:01 - INFO -   07:57:01 │  INFO  │ [VFIO] Starting config space read for device 0000:03:00.0, strict_mode=False
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Reading configuration space for device 0000:03:00.0 via sysfs (non-strict mode)
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:03:00.0/config
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:03:00.0/config
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Reading up to 4096 bytes for extended config space
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Successfully read 4096 bytes from sysfs
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Read config space for device 8086:2723
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Successfully read 4096 bytes via sysfs
07:57:01 - INFO -   07:57:01 │  INFO  │ [SUBS] Subsystem ID extraction - Vendor: 0x8086, Device: 0x0084
07:57:01 - INFO -   07:57:01 │  INFO  │ [SUBS] Subsystem IDs differ from main IDs - Main: 0x8086:0x2723, Subsystem: 0x8086:0x0084
07:57:01 - INFO -   07:57:01 │  INFO  │ [CNFG] Starting BAR extraction from config space (4096 bytes)
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARX] BAR 0 is active (non-zero): 0xdf100004
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARX] BAR 0 properties: type=memory, prefetchable=False, is_64bit=True
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARX] BAR 0 full 64-bit address: 0x00000000df100000
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARX] BAR 0 size from sysfs: 16384 bytes (16.0KB)
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARS] Added BAR 0: BAR 0: memory @ 0x00000000df100000 (64-bit, non-prefetchable, size=0x4000)
07:57:01 - INFO -   07:57:01 │  INFO  │ [BARX] Completed BAR extraction: found 1 active BARs
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for board.name
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Marked 9 variables as critical (no fallbacks)
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for board.name
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for board.fpga_part
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for board.fpga_family
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for board.pcie_ip_type
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for sys_clk_freq_mhz
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for generated_xdc_path
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for board_xdc_content
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for max_lanes
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for supports_msi
07:57:01 - INFO -   07:57:01 │  INFO  │ [FALLBACK] Applied fallback for supports_msix
07:57:01 - INFO -   07:57:01 │  INFO  │ Context built successfully: 8086:2723_adf8746e3276ec9d
07:57:01 - INFO -   07:57:01 │  INFO  │ [PCIL] Template context validation successful
07:57:01 - INFO -   07:57:01 │  INFO  │ Template context built successfully with 17 top-level keys
07:57:01 - INFO -   07:57:01 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:03:00.0
07:57:01 - INFO -   07:57:01 │  INFO  │ Generating SystemVerilog modules
07:57:01 - INFO -   07:57:01 │  INFO  │ Using PCILeech as primary SystemVerilog generation path
07:57:01 - INFO -   07:57:01 │  INFO  │ Generating PCILeech SystemVerilog modules
07:57:01 - ERROR -   07:57:01 │ ERROR  │ [TEMPLATE] Failed to render template 'sv/top_level_wrapper.sv.j2': 'dict object' has no attribute 'enabled'
07:57:01 - ERROR -   07:57:01 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to render template 'sv/top_level_wrapper.sv.j2': 'dict object' has no attribute 'enabled'
07:57:01 - ERROR -   07:57:01 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to render template 'sv/top_level_wrapper.sv.j2': 'dict object' has no attribute 'enabled'
07:57:01 - ERROR - Build failed: 'dict object' has no attribute 'enabled'
07:57:01 - ERROR - Unexpected error: 'dict object' has no attribute 'enabled'
10:57:01 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/10               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/marxon/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board 35t' returned non-zero exit status 1.

ramseymcgrath, 2025-08-16:
Attach the whole log 🙏

AMRICHASFUCK, 2025-08-16:
[build.log](https://github.com/user-attachments/files/21810848/build.log)

casoni-razvan-30323, 2025-08-16:
did someone solve this?

1nferious, 2025-08-16:
(.venv) john@john-System-Product-Name:~/Desktop/PCILeechFWGenerator$ sudo BDF=0000:01:00.0 PYTHONPATH=src .venv/bin/python pcileech.py build --board pcileech_75t484_x1 --bdf 0000:01:00.0
13:53:03 INFO   13:53:03 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
13:53:03 INFO   13:53:03 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
13:53:03 INFO   13:53:03 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
13:53:03 INFO   13:53:03 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
13:53:03 INFO   13:53:03 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
13:53:03 INFO   13:53:03 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
13:53:03 INFO   13:53:03 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
13:53:03 INFO   13:53:03 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
13:53:03 INFO   13:53:03 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
13:53:03 INFO   13:53:03 │  INFO  │ Discovered 8 boards
13:53:03 INFO   13:53:03 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
13:53:03 INFO   13:53:03 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
13:53:03 INFO   13:53:03 │  INFO  │ [BUILD] Starting build for device 0000:01:00.0 on board pcileech_75t484_x1
13:53:03 INFO   13:53:03 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 97e7bdc6233d3546d75530b8bb5a123caa4b34452cb004df8034eb5d95af2705
--> 97e7bdc6233d
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
Get:1 http://archive.ubuntu.com/ubuntu jammy InRelease [270 kB]
Get:2 http://security.ubuntu.com/ubuntu jammy-security InRelease [129 kB]
Get:3 http://archive.ubuntu.com/ubuntu jammy-updates InRelease [128 kB]
Get:4 http://security.ubuntu.com/ubuntu jammy-security/multiverse amd64 Packages [48.5 kB]
Get:5 http://archive.ubuntu.com/ubuntu jammy-backports InRelease [127 kB]
Get:6 http://security.ubuntu.com/ubuntu jammy-security/universe amd64 Packages [1271 kB]
Get:7 http://archive.ubuntu.com/ubuntu jammy/restricted amd64 Packages [164 kB]
Get:8 http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages [1792 kB]
Get:9 http://security.ubuntu.com/ubuntu jammy-security/main amd64 Packages [3209 kB]
Get:10 http://archive.ubuntu.com/ubuntu jammy/universe amd64 Packages [17.5 MB]
Get:11 http://security.ubuntu.com/ubuntu jammy-security/restricted amd64 Packages [5111 kB]
Get:12 http://archive.ubuntu.com/ubuntu jammy/multiverse amd64 Packages [266 kB]
Get:13 http://archive.ubuntu.com/ubuntu jammy-updates/universe amd64 Packages [1575 kB]
Get:14 http://archive.ubuntu.com/ubuntu jammy-updates/restricted amd64 Packages [5299 kB]
Get:15 http://archive.ubuntu.com/ubuntu jammy-updates/multiverse amd64 Packages [75.9 kB]
Get:16 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages [3521 kB]
Get:17 http://archive.ubuntu.com/ubuntu jammy-backports/main amd64 Packages [83.2 kB]
Get:18 http://archive.ubuntu.com/ubuntu jammy-backports/universe amd64 Packages [35.2 kB]
Fetched 40.6 MB in 1s (30.4 MB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-common binutils-x86-64-linux-gnu bzip2 cpp cpp-11 dpkg-dev
  g++ g++-11 gcc gcc-11 gcc-11-base libasan6 libatomic1 libbinutils
  libc-dev-bin libc6-dev libcc1-0 libcrypt-dev libctf-nobfd0 libctf0
  libdpkg-perl libelf1 libexpat1 libgcc-11-dev libgdbm-compat4 libgdbm6
  libgomp1 libisl23 libitm1 libkmod2 liblsan0 libmpc3 libmpdec3 libmpfr6
  libnsl-dev libpci3 libperl5.34 libpython3-stdlib libpython3.10-minimal
  libpython3.10-stdlib libquadmath0 libreadline8 libsqlite3-0 libstdc++-11-dev
  libtirpc-dev libtsan0 libubsan1 linux-headers-5.15.0-151
  linux-headers-5.15.0-151-generic linux-libc-dev lto-disabled-list make
  media-types openssl patch pci.ids perl perl-base perl-modules-5.34
  python3-distutils python3-lib2to3 python3-minimal python3-pkg-resources
  python3-setuptools python3-wheel python3.10 python3.10-minimal
  readline-common rpcsvc-proto xz-utils
Suggested packages:
  binutils-doc bzip2-doc cpp-doc gcc-11-locales debian-keyring g++-multilib
  g++-11-multilib gcc-11-doc gcc-multilib manpages-dev autoconf automake
  libtool flex bison gdb gcc-doc gcc-11-multilib glibc-doc gnupg git bzr
  gdbm-l10n libstdc++-11-doc make-doc ed diffutils-doc wget | curl | lynx-cur
  perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl
  libtap-harness-archive-perl python3-doc python3-tk python3-venv
[... cut ...]
17:54:21 - ERROR - Unexpected error: 'dict object' has no attribute 'enabled'
13:54:21 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/14               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /home/john/Desktop/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:01:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.

sthgi, 2025-08-16:
i get same thing on capture card

Danielix99, 2025-08-17:
Still in the code of your pr this error come up

00:38:53 - INFO -   00:38:53 │  INFO  │ [BARX] Completed BAR extraction: found 3 active BARs
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for board.name
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Marked 9 variables as critical (no fallbacks)
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for board.name
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for board.fpga_part
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for board.fpga_family
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for board.pcie_ip_type
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for sys_clk_freq_mhz
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for generated_xdc_path
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for board_xdc_content
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for max_lanes
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for supports_msi
00:38:53 - INFO -   00:38:53 │  INFO  │ [FALLBACK] Applied fallback for supports_msix
00:38:53 - ERROR -   00:38:53 │ ERROR  │ [VALIDATE] Device identification validation failed for 0000:06:00.0: invalid literal for int() with base 0: '----'
00:38:53 - INFO -   00:38:53 │  INFO  │ Context built successfully: ----:----_----------------
00:38:53 - INFO -   00:38:53 │  INFO  │ [PCIL] Template context validation successful
00:38:53 - INFO -   00:38:53 │  INFO  │ Template context built successfully with 35 top-level keys
00:38:53 - INFO -   00:38:53 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:06:00.0
00:38:53 - INFO -   00:38:53 │  INFO  │ Generating SystemVerilog modules
00:38:53 - INFO -   00:38:53 │  INFO  │ Using PCILeech as primary SystemVerilog generation path
00:38:53 - INFO -   00:38:53 │  INFO  │ Generating PCILeech SystemVerilog modules
00:38:53 - ERROR -   00:38:53 │ ERROR  │ PCILeech SystemVerilog generation failed: device_config must be a dictionary, got TemplateObject. Cannot proceed with firmware generation.
00:38:53 - ERROR -   00:38:53 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: device_config must be a dictionary, got TemplateObject. Cannot proceed with firmware generation.
00:38:53 - ERROR - Build failed: device_config must be a dictionary, got TemplateObject. Cannot proceed with firmware generation.
00:38:53 - ERROR - Unexpected error: device_config must be a dictionary, got TemplateObject. Cannot proceed with firmware generation.

closed 2025-08-22 by ramseymcgrath