VoltCyclone/PCILeechFWGenerator #282: [BUG] New errors

StixuYT, 2025-08-16:
## 📋 Bug Summary

**Brief Description:**
Local build failed with exit code 1

**Severity Level:**

- [ ] **Critical** - System crashes, data loss, or complete failure
- [ ] **High** - Major functionality broken, significant impact
- [ ] **Medium** - Minor functionality issues, workarounds available
- [ ] **Low** - Cosmetic issues, minor inconveniences

## 📦 Environment Details

[systeminfo](https://imgur.com/3nI2EwG)

## 🚨 Problem Description

**Error Messages/Logs:**
```
(dma) daniel@daniel:~/0.13.3$ sudo /home/daniel/dma/bin/python3 pcileech.py check
[sudo] password for daniel: 
01:25:04 INFO   01:25:04 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
01:25:04 INFO   01:25:04 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
01:25:04 INFO   01:25:04 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
01:25:04 INFO   01:25:04 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
01:25:04 INFO   01:25:04 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
01:25:04 INFO   01:25:04 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
01:25:04 INFO   01:25:04 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
01:25:04 INFO   01:25:04 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
01:25:04 INFO   01:25:04 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
01:25:04 INFO   01:25:04 │  INFO  │ Discovered 8 boards
01:25:04 INFO   01:25:04 │  INFO  │ [CHECK] Running VFIO diagnostics
01:25:04 INFO   01:25:04 │  INFO  │ [VFIO] Starting VFIO diagnostics for device: system-wide
01:25:04 INFO   01:25:04 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered

(dma) daniel@daniel:~/0.13.3$ sudo /home/daniel/dma/bin/python3 pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1
01:25:07 INFO   01:25:07 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
01:25:07 INFO   01:25:07 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
01:25:07 INFO   01:25:07 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
01:25:07 INFO   01:25:07 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
01:25:07 INFO   01:25:07 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
01:25:07 INFO   01:25:07 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
01:25:07 INFO   01:25:07 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
01:25:07 INFO   01:25:07 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
01:25:07 INFO   01:25:07 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
01:25:07 INFO   01:25:07 │  INFO  │ Discovered 8 boards
01:25:07 INFO   01:25:07 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
01:25:07 WARNING   01:25:07 │ WARNING│ [VFIO] VFIO constants rebuild failed: 
01:25:07 WARNING   01:25:07 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
01:25:07 INFO   01:25:07 │  INFO  │ [BUILD] Starting build for device 0000:08:00.0 on board pcileech_75t484_x1
01:25:07 WARNING   01:25:07 │ WARNING│ [BUILD] Podman not available or cannot connect

============================================================
⚠️  Podman is not available or cannot connect.
============================================================
[... 263 lines cut ...]
- [ ] Always reproducible
- [ ] Intermittent (occurs sometimes)
- [ ] Rare (occurred once or twice)

**Impact Assessment:**
- [ ] Blocks development/testing
- [ ] Affects firmware functionality
- [ ] Synthesis/build failures
- [ ] Runtime errors
- [ ] Performance issues

**Workaround Available:**
- [ ] Yes (please describe below)
- [ ] No

## ✅ Pre-submission Checklist

**I have:**
- ✅ Checked that I'm using the latest version from the main branch
- ✅ Searched for existing similar issues
- ✅ Read the relevant documentation/wiki pages
- ✅ Included all required information above
- ✅ Removed any sensitive/personal information
- ✅ Tested with minimal reproduction case
- ✅ Verified this isn't a configuration issue

StixuYT, 2025-08-16:
Happens on the new bugfix enabled branch as well.

ramseymcgrath, 2025-08-17:
> Happens on the new bugfix enabled branch as well.

pull and rerun pls

StixuYT, 2025-08-17:
After the new update I get this old error again
```
(dma) daniel@daniel:~/0.13.4-beta.2$ sudo /home/daniel/dma/bin/python3 pcileech.py build --bdf 0000:08:00.0 --board pcileech_75t484_x1
09:02:23 INFO   09:02:23 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
09:02:23 INFO   09:02:23 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
09:02:23 INFO   09:02:23 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
09:02:23 INFO   09:02:23 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
09:02:23 INFO   09:02:23 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
09:02:23 INFO   09:02:23 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
09:02:23 INFO   09:02:23 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
09:02:23 INFO   09:02:23 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
09:02:23 INFO   09:02:23 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
09:02:23 INFO   09:02:23 │  INFO  │ Discovered 8 boards
09:02:23 INFO   09:02:23 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
09:02:23 WARNING   09:02:23 │ WARNING│ [VFIO] VFIO constants rebuild failed: 
09:02:23 WARNING   09:02:23 │ WARNING│ [VFIO] VFIO constants rebuild failed - may cause ioctl errors
09:02:23 INFO   09:02:23 │  INFO  │ [BUILD] Starting build for device 0000:08:00.0 on board pcileech_75t484_x1
09:02:23 WARNING   09:02:23 │ WARNING│ [BUILD] Podman not available or cannot connect

============================================================
⚠️  Podman is not available or cannot connect.
============================================================

The build normally runs in a container for consistency.
However, you can run the build locally on your system.

Note: Local builds require all dependencies to be installed.
      (Vivado, Python packages, etc.)

Would you like to run the build locally? [y/N]: y
09:02:26 INFO   09:02:26 │  INFO  │ [LOCAL] Running local build - board=pcileech_75t484_x1
09:02:26 INFO   09:02:26 │  INFO  │ [LOCAL] Executing local build with args: --bdf 0000:08:00.0 --board pcileech_75t484_x1
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for board.name
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
09:02:26 INFO   09:02:26 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
09:02:26 INFO   09:02:26 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:08:00.0
09:02:26 INFO   09:02:26 │  INFO  │ AdvancedSVGenerator initialized with templating system, PCILeech primary: True
09:02:26 INFO   09:02:26 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
09:02:26 INFO No device profile specified, using live device detection
09:02:26 INFO ➤ Preloading MSI-X data before VFIO binding
09:02:26 INFO   09:02:26 │  INFO  │ MSI-X capability found: 4 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
09:02:26 WARNING   09:02:26 │ WARNING│ MSI-X table offset 0x4 is not 8-byte aligned (actual offset: 0x4, aligned: 0x0)
09:02:26 INFO   • Found MSI-X capability: 4 vectors
09:02:26 INFO ➤ Generating PCILeech firmware …
09:02:26 INFO   09:02:26 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:08:00.0
09:02:26 INFO   09:02:26 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
09:02:26 INFO   09:02:26 │  INFO  │ MSI-X capability found: 4 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
09:02:26 WARNING   09:02:26 │ WARNING│ MSI-X table offset 0x4 is not 8-byte aligned (actual offset: 0x4, aligned: 0x0)
09:02:26 INFO   09:02:26 │  INFO  │ [MSIX] Preloaded MSI-X capability: 4 vectors, table BIR 4, offset 0x0
09:02:26 INFO   09:02:26 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
09:02:26 INFO   09:02:26 │  INFO  │ [PROFILER] Monitoring started
09:02:56 INFO   09:02:56 │  INFO  │ [MSIX] Captured 21171 register accesses with 2 timing patterns
09:02:56 INFO   09:02:56 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:08:00.0
[... cut ...]
09:00:51 INFO   09:00:51 │  INFO  │ [VFIO] Diagnostics completed. Overall status: ok, Can proceed: True

=== VFIO DIAGNOSTIC REPORT ===
Overall: ✔ OK
Proceed: yes

✔ Platform: Linux detected
✔ IOMMU HW: VT‑d / AMD‑Vi supported by CPU
✔ Kernel cmdline: IOMMU enabled in cmdline: amd_iommu=on, iommu=pt, iommu=on, ACS override: pcie_acs_override=downstream, pcie_acs_override=downstream,multifunction
✔ Kernel modules: All VFIO modules loaded: vfio, vfio_pci, vfio_iommu_type1
✔ vfio-pci driver: vfio-pci registered
```