VoltCyclone/PCILeechFWGenerator #284: [BUG] Install Error

smallprogram, 2025-08-17:
OS: Ubuntu 24.04

Run:
```
# Clone the repository
git clone https://github.com/ramseymcgrath/PCILeechFWGenerator.git
cd PCILeechFWGenerator

# Create virtual environment (recommended)
python3 -m venv .venv
source .venv/bin/activate  # On Windows: .venv\Scripts\activate

pip install -e .[tui]
```
Report Error:
```
(.venv) zhusir@zhusirubuntutestpc:~/PCILeechFWGenerator$ pip install -e .
Obtaining file:///home/zhusir/PCILeechFWGenerator
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [72 lines of output]
      /tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!
      
              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:
      
              License :: OSI Approved :: MIT License
      
              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************
      
      !!
        self._finalize_license_expression()
      /tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools_scm/_integration/version_inference.py:51: UserWarning: version of PCILeechFWGenerator already set
        warnings.warn(self.message)
      
      
      An error occurred while building the project, please ensure you have the most updated version of setuptools, setuptools_scm and wheel with:
         pip install -U setuptools setuptools_scm wheel
      
      
      Traceback (most recent call last):
        File "/home/zhusir/.venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/zhusir/.venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/zhusir/.venv/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 132, in get_requires_for_build_editable
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 473, in get_requires_for_build_editable
          return self.get_requires_for_build_wheel(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 331, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 301, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 317, in run_setup
          exec(code, locals())
        File "<string>", line 14, in <module>
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/__init__.py", line 115, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 160, in setup
          dist.parse_config_files()
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/dist.py", line 752, in parse_config_files
          setupcfg.parse_configuration(
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 188, in parse_configuration
          meta.parse()
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 502, in parse
          section_parser_method(section_options)
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 477, in parse_section
          self[name] = value
          ~~~~^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 294, in __setitem__
          parsed = self.parsers.get(option_name, lambda x: x)(value)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 600, in _parse_version
          return expand.version(self._parse_attr(value, self.package_dir, self.root_dir))
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/setupcfg.py", line 419, in _parse_attr
          return expand.read_attr(attr_desc, package_dir, root_dir)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/expand.py", line 182, in read_attr
          spec = _find_spec(module_name, path)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-lce88zip/overlay/lib/python3.12/site-packages/setuptools/config/expand.py", line 196, in _find_spec
          spec = spec or importlib.util.find_spec(module_name)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<frozen importlib.util>", line 92, in find_spec
      ModuleNotFoundError: No module named 'src'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

sthgi, 2025-08-17:
I got the same thing.  while we are working through bugs running from source will work.

smallprogram, 2025-08-17:
> I got the same thing. while we are working through bugs running from source will work.

Can I revert to an earlier source version to resolve this issue?

sthgi, 2025-08-17:
not that will produce firmware

ramseymcgrath, 2025-08-17:
Yea CI builds threw the same, package config typo

ramseymcgrath, 2025-08-17:
complete in v0.13.4-beta.4 onward