VoltCyclone/PCILeechFWGenerator #297: [BUG] branch enabled-bugfix 0.14.4b10 build error Failed to read actual MSI-X table from hardware

smallprogram, 2025-08-20:
### OS
ubuntu22.04
### kernel
6.8.0-65-generic
### headware
Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8161] (rev 15)
### command
```sh
cd PCILeechFWGenerator
git pull
source .venv/bin/activate 
pip install -e .[tui]
sudo modprobe vfio vfio-pci vfio-iommu-type1
echo "10ec 8161" | sudo tee /sys/bus/pci/drivers/vfio-pci/new_id
echo "0000:03:00.0" | sudo tee /sys/bus/pci/devices/0000:03:00.0/driver/unbind 
echo "0000:03:00.0" | sudo tee /sys/bus/pci/drivers/vfio-pci/bind 
$(which python3) pcileech.py check --device 0000:03:00.0
sudo $(which python3) pcileech.py build \
  --bdf 0000:03:00.0 \
  --board pcileech_75t484_x1
```
### report
```
07:26:16 INFO   07:26:16 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.34.1
.git
07:26:16 INFO   07:26:16 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
07:26:16 INFO   07:26:16 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
07:26:16 INFO   07:26:16 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
07:26:16 INFO   07:26:16 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
07:26:16 INFO   07:26:16 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
07:26:16 INFO   07:26:16 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
07:26:16 INFO   07:26:16 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
07:26:16 INFO   07:26:16 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
07:26:16 INFO   07:26:16 │  INFO  │ Discovered 8 boards
07:26:16 INFO   07:26:16 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
07:26:16 INFO   07:26:16 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
07:26:16 INFO   07:26:16 │  INFO  │ [BUILD] Starting build for device 0000:03:00.0 on board pcileech_75t484_x1
07:26:19 WARNING   07:26:19 │ WARNING│ 
============================================================
📦 A new version of PCILeech Firmware Generator is available!
   Current version: 0.13.4
   Latest version:  0.13.4b10
   Build date:      2025-08-16T17:44:00.775071
   Commit hash:     4ad3581

   Update with one of these commands:
   • pip install --upgrade pcileech-fw-generator
   • git pull (if installed from source)

   Release notes: https://github.com/ramseymcgrath/PCILeechFWGenerator/releases
============================================================

07:26:20 INFO   07:26:20 │  INFO  │ Building container image pcileech-fw-generator:latest
[1/2] STEP 1/7: FROM ubuntu:22.04 AS build
[1/2] STEP 2/7: ENV DEBIAN_FRONTEND=noninteractive     LANG=C.UTF-8 LC_ALL=C.UTF-8 TZ=UTC     PIP_BREAK_SYSTEM_PACKAGES=1
--> Using cache 65183df351e840f8090a7b97a51a9e2ae01e37502e558938c8b7743ff2b1716e
--> 65183df351e
[1/2] STEP 3/7: RUN apt-get update && apt-get install -y --no-install-recommends         python3 python3-pip build-essential         linux-headers-generic         pciutils kmod ca-certificates &&     rm -rf /var/lib/apt/lists/*
--> Using cache f4cbe44ef81297dd80d40b3f88740e34fac609621f1499829efe87d5587445b0
--> f4cbe44ef81
[1/2] STEP 4/7: WORKDIR /src
--> Using cache 6d56b9a0c6ef8e0768d7364787a78d0d6ddbc2e7160d5c53bd53870f7aed5d99
--> 6d56b9a0c6e
[1/2] STEP 5/7: COPY vfio_helper.c patch_vfio_constants.py build_vfio_constants.sh ./
--> Using cache 31d3acd3d32cc07476f4411751e899323ae413adac02237265ec54f985588a53
--> 31d3acd3d32
[1/2] STEP 6/7: COPY src/cli/vfio_constants.py ./src/cli/
--> Using cache 288b9105bfc737e8b0c681e4f77dbbe865f5055b9625583cf212f6f656ff2808
--> 288b9105bfc
[... 152 lines cut ...]
}
07:27:07 - INFO -   07:27:07 │  INFO  │ Context built successfully: 10ec:8161
07:27:07 - INFO -   07:27:07 │  INFO  │ [PCIL] Template context validation successful
07:27:07 - INFO -   07:27:07 │  INFO  │ Template context built successfully with 39 top-level keys
07:27:07 - INFO -   07:27:07 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:03:00.0
07:27:07 - INFO -   07:27:07 │  INFO  │ Generating SystemVerilog modules
07:27:07 - INFO -   07:27:07 │  INFO  │ Using PCILeech as primary SystemVerilog generation path
07:27:07 - INFO -   07:27:07 │  INFO  │ Generating PCILeech SystemVerilog modules
07:27:07 - INFO -   07:27:07 │  INFO  │ Generating MSI-X modules
07:27:07 - INFO -   07:27:07 │  INFO  │ Available BARs for MSI-X table lookup: [{'index': 2, 'size': 4096, 'type': 'BarConfiguration'}]
07:27:07 - INFO -   07:27:07 │  INFO  │ Looking for BAR 4 for MSI-X table access
07:27:07 - INFO -   07:27:07 │  INFO  │ No direct BIR match found, analyzing BAR layout for MSI-X table
07:27:07 - INFO -   07:27:07 │  INFO  │ Original config space BARs: []
07:27:07 - WARNING -   07:27:07 │ WARNING│ Using largest memory BAR as fallback for MSI-X table: region 0, size 4096
07:27:07 - INFO -   07:27:07 │  INFO  │ Starting VFIO device fd acquisition for 0000:03:00.0
07:27:07 - INFO -   07:27:07 │  INFO  │ [VFIO] Device 0000:03:00.0 is in IOMMU group 12
07:27:07 - INFO -   07:27:07 │  INFO  │ [VFIO] Successfully obtained device fd 6 for 0000:03:00.0
07:27:07 - WARNING -   07:27:07 │ WARNING│ mmap failed with EINVAL - BAR may not be mappable or accessible
07:27:07 - ERROR -   07:27:07 │ ERROR  │ Failed to read actual MSI-X table from hardware: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO and try again.
07:27:07 - ERROR -   07:27:07 │ ERROR  │ PCILeech SystemVerilog generation failed: Failed to read actual MSI-X table from hardware: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO and try again.
07:27:07 - ERROR -   07:27:07 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: Failed to read actual MSI-X table from hardware: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO and try again.
07:27:07 - ERROR - Build failed: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.
07:27:07 - ERROR - Unexpected error: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.
07:27:08 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/12               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.
```

ramseymcgrath, 2025-08-21:
run on 11

smallprogram, 2025-08-21:
> run on 11

Still not working

```
Successfully tagged localhost/pcileech-fw-generator:latest
c33de05bfa28ba3b274ac5caebe48133357cfbc043ab6ae8d86046e532e736bb
08:34:05 INFO   08:34:05 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:06:00.0
08:34:05 INFO   08:34:05 │  INFO  │ [BIND] Current driver for 0000:06:00.0: vfio-pci
08:34:05 INFO   08:34:05 │  INFO  │ [BIND] Device 0000:06:00.0 already bound to vfio-pci
08:34:05 INFO   08:34:05 │  INFO  │ [CONT] Launching build container - board=pcileech_75t484_x1, tag=latest
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for board.name
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for meta
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for process_var
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for title
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for _td
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for error_name
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Registered default fallback for error_value
08:34:06 - INFO -   08:34:06 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
08:34:06 - INFO -   08:34:06 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:06:00.0
08:34:06 - INFO -   08:34:06 │  INFO  │ AdvancedSVGenerator initialized with templating system, PCILeech primary: True
08:34:06 - INFO -   08:34:06 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
08:34:06 - INFO - No device profile specified, using live device detection
08:34:06 - INFO - ➤ Preloading MSI-X data before VFIO binding
08:34:06 - INFO -   08:34:06 │  INFO  │ MSI-X capability found: 4 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
08:34:06 - INFO -   • Found MSI-X capability: 4 vectors
08:34:06 - INFO - ➤ Generating PCILeech firmware …
08:34:06 - INFO -   08:34:06 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:06:00.0
08:34:06 - INFO -   08:34:06 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
08:34:06 - INFO -   08:34:06 │  INFO  │ MSI-X capability found: 4 entries, table BIR 4 offset 0x0, PBA BIR 4 offset 0x800
[... cut ...]
08:34:48 - ERROR - Unexpected error: Failed to read actual MSI-X table data from hardware. Cannot generate safe firmware without real MSI-X table values. Ensure the device is properly accessible via VFIO.
08:34:48 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/12               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:06:00.0 --board pcileech_75t484_x1' returned non-zero exit status 1.
```

ramseymcgrath, 2025-08-22:
its not your device, rerun on current vers and open a new ticket with the logs if you still see the same error