VoltCyclone/PCILeechFWGenerator #321: [BUG] 0.13.8

sthgi, 2025-08-24:
USB3 PCI

Successfully tagged localhost/pcileech-fw-generator:latest
9e006aca5f5c708c9cfcf771667831318dfdfce822a13799ae3af2896339a0d8
01:22:34 INFO   01:22:34 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
01:22:34 INFO   01:22:34 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
01:22:34 INFO   01:22:34 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
01:22:34 INFO   01:22:34 │  INFO  │ [CONT] Launching build container - board=pcileech_100t484_x1, tag=latest
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for board.name
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for meta
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for process_var
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for title
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for _td
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for error_name
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Registered default fallback for error_value
06:22:35 - INFO -   06:22:35 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
06:22:35 - INFO -   06:22:35 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:03:00.0
06:22:35 - INFO -   06:22:35 │  INFO  │ SystemVerilogGenerator initialized successfully
06:22:35 - INFO -   06:22:35 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
06:22:35 - INFO -   06:22:35 │  INFO  │ No device profile specified, using live device detection
06:22:35 - INFO - ➤ Preloading MSI-X data before VFIO binding
06:22:35 - INFO -   06:22:35 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
06:22:35 - INFO -   • Found MSI-X capability: 8 vectors
06:22:35 - INFO - ➤ Generating PCILeech firmware …
06:22:35 - INFO -   06:22:35 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:03:00.0
06:22:35 - INFO -   06:22:35 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
06:22:35 - INFO -   06:22:35 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
06:22:35 - INFO -   06:22:35 │  INFO  │ [MSIX] Preloaded MSI-X capability: 8 vectors, table BIR 0, offset 0x1000
06:22:35 - INFO -   06:22:35 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
06:22:38 - INFO -   06:22:38 │  INFO  │ [PROFILER] Monitoring started
06:23:09 - INFO -   06:23:09 │  INFO  │ [MSIX] Captured 40794 register accesses with 2 timing patterns
06:23:09 - INFO -   06:23:09 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
06:23:09 - INFO -   06:23:09 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
06:23:09 - INFO -   06:23:09 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
06:23:10 - INFO -   06:23:10 │  INFO  │ [VFIO] VFIO binding established for device 0000:03:00.0 at /dev/vfio/32
06:23:10 - INFO -   06:23:10 │  INFO  │ [MSIX] Analyzing configuration space for device 0000:03:00.0 (VFIO already active)
06:23:10 - INFO -   06:23:10 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:03:00.0/config
06:23:10 - INFO -   06:23:10 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:03:00.0/config
[... 121 lines cut ...]
06:23:12 - INFO -   06:23:12 │  INFO  │ [PCIL] Template context validation successful
06:23:12 - INFO -   06:23:12 │  INFO  │ Template context built successfully with 35 top-level keys
06:23:12 - INFO -   06:23:12 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:03:00.0
06:23:12 - INFO -   06:23:12 │  INFO  │ Generating SystemVerilog modules
06:23:12 - INFO -   06:23:12 │  INFO  │ Generating PCILeech SystemVerilog modules
06:23:12 - ERROR - Post-conversion context top-level types: {'device_signature': 'str', 'header': 'str', 'fifo_type': 'str', 'fifo_depth': 'int', 'data_width': 'int', 'fpga_family': 'str', 'vendor_id': 'str', 'device_id': 'str', 'vendor_id_hex': 'str', 'device_id_hex': 'str', 'vendor_id_int': 'int', 'device_id_int': 'int', 'device_config': 'TemplateObject', 'msix_config': 'TemplateObject', 'pcileech_config': 'TemplateObject', 'timing_config': 'TemplateObject', 'bar_config': 'TemplateObject', 'board_config': 'TemplateObject', 'interrupt_config': 'TemplateObject', 'config_space_data': 'TemplateObject', 'generation_metadata': 'TemplateObject', 'power_management': 'TemplateObject', 'power_config': 'TemplateObject', 'error_handling': 'TemplateObject', 'error_config': 'TemplateObject', 'performance_counters': 'TemplateObject', 'perf_config': 'TemplateObject', 'device_type': 'str', 'device_class': 'str', 'enable_scatter_gather': 'bool', 'enable_interrupt': 'bool', 'enable_clock_crossing': 'bool', 'enable_performance_counters': 'bool', 'enable_error_detection': 'bool', 'enable_custom_config': 'bool', 'device': 'TemplateObject', 'getattr': 'builtin_function_or_method', 'log2': 'function', 'NUM_MSIX': 'int', 'msix_table_bir': 'int', 'msix_table_offset': 'int', 'msix_pba_bir': 'int', 'msix_pba_offset': 'int', 'table_offset_bir': 'int', 'pba_offset_bir': 'int', 'msi_vectors': 'int', 'max_payload_size': 'int', 'enable_perf_counters': 'bool', 'enable_error_logging': 'bool', 'bar': 'list', 'config [... cut ...]
06:23:12 - ERROR - Unexpected error: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.
01:23:12 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

sthgi, 2025-08-24:
Capture Card PCI

Successfully tagged localhost/pcileech-fw-generator:latest
38be2c4df93aca0cb0e2446985f994b514ae171991af38e446a909bda1fa7477
01:24:47 INFO   01:24:47 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
01:24:47 INFO   01:24:47 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
01:24:47 INFO   01:24:47 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
01:24:47 INFO   01:24:47 │  INFO  │ [CONT] Launching build container - board=pcileech_100t484_x1, tag=latest
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for board.name
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for meta
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for process_var
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for title
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for _td
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for error_name
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Registered default fallback for error_value
06:24:47 - INFO -   06:24:47 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
06:24:47 - INFO -   06:24:47 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:03:00.0
06:24:47 - INFO -   06:24:47 │  INFO  │ SystemVerilogGenerator initialized successfully
06:24:47 - INFO -   06:24:47 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
06:24:47 - INFO -   06:24:47 │  INFO  │ No device profile specified, using live device detection
06:24:47 - INFO - ➤ Preloading MSI-X data before VFIO binding
06:24:47 - INFO -   06:24:47 │  INFO  │ MSI-X capability not found
06:24:47 - INFO -   • No MSI-X capability found
06:24:47 - INFO - ➤ Generating PCILeech firmware …
06:24:47 - INFO -   06:24:47 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:03:00.0
06:24:47 - INFO -   06:24:47 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
06:24:47 - INFO -   06:24:47 │  INFO  │ MSI-X capability not found
06:24:47 - INFO -   06:24:47 │  INFO  │ [MSIX] No MSI-X capability found during preload
06:24:47 - INFO -   06:24:47 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
[... cut ...]
06:25:24 - ERROR - Unexpected error: VFIO device access failed
01:25:24 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

sthgi, 2025-08-24:
WLAN PCI

Successfully tagged localhost/pcileech-fw-generator:latest
368ca114de6af088191c3a93e2412e3b9c0d90bcf22cb560908be40327d01f64
01:27:07 INFO   01:27:07 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
01:27:07 INFO   01:27:07 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
01:27:07 INFO   01:27:07 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
01:27:07 INFO   01:27:07 │  INFO  │ [CONT] Launching build container - board=pcileech_100t484_x1, tag=latest
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for board.name
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for meta
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for process_var
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for title
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for _td
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for error_name
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Registered default fallback for error_value
06:27:07 - INFO -   06:27:07 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
06:27:07 - INFO -   06:27:07 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:03:00.0
06:27:07 - INFO -   06:27:07 │  INFO  │ SystemVerilogGenerator initialized successfully
06:27:07 - INFO -   06:27:07 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
06:27:07 - INFO -   06:27:07 │  INFO  │ No device profile specified, using live device detection
06:27:07 - INFO - ➤ Preloading MSI-X data before VFIO binding
06:27:07 - INFO -   06:27:07 │  INFO  │ MSI-X capability not found
06:27:07 - INFO -   • No MSI-X capability found
06:27:07 - INFO - ➤ Generating PCILeech firmware …
06:27:07 - INFO -   06:27:07 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:03:00.0
06:27:07 - INFO -   06:27:07 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
06:27:07 - INFO -   06:27:07 │  INFO  │ MSI-X capability not found
06:27:07 - INFO -   06:27:07 │  INFO  │ [MSIX] No MSI-X capability found during preload
06:27:07 - INFO -   06:27:07 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
[... cut ...]
06:27:43 - ERROR - Unexpected error: No valid MMIO BARs found
01:27:43 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

ramseymcgrath, 2025-08-24:
The first got the furthest, but msix still didnt stick in the context so that bug hasnt been fixed yet
Second one looks better but vfio failure means it tried to go back and look up missing info but the device had been remounted. 
Third one wasnt found with mmio so it mightve been in use too. 
Overall looks better than last release, 2 bugs to fix

ramseymcgrath, 2025-08-24:
build against https://github.com/VoltCyclone/PCILeechFWGenerator/releases/tag/v0.13.9-beta.1 pls

Danielix99, 2025-08-24:
I still get with 0.13.9-beta.1 

23:24:10 - INFO -   23:24:10 │  INFO  │ [PCIL] Template context validation successful
23:24:10 - INFO -   23:24:10 │  INFO  │ Template context built successfully with 39 top-level keys
23:24:10 - INFO -   23:24:10 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:06:00.0
23:24:10 - INFO -   23:24:10 │  INFO  │ Generating SystemVerilog modules
23:24:10 - INFO -   23:24:10 │  INFO  │ Generating PCILeech SystemVerilog modules
23:24:10 - INFO -   23:24:10 │  INFO  │ Generating MSI-X modules
23:24:10 - ERROR -   23:24:10 │ ERROR  │ PCILeech module generation failed: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.
23:24:10 - ERROR -   23:24:10 │ ERROR  │ ERROR TYPE: Data Error
CONTEXT: SystemVerilog generation
DETAILS: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.
SUGGESTION: Check your data format and ensure it's valid according to the expected schema.
23:24:10 - ERROR -   23:24:10 │ ERROR  │ [PCIL] PCILeech firmware generation failed: SystemVerilog generation failed: ERROR TYPE: Data Error
CONTEXT: SystemVerilog generation
DETAILS: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.
SUGGESTION: Check your data format and ensure it's valid according to the expected schema.
23:24:10 - ERROR - Build failed: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.
23:24:10 - ERROR - Unexpected error: MSI-X table data must be read from actual hardware. Cannot generate safe firmware without real MSI-X values.

ramseymcgrath, 2025-08-25:
i need more of the log to see if msix was enabled or not

sthgi, 2025-08-25:
USB3

Successfully tagged localhost/pcileech-fw-generator:latest
451fe333be43f30c12036f168f7ca10e2ee0dc656adfb2fade67b650b3e7be43
18:30:15 INFO   18:30:15 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
18:30:15 INFO   18:30:15 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
18:30:15 INFO   18:30:15 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
18:30:15 INFO   18:30:15 │  INFO  │ [CONT] Launching build container - board=pcileech_100t484_x1, tag=latest
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for board.name
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for meta
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for process_var
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for title
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for _td
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for error_name
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Registered default fallback for error_value
23:30:16 - INFO -   23:30:16 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
23:30:16 - INFO -   23:30:16 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:03:00.0
23:30:16 - INFO -   23:30:16 │  INFO  │ SystemVerilogGenerator initialized successfully
23:30:16 - INFO -   23:30:16 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
23:30:16 - INFO -   23:30:16 │  INFO  │ No device profile specified, using live device detection
23:30:16 - INFO - ➤ Preloading MSI-X data before VFIO binding
23:30:16 - INFO -   23:30:16 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
23:30:16 - INFO -   • Found MSI-X capability: 8 vectors
23:30:16 - INFO - ➤ Generating PCILeech firmware …
23:30:16 - INFO -   23:30:16 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:03:00.0
23:30:16 - INFO -   23:30:16 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
23:30:16 - INFO -   23:30:16 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
[... cut ...]
01:14:46 - ERROR - Unexpected error: No valid MMIO BARs found
20:14:46 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/32               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/pcileechfwgenerator-0.13.9b1/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

mantis-rng, 2025-08-25:
Same for a USB3 PCI 

```
17:27:31 INFO   17:27:31 │  INFO  │ Discovering boards from pcileech-fpga repository...
git version 2.43.0
.git
17:27:31 INFO   17:27:31 │  INFO  │ Discovered board: 35t at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/PCIeSquirrel
17:27:31 INFO   17:27:31 │  INFO  │ Discovered board: pcileech_enigma_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/EnigmaX1
17:27:31 INFO   17:27:31 │  INFO  │ Discovered board: pcileech_pciescreamer_xc7a35 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/pciescreamer
17:27:31 INFO   17:27:31 │  INFO  │ Discovered CaptainDMA board: pcileech_75t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/75t484_x1
17:27:31 INFO   17:27:31 │  INFO  │ Discovered CaptainDMA board: pcileech_35t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t484_x1
17:27:31 INFO   17:27:31 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x4 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x4
17:27:31 INFO   17:27:31 │  INFO  │ Discovered CaptainDMA board: pcileech_35t325_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/35t325_x1
17:27:31 INFO   17:27:31 │  INFO  │ Discovered CaptainDMA board: pcileech_100t484_x1 at /root/.cache/pcileech-fw-generator/repos/pcileech-fpga/CaptainDMA/100t484-1
17:27:31 INFO   17:27:31 │  INFO  │ Discovered 8 boards
17:27:31 INFO   17:27:31 │  INFO  │ [VFIO] Rebuilding VFIO constants for current kernel...
17:27:31 INFO   17:27:31 │  INFO  │ [VFIO] VFIO constants rebuilt successfully
17:27:31 INFO   17:27:31 │  INFO  │ [BUILD] Starting build for device 0000:06:00.0 on board 35t
17:27:32 WARNING   17:27:32 │ WARNING│ [BUILD] Podman not available or cannot connect

============================================================
⚠️  Podman is not available or cannot connect.
============================================================

The build normally runs in a container for consistency.
However, you can run the build locally on your system.

Note: Local builds require all dependencies to be installed.
      (Vivado, Python packages, etc.)

Would you like to run the build locally? [y/N]: y
17:27:33 INFO   17:27:33 │  INFO  │ [LOCAL] Running local build - board=35t
17:27:33 INFO   17:27:33 │  INFO  │ [LOCAL] Executing local build with args: --bdf 0000:06:00.0 --board 35t
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for board.name
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for meta
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for process_var
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for title
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
17:27:33 INFO   17:27:33 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
[... cut ...]
17:28:05 WARNING   17:28:05 │ WARNING│ [FALLBACK] Fallback denied by policy (mode=none) for build-integration: VFIO device access failed
17:28:05 ERROR   17:28:05 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Build integration generation failed: VFIO device access failed
17:28:05 ERROR Build failed: VFIO device access failed
17:28:05 ERROR Unexpected error: VFIO device access failed
17:28:05 ERROR   17:28:05 │ ERROR  │ [LOCAL] Local build failed after 31.4s: Local build failed with exit code 1
17:28:05 ERROR Build failed: Local build failed with exit code 1
```

mantis-rng, 2025-08-25:
@ramseymcgrath lmk if you want to connect in some way to test your fixes in a more efficient setup