VoltCyclone/PCILeechFWGenerator #334: [BUG] 0.13.11

sthgi, 2025-09-04:
USB3

Successfully tagged localhost/pcileech-fw-generator:latest
6406220645180e13460c710c5f60a297908b9795f84e32fb60697043e54fdf8e
22:53:11 INFO   22:53:11 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
22:53:11 INFO   22:53:11 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
22:53:11 INFO   22:53:11 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
22:53:11 INFO   22:53:11 │  INFO  │ [CONT] Launching build container - board=pcileech_100t484_x1, tag=latest
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for board.name
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_part
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for board.fpga_family
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for board.pcie_ip_type
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for sys_clk_freq_mhz
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for generated_xdc_path
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for board_xdc_content
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for max_lanes
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for supports_msi
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for supports_msix
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for ROM_BAR_INDEX
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for ROM_HEX_FILE
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for ROM_SIZE
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for CACHE_SIZE
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SHDW_HI
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for CONFIG_SPACE_SIZE
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for CUSTOM_WIN_BASE
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for kwargs
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for meta
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for opt_directive
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for phys_opt_directive
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for place_directive
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for route_directive
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_n_pin
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for pcie_clk_p_pin
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for pcie_rst_pin
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for pcie_config
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for process_var
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for reg_value
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for temp_coeff
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for title
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for transition_delays
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for varied_value
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for voltage_var
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for _td
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for from_state_value
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for to_state_value
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for error_name
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Registered default fallback for error_value
03:53:11 - INFO -   03:53:11 │  INFO  │ [FALLBACK] Marked 26 variables as critical (no fallbacks)
03:53:11 - INFO -   03:53:11 │  INFO  │ [PCIL] Initializing PCILeech generator for device 0000:03:00.0
03:53:11 - INFO -   03:53:11 │  INFO  │ SystemVerilogGenerator initialized successfully
03:53:11 - INFO -   03:53:11 │  INFO  │ [PCIL] PCILeech generator components initialized successfully
03:53:11 - INFO -   03:53:11 │  INFO  │ No device profile specified, using live device detection
03:53:11 - INFO -   03:53:11 │  INFO  │ ➤ Preloading MSI-X data before VFIO binding
03:53:11 - INFO -   03:53:11 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
03:53:11 - INFO -   03:53:11 │  INFO  │   • Found MSI-X capability: 8 vectors
03:53:11 - INFO -   03:53:11 │  INFO  │ ➤ Generating PCILeech firmware …
03:53:11 - INFO -   03:53:11 │  INFO  │ [PCIL] Starting PCILeech firmware generation for device 0000:03:00.0
03:53:11 - INFO -   03:53:11 │  INFO  │ [MSIX] Preloading MSI-X data from sysfs before VFIO binding
03:53:11 - INFO -   03:53:11 │  INFO  │ MSI-X capability found: 8 entries, table BIR 0 offset 0x1000, PBA BIR 0 offset 0x1080
03:53:11 - INFO -   03:53:11 │  INFO  │ [MSIX] Preloaded MSI-X capability: 8 vectors, table BIR 0, offset 0x1000
03:53:11 - INFO -   03:53:11 │  INFO  │ [MSIX] Capturing device behavior profile for 30.0s
03:53:14 - INFO -   03:53:14 │  INFO  │ [PROFILER] Monitoring started
03:53:46 - INFO -   03:53:46 │  INFO  │ [MSIX] Captured 40603 register accesses with 2 timing patterns
03:53:46 - INFO -   03:53:46 │  INFO  │ [BIND] Starting VFIO binding process for device 0000:03:00.0
03:53:46 - INFO -   03:53:46 │  INFO  │ [BIND] Current driver for 0000:03:00.0: vfio-pci
03:53:46 - INFO -   03:53:46 │  INFO  │ [BIND] Device 0000:03:00.0 already bound to vfio-pci
03:53:46 - INFO -   03:53:46 │  INFO  │ [VFIO] VFIO binding established for device 0000:03:00.0 at /dev/vfio/32
03:53:46 - INFO -   03:53:46 │  INFO  │ [MSIX] Analyzing configuration space for device 0000:03:00.0 (VFIO already active)
03:53:46 - INFO -   03:53:46 │  INFO  │ [CNFG] Attempting to read config space from /sys/bus/pci/devices/0000:03:00.0/config
03:53:46 - INFO -   03:53:46 │  INFO  │ [CNFG] Config space file exists: /sys/bus/pci/devices/0000:03:00.0/config
[... 612 lines cut ...]
04:06:23 - INFO -   04:06:23 │  INFO  │ [BARS] Added BAR 0: BAR 0: memory @ 0x00000000c4000000 (64-bit, non-prefetchable, size=0x10000)
04:06:23 - INFO -   04:06:23 │  INFO  │ [BARX] Completed BAR extraction: found 1 active BARs
04:06:23 - INFO -   04:06:23 │  INFO  │ [LOOKUP] Pre-fallback device_info (shape): {'command': 'str', 'status': 'str', 'cache_line_size': 'str', 'latency_timer': 'str', 'header_type': 'str', 'bist': 'str'}
04:06:23 - INFO -   04:06:23 │  INFO  │ [LOOKUP] Pre-fallback device_info (sanitized): {
  "bist": "0",
  "cache_line_size": "0",
  "command": "2",
  "header_type": "0",
  "latency_timer": "0",
  "status": "16"
}
04:06:23 - INFO -   04:06:23 │  INFO  │ Context built successfully: 14e4:1677
04:06:23 - INFO -   04:06:23 │  INFO  │ [PCIL] Template context validation successful
04:06:23 - INFO -   04:06:23 │  INFO  │ Template context built successfully with 39 top-level keys
04:06:23 - INFO -   04:06:23 │  INFO  │ [VFIO] VFIO binding cleanup completed for device 0000:03:00.0
04:06:23 - INFO -   04:06:23 │  INFO  │ Generating SystemVerilog modules
04:06:23 - INFO -   04:06:23 │  INFO  │ Generating PCILeech SystemVerilog modules
04:06:23 - INFO -   04:06:23 │  INFO  │ Generated 4 PCILeech SystemVerilog modules
04:06:23 - INFO -   04:06:23 │  INFO  │ Generated 4 SystemVerilog modules
04:06:23 - INFO -   04:06:23 │  INFO  │ Generating additional firmware components
04:06:23 - WARNING -   04:06:23 │ WARNING│ [FALLBACK] Fallback denied by policy (mode=none) for build-integration: VFIO device access failed
04:06:23 - ERROR -   04:06:23 │ ERROR  │ [PCIL] PCILeech firmware generation failed: Build integration generation failed: VFIO device access failed
04:06:23 - ERROR -   04:06:23 │ ERROR  │ Build failed: VFIO device access failed
04:06:23 - ERROR -   04:06:23 │ ERROR  │ Unexpected error: VFIO device access failed
23:06:24 ERROR Build failed: Command 'podman run --rm --privileged               --device=/dev/vfio/18               --device=/dev/vfio/vfio               --entrypoint python3               --user root               -v /root/PCILeechFWGenerator/output:/app/output               -v /lib/modules/$(uname -r)/build:/kernel-headers:ro               pcileech-fw-generator:latest               -m src.build --bdf 0000:03:00.0 --board pcileech_100t484_x1' returned non-zero exit status 1.

i-yyi, 2025-09-04:
Same issue, mark

ramseymcgrath, 2025-09-04:
fixed

closed 2025-09-04 by ramseymcgrath